jasonacox/pypowerwall #1: Powerwall Solar Vitals and String Data

jasonacox, 2021-12-05:
This is a cross-reference link to a discussion at https://github.com/vloschiavo/powerwall2/issues/51 on the Tesla Powerwall Gateway `/api/devices/vitals` endpoint.  This issue is to troubleshoot and help provide help getting real time string data to trend and identify system issues.

By "string" I mean each string of solar panels that are connected together and terminated at the Powerwall solar inverter.  The  Powerwall+ allows for 4 strings each, identified as A, B, C and D.  If you have multiple inverters, the pyPowerwall `pw.strings()` function will append a number to the letter to accommodate multiple inverters (A, B, C, D, A1, B1, C1, D1, A2, ...).

## Vitals Background
* The `/api/devices/vitals` endpoint responds with a protobuf binary payload.  Thanks to the great work by @brianhealey, we have a [tesla.proto](https://github.com/jasonacox/pypowerwall/blob/main/tesla.proto) file that can be used to create python interfaces to the data.  pyPowerwall now uses this to decode the *vitals* payload through two simple functions, `vitals()` and `strings()`:

```bash
# Install pyPowerwall package
pip install pypowerwall
```

```python
import pypowerwall

# Credentials for your Powerwall - Customer Login Data
password='password'
email='email@example.com'
host = "localhost"                # Change to the IP of your Powerwall
timezone = "America/Los_Angeles"  # Change to your local timezone/tz

# Connect to Powerwall
pw = pypowerwall.Powerwall(host,password,email,timezone)

# Display Vitals (returns details as dictionary or JSON)
vitals = pw.vitals(jsonformat=False)
print(vitals)

# Display String Data 
strings = pw.strings(jsonformat=False)
print(strings)

```

## String Data

I wanted to be able to capture string data to see if my system was performing correctly.  If you want to see it in human readable format, just specify jsonformat as True like this:
```python
strings = pw.strings(jsonformat=True)
print(strings)
```
Output:
```json
{
    "A": {
        "Connected": true,
        "Current": 0.48,
        "Power": 115.0,
        "State": "PV_Active",
        "Voltage": 251.8
    },
    "B": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -2.3999999999999995
    },
    "C": {
        "Connected": true,
        "Current": 1.82,
        "Power": 526.0,
        "State": "PV_Active",
        "Voltage": 293.6
    },
    "D": {
        "Connected": true,
        "Current": 1.81,
        "Power": 521.0,
        "State": "PV_Active_Parallel",
        "Voltage": 294.0
    }
}
```
I push that into a tiny installation of Splunk (see [TinySplunk](https://github.com/jasonacox/TinySplunk)) and get a graph like this:

[image]

Here is an example script I used to pull the string data and stuff it into Splunk:

```python
# pyPowerwall - Powerwall Solar String data to Splunk

from splunk_http_event_collector import http_event_collector
import logging
import sys
import pypowerwall

# Credentials for your Powerwall - Customer Login Data
password='password'
email='email@example.com'
host = "localhost"                # Change to the IP of your Powerwall
timezone = "America/Los_Angeles"  # Change to your local timezone/tz

# Connect to Powerwall
pw = pypowerwall.Powerwall(host,password,email,timezone)

# Grab strings data from Powerwall
strings = pw.strings(jsonformat=False)

# init logging config, this would be job of your main code using this class.
logging.basicConfig(format='%(asctime)s %(name)s %(levelname)s %(message)s', datefmt='%Y-%m-%d %H:%M:%S %z')

# Create event collector object, default SSL and HTTP Event Collector Port
http_event_collector_key = "xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx"
http_event_collector_host = "10.1.1.101"

# Setup Splunk HEC Connector
splunk = http_event_collector(http_event_collector_key, http_event_collector_host)
splunk.log.setLevel(logging.ERROR)
    
# perform a HEC reachable check
hec_reachable = splunk.check_connectivity()
if not hec_reachable:
    sys.exit(1)

# break apart JSON into event items
event = {}
try:
    for k, v in strings.items():
        # key and value are payload event items
        event.update({k:v})
    
    # Build payload with metadata information
    payload = {}
    payload.update({"index":"main"})
    payload.update({"sourcetype":"solar-strings"})
    payload.update({"source":"http-stream"})
    payload.update({"host":"pypowerwall"})
    payload.update({"event":event})
    splunk.sendEvent(payload)
    
    splunk.flushBatch()

except:
    print("ERROR: Unable to send %s" % SENSOR_FILE)
```

## Neruio Solar Meter Monitoring

I also have occasional trouble with the Neruio solar meter that Tesla installed in my Powerwall+ so I also have scripts to watch that.  It shows up in the `vitals` payload as a device that I monitor and capture to create alerts (e.g. if it doesn't show up in the `vitals` payload that means it has lost connection with the gateway and shut down solar production):

```json
"NEURIO--VAHxxxxxxxxxx": {
        "NEURIO_CT0_InstRealPower": 1333.5606370817025,
        "NEURIO_CT0_Location": "solar",
        "Parent": "STSTSM--xxxxxxx-00-E--TGxxxxxxxxxxxx",
        "firmwareVersion": "1.6.1-Tesla",
        "lastCommunicationTime": "1638720520",
        "manufacturer": "NEURIO",
        "partNumber": "",
        "serialNumber": "VAHxxxxxxxxxx"
    }
```
I have a script that pulls this out of the vitals payload and stuffs it into Splunk as well.  Below is an example showing a dropout where the Neurio loses connection and reconnects:

[image]

adamurban, 2021-12-16:
Does this string function only work if you have a powerwall+ configuration?

I have tesla inverters, but they're configured standalone. Config is 2 7.6kW tesla inverters, 3 powerwalls and one gateway... So 8 MPPTs total I'd like to track. But I have yet to see any string level data on the gateway. The only place I can see it is via tedapi and/or on the inverter's local web interface, but that's mostly useless for monitoring as it requires connecting to the inverter's dedicated SSID(s) and powercycling them to read the data.

Here's what I see when I try the new strings() output: 

```
DEBUG:pypowerwall [0.1.0]

DEBUG:loaded auth from cache file .powerwall
DEBUG:Starting new HTTPS connection (1): 10.0.0.192:443
DEBUG:https://10.0.0.192:443 "GET /api/system_status/soe HTTP/1.1" 200 32
Battery power level: 71%
DEBUG:Starting new HTTPS connection (1): 10.0.0.192:443
DEBUG:https://10.0.0.192:443 "GET /api/meters/aggregates HTTP/1.1" 200 None
Power response: {'site': 2533.170000076294, 'solar': -12, 'battery': -10, 'load': 2499.3416846925347}
Grid Power: 2.53kW
Solar Power: -0.01kW
Battery Power: -0.01kW
Home Power: 2.50kW
Grid raw: {'last_communication_time': '2021-12-16T02:33:41.135759439-04:00', 'instant_power': 2533.170000076294, 'instant_reactive_power': -1190.289999961853, 'instant_apparent_power': 2798.882014893754, 'frequency': 60.0099983215332, 'energy_exported': 264944.78814683843, 'energy_imported': 3746632.778888163, 'instant_average_voltage': 169.18782024742725, 'instant_average_current': 27.628999999999998, 'i_a_current': 0, 'i_b_current': 0, 'i_c_current': 0, 'last_phase_voltage_communication_time': '0001-01-01T00:00:00Z', 'last_phase_power_communication_time': '0001-01-01T00:00:00Z', 'timeout': 1500000000, 'num_meters_aggregated': 2, 'instant_total_current': 27.628999999999998}
Solar raw: {'last_communication_time': '2021-12-16T02:33:41.190430341-04:00', 'instant_power': -12, 'instant_reactive_power': 90, 'instant_apparent_power': 90.79647570252934, 'frequency': 0, 'energy_exported': 3078979.3115712255, 'energy_imported': 22353.287072459854, 'instant_average_voltage': 214.6010950577839, 'instant_average_current': 0.393, 'i_a_current': 0, 'i_b_current': 0, 'i_c_current': 0, 'last_phase_voltage_communication_time': '0001-01-01T00:00:00Z', 'last_phase_power_communication_time': '0001-01-01T00:00:00Z', 'timeout': 1500000000, 'num_meters_aggregated': 1, 'instant_total_current': 0.393}
DEBUG:Starting new HTTPS connection (1): 10.0.0.192:443
DEBUG:https://10.0.0.192:443 "GET /api/devices/vitals HTTP/1.1" 200 None
DEBUG:Found 13 devices.
String Data: {}

DEBUG:Found 13 devices.
```

My vitals show 13 devices:
1 NEURIO
1 STSTSM
3 TEPINV
3 TEPOD
3 TETHC
1 TESLA
1 TESYNC

jasonacox, 2021-12-17:
Hi @adamurban - Thanks for the note!  Unfortunately, you may have a very valid point.  There is a strong possibility that this is only available on the Powerwall+.   I don't see a PVAC or PVS device data in your vitals list. That is where I find the string data on my Powerwall+ setup.   Do you see the string data when you log in to the Tesla gateway portal?  For me it shows this:

[image]

The web portal is actually rendering this data from the protobus payload from `/api/devices/vitals`.  If you do see it on the portal, there may be another API endpoint or something we are missing in the vitals output.  Look at the output for `pw.vitals()` and see if you see anything resembling this:

```json
    "PVAC--1538100-00-F--...": {
...
        "PVAC_PVCurrent_A": 0.03,
        "PVAC_PVCurrent_B": 0.0,
        "PVAC_PVCurrent_C": 0.12,
        "PVAC_PVCurrent_D": 0.06,
        "PVAC_PVMeasuredPower_A": 1.0,
        "PVAC_PVMeasuredPower_B": 0.0,
        "PVAC_PVMeasuredPower_C": 3.0,
        "PVAC_PVMeasuredPower_D": 2.0,
        "PVAC_PVMeasuredVoltage_A": 15.8,
        "PVAC_PVMeasuredVoltage_B": 33.2,
        "PVAC_PVMeasuredVoltage_C": 14.8,
        "PVAC_PVMeasuredVoltage_D": 14.0,
...
        "PVAC_PvState_A": "PV_Disabled",
        "PVAC_PvState_B": "PV_Disabled",
        "PVAC_PvState_C": "PV_Disabled",
        "PVAC_PvState_D": "PV_Disabled",
...
    },
    "PVS--1538100-00-F--...": {
...
        "PVS_StringA_Connected": true,
        "PVS_StringB_Connected": false,
        "PVS_StringC_Connected": true,
        "PVS_StringD_Connected": true,
...
    },
```

dailow, 2022-01-01:
It appears the neurio data is no longer available from the vitals API with the latest firmware update.  Is there any other way to get the neurio data?  Seems like the data from the /meters API isn't available while the system is running.

jasonacox, 2022-01-01:
I noticed the same thing.  One benefit seems to be that when the Neurio disconnects from the gateway (which happens frequently for me) it no longer stops solar generation.  That was seriously annoying.  Early this week Tesla scheduled a maintenance visit to "fix reporting issues" related to the solar meter (Neurio) which amounted to them re-commissioning the system.  After the commissioning they said it was reporting to Tesla but it still wasn't showing up in the `/api/devices/vitals`API.  

I noticed that the /systems portal (installer login) also no longer shows the Neurio device.  The portal uses `/api/devices/vitals` so that makes sense, but I was hoping it would reveal a new API for the Neurio.  I'll poke around and see if I can find it anywhere else.  However, I suspect this update is the first of Tesla reducing dependency on the Neurio.  It is an expensive and highly unreliable device that they could do themselves through the inverter, at least for the Powerwall+.

I have no problem getting data from the `/api/meters/aggregates` API endpoint.  Does this work for you?

```python
import pypowerwall

# Credentials for your Powerwall - Customer Login Data
password='password'
email='email@example.com'
host = "localhost"                # Change to the IP of your Powerwall
timezone = "America/Los_Angeles"  # Change to your local timezone/tz

# Connect to Powerwall
pw = pypowerwall.Powerwall(host,password,email,timezone)

# Poll /meters API
print(pw.poll('/api/meters/aggregates'))
```

I noticed that for `/api/meters/status` I'm now getting this which is probably because the Neurio has failed AGAIN. 

```json
{"status":"failed_meter","errors":["context deadline exceeded"],"serial":"VAHxxxxxxxxxx"}
```

dailow, 2022-01-01:
I have two Neurio's because I have a seperate solar array and was trying to monitor them separately, which I could before since each meter shows up in the vitals API.  For now I am just calculating it by subtracting the total solar load from all the string production.  Thank you for putting this together, nice not to have to maintain my custom scripts!

s0schen, 2022-01-02:
> # Credentials for your Powerwall - Customer Login ***@***.***'host = "localhost"                # Change to the IP of your Powerwalltimezone = "America/Los_Angeles"  # Change to your local timezone/tz

Interesting about the Neurio.  They dropped off my installer page as well.
My pre PTO solar output is much greater today with finally a smooth looking
graph. Output today is about 30% increased.

I wonder if my jagged looking solar output was due to neurio issues?

string voltage yesterday was steady but today it is jagged with a result of
more current and total output.

Firmware 21.44 223a5cd

jasonacox, 2022-01-03:
Hi @s0schen, Yes, I believe you are correct.  I noticed the same thing. When the Neurio was running with the old firmware, it would shut down solar (often for just minutes) because the Neurio would go into a bad state.  I was so frustrated with it, that I actually re-comissioned my system myself and purposefully removed the Neurio from the configuration.  The Neruio is crap technology that needs to be replaced.

It turns out that the Neurio really isn't needed if you have a Powerwall+ (and possibly others).  I was getting way better and more consistent solar production without the Neurio, as you observed.  If you are interested in the full story of my Tesla + Neurio adventures (nightmares?), you can read it here: https://www.jasonacox.com/wordpress/archives/1017

I suspect Tesla is discovering the same thing and this new Firmware doesn't stop solar production when the Neurio freaks out.  It will be interesting to see if this holds up our PTO (I'm pre-PTO as well!)

FWIW for others wondering what Firmware version they are running, in the latest pypowerwall version (v0.1.2), I added some easy to use functions to pull firmware version, uptime, etc:

```python
import pypowerwall
pw = pypowerwall.Powerwall(host,password,email,timezone)

# Some System Info
print("Site Name: %s - Firmware: %s - DIN: %s" % (pw.site_name(), pw.version(), pw.din()))
print("System Uptime: %s\n" % pw.uptime())
```

dailow, 2022-01-03:
Thank you @jasonacox for the link to your blog and all the steps you tried.  Every other week I keep seeing the production of all the strings drop for a minute and then resume, which I thought might have been a software issue because resetting everything fixes the issue. I am still seeing the same issue with the latest firmware, so not sure if it's the same issue but I may try try removing the neurio altogether to see if it helps.  In the latest firmware on the CT screen there is a checkbox now that asks if it's for monitoring a PW+ inverter and that it's only used for metering.

jasonacox, 2022-01-03:
Thanks @dailow !  If you are able and willing, can you grab some screenshots of what you see (make sure you mask any confidential data)?  Also, same, if you decide to 'Run Wizard" to remove the Neurio?  

Fair warning, for others too, I suspect removing the Neruio may impact Tesla's efforts to finish PTO.  They already contacted me to "do maintenance to fix metering before PTO" last week which mean having them come out and re-commission with the Neurio. :)

dailow, 2022-01-03:
Here you can see that my solar graph looks pretty choppy, even though it was pretty sunny outside without really any clouds.  I then looked at data that I capture from the vitals API and see that the power for all my strings drop to 0 at the same time, so I reboot my inverter and you can see it looks smooth afterwards.  What I can't explain is it seems to also impact how the PW draws during peak times.  As you can see prior to the reboot even during peak hours when the sun is down, it will sporadically draws from the grid, but after the reboot its flat and will work like that for a while until I have to reboot it again.  Last screenshot you can see the new toggle where you have to say if the CT is measuring a PW+ or not, not really sure what the Revenue Grade Metering is referring to.  Next time it happens I will check to see if the Neurio status is online, but I am leaning toward some SW issue potentially with the TOU setting.

[image]
[image]
[image]
[image]
[image]

jasonacox, 2022-01-03:
Thanks @dailow !

That definitely looks like the pattern I would see when the Neurio was acting up.  When you see the jaggies, if you open the Tesla Gateway box and look at the light on the Neurio, you will see it flicker between purple and blue.  That's an indication it has become unstable.  I just unplug the Neurio power (at the bottom, be careful, 120v) let it sit for 5s and plug it back in.  That usually clears it up. 

On my system, when the Neurio was in a bad state, it would also impact Powerwall supply too (when the Powerwall were powering the house) and I would see Grid power spike up too.  Again, resetting the Neurio would fix it for a while.

On the "Is this measuring a Powerwall+ Inverter" bit - thanks for the post!  I've done some research on the Neurio.  It's not a quality device by any means, but for some reason it is certified as a "Revenue Grade Meter" as you mention.  That means that Tesla can use that to show solar generation to get Solar renewable energy certificates (SRECs) which they had us assign to them during in the purchase agreement.  They sell these to companies needing renewable energy credit offsets.  I asked to have it removed from my contract, but Tesla refused.  I did some more research and discovered it would be very difficult for me to sell SRECs for such a small system.  The bartering systems are geared toward massive gigawatt offsets so I was less concerned about Tesla getting them.  However, this is clearly the reason they want a certified revenue grade meter in the system. I suspect they are trying to get the Powerwall+ native inverter's meter certified but until they do, I fully expect them to hound us to keep it running. 

I agree that it is likely a software issue, but I suspect it is due to the Neurio's firmware which was customized for Tesla. Others are using the retail [Neurio](http://neur.io) device (now owned by Generac) called the "Generac W2HEM GNRC PWRview Monitor" ([Link](https://www.amazon.com/Generac-W2HEM-PWRview-Monitor-Orange/dp/B0874G7T8H/ref=asc_df_B0874G7T8H/?tag=hyprod-20&linkCode=df0&hvadid=459573051287&hvpos=&hvnetw=g&hvrand=12094941469286589350&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1014362&hvtargid=pla-955173521012&psc=1)) seem to have good reviews.  When I did have device vitals on the Neurio, I noticed that despite other firmware upgrades, the Neruio was never upgraded and likely cannot be upgrade due to the special W2-Tesla code burned into it (which Generac support confirmed).

I could be wrong about your case.  Some of those jags look like clouds, but the ones that drop to zero are similar to my Neurio issues.  These jags could also be indicators of problems with the inverter too.  That may be caused by frequency changes or solar voltage rise.  I have been monitoring all the voltages on the gateway (pulling from vitals) and it has been surprising to see the variance.  I wonder if the inverters trigger down when grid voltages rise.  I hear they will for frequency changes (as a signal by the utility to stop producing in cases of over production).

Voltages over 24hr
[image]

Frequencies over 24hr
[image]

dailow, 2022-01-03:
Thanks for all the data points @jasonacox .  Weird that the Neurio would impact the Powerwall supply, but I hope Tesla can fix these issues in a future firmware update.  Interesting point about the SREC's, I hadn't really thought about it wasn't really available in California.  I recently just got PTO, so if I see this happen again I am going to just go ahead and remove the Neurio from my system.

s0schen, 2022-01-03:
Here are my graphs if its of interest.  Two PW+ & Two PW   No SRECs’s in TX.     Thanks @jasonacox <https://github.com/jasonacox>.

jasonacox, 2022-01-03:
Wow, 4 PWs total?  Nice setup @s0schen !  However, the  graphs didn't come through (email?) - can you paste them on github: https://github.com/jasonacox/pypowerwall/issues/1

s0schen, 2022-01-03:
[image]
[image]

second try.

Yes 4..  some buy a 3rd car, others put in a pools.  We bought powerwalls. :)   I'm glad pypowerwall and influxdb recorded from the second inverter without additional coding btw.   Thanks.

dailow, 2022-01-05:
@jasonacox  A new firmware update was pushed to my unit this morning and looks like they brought back the Neurio data in the /vitals API, but only for my other Neurio device so I think they are right they are working on some updates around monitoring.  Didn't capture the previous of Neurio firmware, but mine currently shows 1.6.1-Tesla. 

Added more details to how the CT is configured (ignore, I was just seeing what the changes did)
[image]

s0schen, 2022-01-05:
Same update I think.  Firmware 21.44.1 c58c2df3 (c58c2df39ec)   I see the Neurio's back on the system information page and listed as Solar Revenue Only.     I don't have a CT2.   Just a CT1 for each PW+

Changed Influx raw retention to 7 days so I can see a week of string data.

dailow, 2022-01-05:
> Same update I think. Firmware 21.44.1 c58c2df3 (c58c2df39ec) I see the Neurio's back on the system information page and listed as Solar Revenue Only. I don't have a CT2. Just a CT1 for each PW+
> 
> Changed Influx raw retention to 7 days so I can see a week of string data.

Oh interesting.   I have two Neurio's in my system because I have an existing solar system but only one gets listed.  The one for the PW+ only has one CT, but doesn't show up in the system portal for some reason.

jasonacox, 2022-01-06:
Very interesting!  Same here.  I see Firmware `21.44 223a5cd` upgraded to `21.44.1 c58c2df3`.

I believe  device /vitals API only reports the PW+ Neurio.  Is that what you are seeing?  

I now see the Neurio in my PW+ /vitals:
```json
    "NEURIO--VAHxxxxxxxxxx": {
        "NEURIO_CT0_InstRealPower": -16.239999771118164,
        "NEURIO_CT0_Location": "solarRGM",
        "Parent": "STSTSM--1232100-00-E--TGxxxxxxxxxxxx",
        "firmwareVersion": "1.6.1-Tesla",
        "lastCommunicationTime": "1641439535",
        "manufacturer": "NEURIO",
        "partNumber": "",
        "serialNumber": "VAHxxxxxxxxxx"
    },
```

Also, on to the theory that they are switching to using the Neurio for revenue grade metering only  (RGM perhaps for SREC as we discussed), I noticed the location name has changed.  Previously, the Neurio would report this location: 
```
     'NEURIO_CT0_Location': 'solar'
```
Now it reports:       
```
     "NEURIO_CT0_Location": "solarRGM"
```

I'm polling /vitals and had a graph for the Neurio when it was there before.  It flatlined during the 21.44 firmware but is now back with 21.44.1.  I can see that my upgrade happened at about 8:55am PT:

[image]

It will be interesting to see if connectivity issues still occur with the Neurio and how it impacts the rest of the PW+, if at all.

brianhealey, 2022-01-16:
@jasonacox FYI, I updated the protocol buffers to fix some issues I was seeing.

In terms of getting your NEURIO devices to show, I found that it depends on what you have set in the wizard for your meters. Depending on which settings I use, they might show in the UI (or not).

There are quite a few endpoints that might be interesting for people looking for their meter/strings details.  Hint: download the app.js from your powerwall and search for `api.url`.

jasonacox, 2022-01-17:
Thanks @brianhealey !  Thanks for the updates!!!   And thanks for the PR.  Make sure the email you used for the commit is included in your Github account (go to https://github.com/settings/emails) so that you get recognized as the contributor.

And thanks for the app.js tip... :eyes:

jasonacox, 2022-01-21:
> There are quite a few endpoints that might be interesting for people looking for their meter/strings details. Hint: download the app.js from your powerwall and search for `api.url`.

That's a treasure - here is a list of endpoints in app.js:
https://github.com/jasonacox/pypowerwall/blob/main/docs/api.txt

I have started a list of the device names that show up in the vitals output (e.g. TETHC) along with their ECU type, alerts and a brief description.  These are just "best guesses" at this point but may be helpful for others (feedback welcome!): https://github.com/jasonacox/pypowerwall/tree/main/docs#devices

dailow, 2022-01-21:
Thanks for documenting all of this @jasonacox!  My PW recently was updated and can see that my Neurio firmware is at 1.7.1-Tesla, hope that will help fix some of the issues you were seeing.  I saw this new alert pop up after the update, but don't know what it refers to.  

THC_w155_Backup_Genealogy_Updated

jasonacox, 2022-01-24:
Agree! I'm still on `1.6.1-Tesla`. and see the Neurio drop out several times a day (see graph below).  Since the last PW firmware upgrade, Neurio glitching doesn't seem to impact *real* solar generation, just metrics from the Neurio  RGM meter.  Hopefully the Neurio firmware upgrade will fix that.  Did your PW firmware upgrade?  I'm still on `21.44.1 c58c2df3`. 

[image]

"THC_w155_Backup_Genealogy_Updated" - Genealogy?  Very interesting.

dailow, 2022-01-24:
I'm on 22.1 92118b67 now.  No obvious changes from what I can see.

guillochon, 2022-05-18:
I'm not sure if this is the right issue to raise this on, but I just got my Solaredge inverter replaced with a Tesla brand inverter and I was wondering if pypowerwall could be used to get any info from the inverter? I see that `/tedapi/din` returns something but none of the other endpoints seem to work that I've tried...

jasonacox, 2022-05-19:
Hi @guillochon - I'm not sure.  I see in this https://www.reddit.com/r/TeslaSolar/comments/plew0k/tesla_inverter_local_api/ thread they talk about having to do a physical auth to log in to get to the API data on that inverter.  Do you have a Powerwall or Tesla Gateway in addition to the inverter?   I know pyPowerwall can talk to the gateway and would be curious if it sees the new inverter. 

How did you get to the /tedapi/din?

guillochon, 2022-05-19:
I do not have a Powerwall, just the inverter. The /tedapi/din is just from the inverter's IP address, which I can see connected to my router. In that reddit thread it says you can POST against these endpoints and get some info back but I wasn't sure if that'll work for what I've got (I'm a bit confused on what's the "gateway" here...all I have is a single white box on my wall that's connected to WiFi without the need for another device to act as a gateway).

guillochon, 2022-05-19:
Some of the endpoints I try to hit give me 404s, and some, e.g. `/api/1/vitals`, give me an error: `wsgi handler failed`

jasonacox, 2022-05-20:
That makes sense now. The gateway I'm talking about is only used with Powerwalls to disconnect your house physically from the grid (allowing you to have solar or battery powering your house even when the grid is down). 

I'm assuming you are just using a browser or command line tool to query these API endpoints.  Is that correct? The second error message indicates they are using something of an internal proxy to service up different API data backends.  WSGI stands for Web Server Gateway Interface.

Do you get any pages to render in a browser when you go to the IP address of your inverter?  If so, using a tool like Chrome's View->Developer->Developer Tools function will show you API calls that the browser is making.  It could give insight into what is needed to get the data you seek from your inverter, provided the data is on the webpage.  This is essentially how we found most of the API for the Powerwall used in pyPowerwall.

guillochon, 2022-05-20:
@jasonacox That's correct, just a browser. Going to the IP address of the inverter just shows a generic page saying that "inverter cannot be configured on this network." All I could glean looking from the source was the `/tedapi/din` call.

That being said, I've figured out that the Tesla API can be accessed from a computer using the `teslapy` package and authenticating with a refresh token. That at least allows me to see the current power generation in watts. That's of course getting the data from their servers, but so far it seems pretty reliable.

jasonacox, 2022-05-21:
Nice job @guillochon !   Can you share your `teslapy` code and the steps you took to get the token?  I think there are many others here that would find that helpful.

guillochon, 2022-05-27:
Yeah sure, here is what I wrote up. I run this as a cron job every 5 minutes, it reads the data from the API and then publishes that to an MQTT topic, which I then read in Homebridge using the mqttthing plugin:

```
import teslapy
import paho.mqtt.publish as publish

refresh_token = "your_refresh_token_goes_here"

with teslapy.Tesla('guillochon@gmail.com') as tesla:
    if not tesla.authorized:
        tesla.refresh_token(refresh_token=refresh_token)

    power = str(tesla.solar_list()[0]["solar_power"])
    print(f"{power} watts")
    
    publish.single("tesla_solar/solar_power", power)
    publish.single("tesla_solar/status", "true")
```

To get my refresh token I followed the instructions here, you need to download either an iOS or Android app to do it: https://teslascope.com/help/generating-tokens

My understanding is that you can get historical data from this same API if you wish, but I was only interested in getting the current values to use in some home automation logic.

thetomfox, 2022-09-20:
Thank you for the tremendous effort putting this together.

Everything seems to be working for me, but I'm not getting any data for String Voltage, String Current, String Power, or Inverter Power. When I poll /strings I get an empty set back --- {}

Any ideas on what might be going on?

2x Part 3012170-05-B Powerwalls
1 Inverter
1 TEG

reported version is 22.18.3 21c0ad81

jasonacox, 2022-09-21:
Thanks @guillochon !!

@thetomfox - Do you have a Powerwall+?  The 3012170-05-B can be used for Powerwall+ but since you mention "Inverter", if that is not a Powerwall+, unfortunately you will not get the String or Inverter data from the Powerwall API.

thetomfox, 2022-09-21:
Honestly, I'm not sure how to tell other than looking at the serial number. 

I definitely have a standalone inverter. It is mounted to the wall right next to the gateway and the powerwalls. hmm. There must be a way to poll the inverter directly... That's annoying. Thanks for the reply!

jasonacox, 2022-09-22:
Thanks @thetomfox - please let us know if you figure out a way to poll it!  Several in the community have the same setup.  

Just in case are just missing it, you can look at the Powerwall vitals payload: http://localhost:8675/vitals (replace localhost with the address of the computer running your dashboard).

wcombs, 2022-12-23:
@thetomfox - I'm having same issue with empty set {} returned for strings. Wondering if any luck getting strings-level data with your setup yet.

My just-installed (all Tesla) setup is:

2x Inverters
1x Powerwall
1x Gateway (firmware 22.26.2 8cd8cac4)

thetomfox, 2022-12-23:
I just gave up. Not important enough to spend the time so far.

wcombs, 2022-12-23:
> I just gave up. Not important enough to spend the time so far.
> […](#)

Got it thanks. Will report back if I find a solution

jasonacox, 2022-12-23:
Thanks for the updates!

It seems that only the Powerwall+ (built-in Tesla inverter, not external) provides string data. However, I could be wrong. I only have a Powerwall+ so I'm not able to see other API endpoint that could contain more information for external inverters. 

In case anyone has the time to explore, the [API endpoints](https://github.com/vloschiavo/powerwall2) used by pyPowerwall were discovered using the Tesla gateway UI built for local customer access and commissioning (see Tesla [instructions](https://www.tesla.com/support/energy/powerwall/own/monitoring-from-home-network)). By viewing "web page source code" we can see the API calls and Javascript used to display information about the system.  If you see string or other important inverter data on your Gateway UI, you can use pyPowerwall to fetch those payloads.  For example:

```python
import pypowerwall

pw = pypowerwall.Powerwall(host,password)

print(pw.poll('/api/troubleshooting/problems'))
```

I know many people would love to get this data for their external inverters. Please post if you discover anything. 🙏  Happy holidays and Merry Christmas everyone! 🎄

Caladain, 2023-10-29:
Perhaps I can be of help.  I have a weird setup, two powerwall+’s with one Neurio and a gateway built into one of the inverters that pulls the string data from both of those inverters just fine.  I have a third inverter, that has its own gateway board and neurio, and that one does not return the string data with the normal APIs calls.

This gateway refuses any sort of web interface (just a single page saying grab the Tesla pros app, no way to bypass)

I did some packet capturing, and it appears that the Teslapro app (which does show the individual string data) is hitting the tedapi/v1 endpoint with a variety of what I assume are protobuf rpc.  One of which is definitely the vitals information.  I *can* hit this endpoint over my local network, but it bounces my plain Post with a not authorized response.

It does also hit tedapi/din (which I can also manually do) to get this id, which appears as part of all the other post’s params.  I can do a full packet capture and upload somewhere if that would help.

jasonacox, 2023-10-30:
Hi @Caladain !  Thanks for the offer to help. Interesting discovery about `tedapi/v1` endpoint.  The authentication is going to be tricky.  We are able to auth using the customer email/password for most systems but you mention you can't bypass the Tesla Pros upgrade screen (can you give us a screenshot?). 

You mention the Tesla Pros app works.  I haven't tried it myself, probably should. Can you explain how you connect it to your system? I wonder if there are any credentials used that we could replicate. 

```python
import pypowerwall

# Connect to endpoint
pw = pypowerwall.Powerwall(host,password)

# Try to grab raw protobuf payload from API
result = pw.poll(api='/tedapi/v1', raw=True, force=True)

print(result)
```

jasonacox-sam, 2026-09-06:
This was the original issue tracking solar vitals/string data for *external* (non-Powerwall+) inverters. The long thread established that string data only comes from integrated Tesla inverters (PW+), walked through teslapy cloud access for external inverters, and ultimately led to the TEDAPI work that now powers local inverter/gateway access — including the new bearer-auth mode in [v0.17.0](https://github.com/jasonacox/pypowerwall/releases/tag/v0.17.0), which matters especially for inverter-only setups without RSA provisioning.

Closing as superseded by the current TEDAPI architecture. Please open new issues for any specific inverter data gaps in current releases.

— Sam ⚡

closed 2026-09-06 by jasonacox-sam