jasonacox/pypowerwall #5: Modification to poll() defaults

wcwong, 2022-02-28:
Two things related to poll()

1. When creating the Powerwall object, it would be good to pass in a cache_expiration parameter that can adjust the cache expiration
2.  It would be good to be able to override the cache with a Fetch=True parameter to poll()

If I make a pull request and make these changes, would you be interested in accepting them?

jasonacox, 2022-02-28:
Hi @wcwong, yes, please submit a PR.  Your 2nd point was on my to-do list and the 1st one is a great addition.  Thanks for the suggestion and help!

wcwong, 2022-02-28:
ok. will do but please don't wait on me. I'm not sure how much time I'm going to have until the weekend and I still need to figure out PRs (the last time I contributed something, it was back in the days of email'ing diffs... :)

jasonacox, 2022-03-01:
Ha!  Thanks, @wcwong, but now I really want you to have the opportunity to submit a PR!  It's so much easier now.  Github will do a lot of the work for you, and honestly, I would love to have your help. :)

If you are willing, give it a try this weekend.

closed 2022-03-05 by jasonacox

jasonacox, 2022-03-05:
Hi @wcwong Thanks for the PR!  Good job.  

I noticed that you are not showing up as a contributor.  This can happen if the email you used to commit the change is not mentioned in your github profile.  You should be able to fix it:

1) Go to https://github.com/settings/emails
2) Add your email to 'Add Email address' 

The email address you used can be seen here: https://github.com/jasonacox/pypowerwall/commit/2454b5ed24e402659931effe3236555568819037.patch

PR #8 (2022-03-05): Updates to __init__ and poll()
 - Closes #5
 - Added pwcacheexpire to __init__ to allow adjustment of the cache time
 - Added timeout to __init__ to allow adjustment of the http timeout
 - added force to pol() to bypass the cache regardless of the timeout
 - moved the check for the vitals to after the cache check (minor optimization)
commit 93592af5 (2022-03-05): Updates to __init__ and poll()
 - Closes #5
 - Added pwcacheexpire to __init__ to allow adjustment of the cache time
 - Added timeout to __init__ to allow adjustment of the http timeout
 - added force to pol() to bypass the cache regardless of the timeout
 - moved the check for the vitals to after the cache check (minor optimization)