jasonacox/pypowerwall #6: detailed battery structure

wcwong, 2022-02-28:
You can get detailed battery information from the 'system_status' api call.   In particular, you get
```
            "Type": "",
            "PackagePartNumber": "3012170-10-B",
            "PackageSerialNumber": "TGxxxxx",
            "disabled_reasons": [],
            "pinv_state": "PINV_GridFollowing",
            "pinv_grid_state": "Grid_Compliant",
            "nominal_energy_remaining": 13803,
            "nominal_full_pack_energy": 13803,
            "p_out": 0,
            "q_out": 0,
            "v_out": 248.60000000000002,
            "f_out": 60.007999999999996,
            "i_out": -0.4,
            "energy_charged": 12780,
            "energy_discharged": 740,
            "off_grid": false,
            "vf_mode": false,
            "wobble_detected": false,
            "charge_power_clamped": false,
            "backup_ready": true,
            "OpSeqState": "Active",
            "version": "b0ec24329c08e4"
```

What do you think of having a `detailed_battery()` call that returns this information plus adds a `temperature `value from the `vitals()` data?

The parameters would follow the the verbose=True to return the JSON.  Otherwise, return a dict structure per battery with the two nominal energy readings and the temperature.

jasonacox, 2022-02-28:
I thought most of this was in `vitals` already, but there are several good gems here!   I would say that for verbose we combine the battery data from `system_status` and `vitals` (TETHC, TEPOD and TEPINV), per battery.  For non-verbose, as you say with perhaps the addition of something like *instant_apparent_power* per battery as well.  It shows up on the TEG installer portal so it must be broken down in some of the api's (but not in vitals unless I'm just missing it).

wcwong, 2022-03-06:
Take a look at the PR I just put in for what I was thinking for this. 

I only did TETHC but created issue #11  for integrating TPOD and TEPINV data.

PR #10 (2022-03-06): Implement system_status() and battery_blocks()
 - Fixes #6
 - implements call to /api/system_status with system_status()
 - implements battery_blocks() by merging data from calls to
   system_status() and vitals()
commit 5f4dc609 (2022-03-06): Implement system_status() and battery_blocks()
 - Fixes #6
 - implements call to /api/system_status with system_status()
 - implements battery_blocks() by merging data from calls to
   system_status() and vitals()
closed 2022-03-06 by jasonacox