jasonacox/pypowerwall #20: Implement HTTP persistent connections for API requests

mcbirse, 2022-07-25:
I've noticed API requests to the Powerwall gateway are not using HTTP persistent connections (aka keep-alive or connection re-use), which results in every request opening a new connection to the gateway and closing again.

There can be a lot of requests being sent to the gateway, even more so if the Power Flow is being viewed via the proxy as with the default Grafana dashboard of the Powerwall-Dashboard (~7 additional requests every few seconds).

I assume the constant open/close of HTTP connections might cause more load on the Powerwall, not to mention the continual requirement for new TCP sockets and such. My Powerwall has been a bit flaky sometimes where it stopped responding for a while, which could be attributed to overloading it perhaps?

So, to potentially reduce the load and make pypowerwall more efficient with API requests to the gateway, I was thinking it would be better to utilise HTTP connection re-use and pooling.

This appears to be simple enough by using a [Session object](https://requests.readthedocs.io/en/latest/user/advanced/) of the Requests module. I will submit a PR with my code changes for this.

Using the Session object results in HTTP connections to the gateway remaining established via keep-alive and being re-used with each API request. There could be multiple connections to the gateway (pooling), but this will be limited by the max pool size (10 by default, which seemed fine in my testing).

I ended up doing a fair amount of before/after testing of this change, as I wanted to quantify and validate that it was in fact beneficial?

I also did resilience testing with Powerwall-Dashboard and the pypowerwall proxy, to ensure there were no adverse affects. i.e. Testing what happens if an established HTTP connection was purposefully broken, invalidating the auth cookie to see if a re-login would occur, disrupting the network connection to the gateway for a while, etc. All my tests passed, with the connection to the gateway re-establishing okay without any user intervention or having to restart the pypowerwall proxy. :thumbsup:

Anyway, I thought I would include findings from my before/after testing for anyone interested...

@jasonacox - let me know what you think? This appears to be a very worthwhile change I believe. I think I just missed the v0.5.1 release as I was typing this out when released! v0.6.0 maybe? 😊 

And sorry for such a long post! But hopefully the below testing might also be informative to others (as it was for me).

---

Firstly, I wanted to see what happens regarding HTTP connections direct to the gateway using a standard web browser. As my Powerwall is in a DMZ on my network, I used my firewall to view the TCP connections.

When browsing direct to the gateway (i.e. not via pypowerwall proxy), only 1 persistent HTTP TCP connection is used. With the browsers dev tools, the network activity showed requests being sent every second or so over this connection.
[image]

By comparison, when viewing the TCP connections to the gateway coming from the Powerwall-Dashboard and pypowerwall, I was seeing constant open/established/close of connections, using different ports.
[image]

After changing pypowerwall to use persistent HTTP connections however, it then behaves like a standard web browser by using keep-alive to keep connections established and re-use them for API requests, with multiple connections up to the pool max. Nice!
[image]

I also used the example test script to check connection use and measure any performance improvements. For the test purposes, I set `pwcacheexpire=0` in the script to disable the pypowerwall caching.

Before the changes, debug output shows each API request creates a new HTTP connection.
[image]

After modifying pypowerwall to use persistent HTTP connections, only 1 connection is established and used for each API request.
[image]

Regarding the performance differences for the above, I measured the elapsed time of sending the set of requests over several iterations, with a 5 second delay between each set.

Before any changes, the time taken for each set of requests to complete was ~1 second.
[image]

When re-using a single HTTP connection however, the response time was significantly reduced, taking around only 0.25 seconds.
[image]

So, there definitely appears to be a benefit here, potentially reducing the load on the Powerwall gateway, while also improving response times!

jasonacox, 2022-07-25:
Hi @mcbirse - Impressive work!  I love this! Good analysis and testing.  Currently pypowerwall caches the calls but you are right, after TTL, it creates a new TCP session each time which is unnecessary.  Please do submit a PR and feel free to take us to 0.6.0.

Based on your research, connection pooling and persistence should be the default. I am wondering if we could present the option to toggle it off or perhaps adjust pool size.  What do you think? 

Again, great work!  This is a great improvement.

mcbirse, 2022-07-25:
> Based on your research, connection pooling and persistence should be the default. I am wondering if we could present the option to toggle it off or perhaps adjust pool size. What do you think?

That's a great idea. It would give the user the option to control this based on their use case. So in the class definition I could add an option for the pool size like below:

```py
def __init__(self, host="", password="", email="nobody@nowhere.com", \
    timezone="America/Los_Angeles", pwcacheexpire=5, timeout=10, poolmaxsize=10):
```

And if the user wanted to disable this, I could add some conditions to the code so if they had set `poolmaxsize=0` it would essentially function as it does now, without persistent connections.

I'll work on adding this functionality before submitting the PR.

Defaulting to using persistent connections makes sense from my testing, and the sweet spot for max pool size certainly seems to be about 10.

If using Powerwall-Dashboard it appears most of the time, about 6-7 connections are established, so it does not reach the default max pool size. This is probably due to e.g. the number of input urls configured in telegraf which is 6. If you use Powerwall-Dashboard and are viewing the Power Flow panel, this increases the connection use while being viewed.

I noticed if you *reduce* the max pool size too much, this will result in similar behaviour to before, where connections are continually opened/closed, as they get discarded when the pool is full:

```ps1
[urllib3.connectionpool] [WARNING] Connection pool is full, discarding connection: 10.255.222.2. Connection pool size: 3
```

This doesn't stop anything from working per se, it just means a new connection is opened again to serve the request.

mcbirse, 2022-07-27:
> Please do submit a PR and feel free to take us to 0.6.0.

Done - hope you like! 😊 

Some notes on these changes:

- Requests to Gateway will now re-use persistent http connections which reduces load and increases response time
- Default connection `poolmaxsize=10` to align with Session object defaults. Note: pool use applies to multi-threaded use of pyPowerwall only, e.g. as with the pyPowerwall Proxy Server
- Added env `PW_POOL_MAXSIZE` to proxy server to allow this to be controlled (persistent connections disabled if set to zero)
- Added env `PW_TIMEOUT` to proxy server to also allow timeout on requests to be adjusted
- For release prep, pyPowerwall version changed to v0.6.0 and proxy version to t17 (release notes still required)

jasonacox, 2022-07-27:
Nicely done @mcbirse !  Beautiful addition with an elegant approach. ❤️  I'm running tests including some soak tests tonight.  If everything looks good, I'll target a release of v0.6.0 to PyPI tomorrow.  Thank you! 🙏

jasonacox, 2022-07-28:
Soak test looked good - no memory leaks.  New version has been pushed to PyPI and Docker Hub for Powerwall-Dashboard.  Thanks again, Michael!

[image]

mcbirse, 2022-07-28:
Awesome! You're welcome Jason glad I could contribute to such as great project! 👍

jasonacox, 2022-08-10:
No issue reported. Closing this, thanks to @mcbirse great work. ❤️

closed 2022-08-10 by jasonacox