jasonacox/pypowerwall #22: Probably not an issue but..

Stephen2615, 2022-08-10:
Firstly, another outstanding bit of coding. 

The documentation says: This python module can be used to monitor and **control** Tesla Energy Powerwalls. 

I imagine this has been discussed many times but there is a train of thought that forcing the backup level to 100% forces the Powerwall to charge no matter what option is chosen and thus override the Tesla "algorithm". I know people who wait until the off peak time (10 PM or perhaps 11 PM for daylight savings) to change their battery backup reserve to 100% and then get up before the peak time (usually 7 AM) to ensure they are not charging during peak hours.

So can this code do that? I can't (probably not looking hard enough) find anything to say set backup reserve to 100%. Then something could poll the battery to ensure it reaches 100% before peak time, set the battery reserve back to say 15%. Importing from the grid seems to be set to 3.5 kWh so a 13.5 kWh battery needs around 4 hours to charge from 10%. 

I generally just leave my battery alone but it would be interesting to see how this could work. A couple of wet cloudy days really screws with charging and more often than not, the "algorithm" doesn't pick the weather and my battery will not charge. 

Regards

Stephen

jasonacox, 2022-08-10:
Hi @Stephen2615 - Thanks for opening this!  

That's a great use case and valid point.  We can make all API calls via `pypowerwall` but some of the "write" functions have been removed in later firmware version. Some thoughts:

* In general, if you are able to make adjustments via the customer login to the Powerwall gateway portal, we can script it in `pypowerwall`.  It may require adding a POST method extension to the library.
* If it requires "installer" level access, it would require a token generated via toggling one of the Powerwalls power switches.  It may not be a seamless  approach (some manual bits) but could be a way to reach the automated outcome we would want.

I need to explore this bit to see if we could get it to work for the battery reserve settings.

Stephen2615, 2022-08-10:
Thanks for the very prompt response. 
One of the forums here in Australia mentioned that it can be done but that is where it stopped. Someone said to look at this link but it doesn't mean anything to me. https://www.teslaapi.io/powerwalls/commands

jasonacox, 2022-08-10:
> https://www.teslaapi.io/powerwalls/commands

Thanks! Yes, I did see that in my investigation as well (see) but it is for accessing the public API, owner-api.teslamotors.com (not your local Tesla Powerwall gateway). It seems non-trivial.

jasonacox, 2022-10-23:
Hi @Stephen2615 - We now have a tool that can help you set the Powerwall backup reserve level for the batteries via command line.  This is based on some great work by @mcbirse (from [here](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history#tesla-history-import-tool)) that streamlines the Tesla Cloud token setup process.  

The tool details are here: https://github.com/jasonacox/pypowerwall/tree/main/tools#set-reserve and the script is here: [set-reserve.py](https://github.com/jasonacox/pypowerwall/blob/main/tools/set-reserve.py).

```bash
# Setup / Login to Tesla Cloud and get token
pip install python-dateutil teslapy
python3 set-reserve.py --login
```

Example

```
$ python3 set-reserve.py --read
READ: Current Battery Reserve Setting: 20% for 2 Powerwalls

$ python3 set-reserve.py --set 25
SET: Current Battery Reserve Setting: 25% - Response: Updated

$ python3 set-reserve.py --read
READ: Current Battery Reserve Setting: 25% for 2 Powerwalls
```

Stephen2615, 2022-10-24:
@jasonacox and @mcbirse 
What can I say other than this is awesome. We have had endless weeks of cloud and rain and the Tesla algorithm just gets confused and doesn't do what I really want which is to force the battery to use off peak. Thank you so much. I am indebted to your brilliance.

willplaice, 2022-10-24:
amazing news - thanks Jason, I am having an issue with creating the config file -   python3 set-reserve,py --login
python3: can't open file '/home/parallels/Powerwall-Dashboard/set-reserve,py': [Errno 2] No such file or directory

I tried creating a blank file - same error ?

willplaice, 2022-10-24:
amazing-  instructions that you're expecting the page not found could be clearer, but this is great I can make it do WHAT I WANT !  thanks @jasonacox and @mcbirse

jasonacox, 2022-10-25:
Thanks @willplaice !!  Appreciate the feedback.

> the page not found could be clearer

Thanks for this - Any suggestion on how we could make it clearer?    Maybe:

```
NOTE: After you log in, it will take you to a 404 Page Not Found error page - do not panic, 
this is what you want.
```

As for a scheduler, I suspect there are several open source options.  I'm familiar with Rundeck but don't know if that is what you are looking for.  If you find something, let us know!

willplaice, 2022-10-25:
I think that NOTE would make it more obvious

I went with cockpit to expose the scheduler over http.... not great but working.

philip98642, 2023-01-04:
Thank you @jasonacox for the set reserve tool, I found it in October and have been using it every night to force my powerwall to fully charge. I have  2 additional scenarios but my coding skills aren't up to making the modifications so I thought I should write them down in case anyone else has similar ideas:

1) My energy supplier can decide when to charge my car so on a sunny day, when the solar is generating, it would be useful to have a command like set-reserve.py --set current to set the reserve to the current battery level to simply stop any charge / discharge during the time my car is charging (and my electricity tarriff falls to off-peak rate). I guess I could also script this if there was an easy way to get the current battery level out of the script. 

2) I have noticed that my powerwall will charge at 5kW in "Time Based-control" and at 1.7kW in "Self-powered" mode. I get 6 hours off peak energy per night so I can charge approx 60% at at 1.7kW and I would prefer not to wear out the battery every night by charging at 5kW when I only really need an hour or so at that rate. Could a --set-mode (Self-Powered or Time Based-control) option be added which could then be used in a script to swap the mode mid way through the night? (also I have noticed that the Time Based-control takes approx 15 mins to stop importing after the reserve is dropped to 20%, even if the charge finished hours previously, whereas the self-powered one acts instantly thus preventing any further grid import.  

Thanks for reading these ideas, Best Regards

Philip

jasonacox, 2023-01-05:
Hi @philip98642 - Great suggestions!

1. Wish granted :) - I made the change.  Pull down the latest and you can use `--set current` or `--current`. It will set the reserve to the current battery level.  Let me know if you see any issues.
2. I believe this is possible but we should create a separate script instead of overloading `set_reserve`.  Maybe we should just create a `set_powerwall` script and add all the options you would find in the app?  Anyway, I'll add a TODO for this unless someone else has time to create the `set_mode` or the more uber `set_powerwall` script for us.

jasonacox, 2023-01-05:
Ok, I guess it was easier than I thought.  I added a `set-mode.py` [tool](https://github.com/jasonacox/pypowerwall/tree/main/tools#set-mode).   

```sh
# Show current mode
python3 set-mode.py --read

# Set Self-Powered Mode
python3 set-mode.py --set self

# Set Time-Based Control Mode
python3 set-mode.py --set time
```

See https://github.com/jasonacox/pypowerwall/tree/main/tools#set-mode

Let me know if you see any issues.

Stephen2615, 2023-01-05:
@jasonacox, 

If only more things in life were this easy to fix. Thanks again for your help

Stephen

philip98642, 2023-01-06:
@jasonacox Thank you so much for such a fast reply, I will make time at the weekend to download and test the code in my environment. Cheers Philip

jasonacox, 2023-03-12:
Closing this...

## Summary

Setting the minimum backup reserve or switching between "time based" and "self powered" mode on the Powerwalls currently cannot be set via local (pypowerwall) API calls due to security restrictions in the Firmware (if this changes, please create a new issue or re-open this). 

However, there is a way to programmatically adjust these with the Tesla Owner API (cloud).  

## Set-Reserve and Set-Mode Tools

Please see  https://github.com/jasonacox/pypowerwall/tree/main/tools for Powerwall `set-reserve.py` and `set-mode.py`.  These tools are based on the the amazing [tesla_history.py](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history) tool by Michael Birse @mcbirse that imports Tesla cloud history into the [Powerwall-Dashboard](https://github.com/jasonacox/Powerwall-Dashboard).

closed 2023-03-12 by jasonacox