jasonacox/pypowerwall #24: Powerwall Firmware Upgrade 22.26.1-foxtrot 4d562eaf - Bugs in Proxy

jasonacox, 2022-10-14:
The latest Powerwall Firmware upgrade to 22.26.1-foxtrot 4d562eaf resulted in a few bugs with the pyPowerwall proxy.

* The `/version` API does not respond.  This is due to an exception caused by the unexpected "-foxtrot" suffix.
* The iFrame rendered animation via proxy may cause flashing and results in a "compliance" text link in the graphic that wasn't there before.

[image]

@danisla Do you know if the "compliance" text in the animation graphic could be easily fixed?

danisla, 2022-10-15:
I am not able to test this because I'm stuck on 22.18.3. However, in general, you should be able to hide the `COMPLIANCE` text in the black.js file. Once you know the selector, you can just do something like this: `$(".element-selector").hide()`

jasonacox, 2022-10-16:
You're brilliant @danisla !  Thank you!  Here is the culprit:

```html
<a id="compliance" class="btn btn-link col-xs-6" href="/compliance"><span>COMPLIANCE</span></a>
```

I just added `.btn` to the existing .hide() line in clear.js (and the others):

```js
$('.overview-menu, #logout, .footer, .compact-btn-row, .toast-list, .power-flow-header, .btn').hide();
```

[image]

As a note, I don't know why my system upgraded or how to force it. I do not know that I've never seen an upgrade with an alpha suffix (e.g. -foxtrot) so I wonder if this is how they do roll-out testing. The suffix broke pypowerwall proxy's version function because it was computing a INT to send back in the json payload along with the string (for anyone graphing version upgrades).

closed 2023-03-12 by jasonacox