jasonacox/pypowerwall #27: Is there a description of the fields?

dakky, 2023-02-01:
I'm trying create some dashboards based on the data collected by pypowerwall and my greatest problem is a understanding of the fields and their values

i.E:
* what is `site` and `load`
* what are the 3 power fields and their difference?
  ```
instant_power: 62
instant_reactive_power: -253
instant_apparent_power: 260.48608408128064
``` 
* what are the `energy_imported` and `energy_exported` fields?

Thanks for clarification

Dakky

jasonacox, 2023-02-02:
HI @dakky 

Unfortunately, Tesla hasn't provide any details as to the official descriptions of these.  Here is what we have generally condluded:

* Site = utility grid
* Load = home (think of it as the "load" that the battery or grid powers)
* instant_power = current power (instant) - also called "true power" in wattage (W) 
* instant_reactive_power = The dissipated power resulting from inductive and capacitive loads measured in volt-amperes reactive (VAR)
* instant_apparent_power = The combination of reactive and true power measure in volt-amperes (VA)
* energy_imported  = kWh pulled from grid over a duration of time (since Powerwall commissioning it seems)
* energy_exported = kWh pushed to grid

If you are building a dashboard, you might check out https://github.com/jasonacox/Powerwall-Dashboard#powerwall-dashboard either to use or help you build yours.

dakky, 2023-02-02:
Thanks 4 the feedback @jasonacox 
 It might be a good idea to collect these infos in the documentation? I can imagine that more people struggle with that issue.

Concerning your dashboard: Yes thats my target. But I have a slightly different setup:
* I already have a influxdb2 up and running 
* influx2 unfortunetly/luckily it uses  a new scripting language for quering.

So basically I am rewriting the queries queries in the dashboard for influxv2, which is quite a hassle because I have to read the json and try to understand whats happening there. The rest of the setup is working fine: pypowerwall runs out of the box, telegraf just needs a new output and now the visualization is the real struggle :-D

jasonacox, 2023-02-03:
Thanks for the suggestion!  I add a glossary: https://github.com/jasonacox/pypowerwall/blob/main/README.md#glossary

Is it possible to run a instance of influxdb 1.8 as well as 2.x?  That would save the effort.  Getting the QL syntax updated to Flux and porting the continuous queries to tasks doesn't seem easy.  Of course, if you do, I'm sure others will appreciate your work and want to know how you did it.

dakky, 2023-02-03:
after fiddling around with more frustration than success i added an influxdb1.8 to my setup. Then everything works out of the box

jasonacox, 2023-02-04:
Thanks for the update @dakky  - and great job getting it to work!  

I anticipate there will come day we have to migrate to 2.x.  Unfortunately, the migration path (data and CQs) doesn't seem easy or clean, at least from my research.  Hopefully someone in this community will discover an easy path. :)

closed 2023-03-12 by jasonacox