jasonacox/pypowerwall #28: Please tag new release for 0.6.1

zi0r, 2023-03-10:
I'm maintaining the FreeBSD port for pypowerwall and have noticed that the new Powerwall-Dashboards require the /alerts/pw call to exist.

Would you be able to tag a new release so I can update the port and get this functionality out to the world?

Thanks!

jasonacox, 2023-03-11:
Oh no!! 😲  You are right.  The /alerts/pw updates should have been tagged in with the PyPI v0.6.0 version push.  I don't know how I missed that.  Thanks for opening this! 🙏 I'm going to rev up to 0.6.1.

jasonacox, 2023-03-11:
Released: 
* Tag: https://github.com/jasonacox/pypowerwall/releases/tag/v0.6.1
* PyPI: https://pypi.org/project/pypowerwall/0.6.1/
* DockerHub: https://hub.docker.com/repository/docker/jasonacox/pypowerwall/general

Please test and let me know if you see the latest.

zi0r, 2023-03-11:
Thanks!  Port updated:

https://cgit.freebsd.org/ports/commit/?id=40ee627f1fc4fa44548abdd4cf68095f00f0b1e1

closed 2023-03-11 by zi0r