jasonacox/pypowerwall #31: Proxy cache-control not set correctly

dkerr64, 2023-03-20:
Setting Cache-Control in server.py [here](https://github.com/jasonacox/pypowerwall/blob/4dc18b0ab8c5b58dc1d84ca95e9083fb3249e7aa/proxy/server.py#L292) is not done correctly.  This line of code
```python
self.send_header("Set-Cookie", "Cache-Control: no-cache")
```
is incorrect as cache control is not a cookie, it is it's own header field.  The correct code is:
```python
self.send_header("Cache-Control", "no-cache, no-store")
```
Note that I added `no-store`.   I did this because when analyzing what the Testa Powerwall returns when I connect the browser directly to the Powerwall, I see both no-cache and no-store.  I believe this is to support older browsers.

Okay, so, by default pyPowerwall should set `no-cache, no-store` by fixing this bug.  But not caching has severe performance implications when reloading a page with embedded power flow animation graphics.  We should provide an option that lets a user indicate that they want to permit a browser to cache resources.  I propose a setting.
`PW_PROXY_CACHE=<seconds>`
and if this is set to anything non-zero then
```python
proxy_cache = int(os.getenv("PW_PROXY_CACHE", "0"))
if proxy_cache > 0:
    self.send_header("Cache-Control", "max-age={}".format(proxy_cache))
else:
    self.send_header("Cache-Control", "no-cache, no-store")
```
This dramatically improves page load (after the first load) until whatever number of seconds expires and then the browser will fetch the resources again.  A user can always force the reload by holding down the shift key.

[image]

Notice the resources that are now retrieved from memory.

Now, we could further refine this.  Rather than setting the header before the data is retrieved from Powerwall, wait until after so that the content-type is known.  If content-type is `text/css`, `application/javascript` or `image/png` then use max-age else do not permit caching.  Specifically `application/json` should never be permitted to cache. This is what appears to be happening by default, but I don't know if that is coincidence or not...  It's not really coincidence (as coded right now) because that data is coming from `<ipaddr>:8675/api/xxxx` that are in the `ALLOWLIST` so it is not going through the proxy code.

So, it might be safer to move the header code down a few lines where we already have...
`ftype = r.headers['content-type']`
and then add
```python
proxy_cache = int(os.getenv("PW_PROXY_CACHE", "0"))
# Allow browser caching, if user permits, for CSS, JavaScript and PNG images...
if proxy_cache > 0 and (ftype == 'text/css' or ftype == 'application/javascript' or ftype == 'image/png'):
    self.send_header("Cache-Control", "max-age={}".format(proxy_cache))
else:
    self.send_header("Cache-Control", "no-cache, no-store")
```
And finally, I would move the first line up to where all the other environment variables are retrieved.

David

dkerr64, 2023-03-20:
I notice that `server-r.py` is very similar but does not include the (buggy) attempt to set Cache-Control.  But this proposed fix should probably be applied to that file as well?

thekev, 2023-03-20:
Also having this same problem. It's breaking other applications on the same host until I clear the bad cookie. Looks like this is blamed on 76a2344c.

jasonacox, 2023-03-21:
Brilliant @dkerr64 !!  I love it.  Do you want to submit the Pull Request? I'm happy to make the changes but prefer to make sure you get credit.

As for `server-r.py`, that was a prototype that has fallen behind a bit. It was attempt to add a simple auth cookie restriction to help with public access. It's a bit clunky and frankly not very secure. If it worked, the functionality would have been ported to `server.py`.  Long way to say: don't worry about adding the updates to `server-r.py`. 😉

closed 2023-03-22 by jasonacox

jasonacox, 2023-03-22:
Thanks David! This is great. I updated release to t25 and will test and push to latest on dockerhub soon.

jgleigh, 2023-03-24:
Just tested this and it's working great. Thanks!

dkerr64, 2023-03-25:
@jasonacox is this on docker hub yet?  I need it there to easily update my Powerwall-Dashboard server.

Thanks
David

jasonacox, 2023-03-25:
Yes the update is now available as jasonacox/pypowerwall:latest

dkerr64, 2023-03-25:
> Yes the update is now available as jasonacox/pypowerwall:latest

Thank you.  I just realized that I needed to run the `upgrade.sh` script.  I did that and it now flys.  Loading the power flow graphic is so much faster now that it is cached.

jasonacox, 2023-03-25:
Awesome!  Thanks for your help and contribution @dkerr64 !!

commit 297e5abc (2023-03-22): Merge pull request #32 from dkerr64/main

Fix Cache-Control no-cache header and allow for setting max-age, fixes #31