jasonacox/pypowerwall #34: Powerflow background is transparent, not black.

venturanc, 2023-04-06:
When I access x.x.x.x:8675/ for the powerflow animation, the background of the page is transparent, and I can't see the current battery percentage without highlighting the text on the page. 

Did I miss a configuration step?  

[image]

If I look at the page source and change the CSS background to black, it looks great, but then I think it autorefreshes and goes back to background=transparent.

jasonacox, 2023-04-07:
Hi @venturanc!

How are you running the proxy?  Docker or direct via command line?  There are some envrionmental settings that change the color settings (see https://github.com/jasonacox/pypowerwall/tree/main/proxy/web):
```bash
# default
PW_STYLE=clear

# others
PW_STYLE=black
PW_STYLE=white
PW_STYLE=dakboard
PW_STYLE=grafana
```
Also, try this: http://x.x.x.x:8675/example.html - it should render an iFrame example like this:

[image]

venturanc, 2023-04-07:
Ahh! Thanks! I'm running the docker version. The example html link does work too. Sorry that I missed that. I'll close this issue that was my user error.

BTW, thanks so much for developing this! I'm using this to get Tesla Powerwall info into Smartthings for home automation (switching off devices if grid outage, reminders to do laundry/dishes when the powerwall is full and excess solar production etc.)

I also have a python script that gets the grid power usage every 2 minutes from pypowerwall during solar production time and if my car is charging, it uses Teslapy to adjust the car charging speed to try and maintain minimal to no grid exports to avoid non-bypassable charges, and it stops charging at 4pm when the 4-9 peak period starts to not drain the powerwall. it was pretty easy to modify the TesSense project and get the data from pypowerwall instead of Sense!

jasonacox, 2023-04-07:
Thanks @venturanc !  I would love to see some of your scripts if you are ever willing to share.  Post them here if it is easy or we have a section where cool automation contributions can be added: https://github.com/jasonacox/pypowerwall/tree/main/tools - No pressure. Thanks for the kind words. 🙏

venturanc, 2023-04-07:
Sure @jasonacox - I did a writeup about how to use PyPowerwall with SmartThings [`here: `](https://community.smartthings.com/t/obsolete-tesla-powerwall-manager-smart-app-and-device-handler/170347/68?u=natec)

I need to spend some more time cleaning this up, and perhaps add a web API call to start/stop the solar charging, and maybe tie it all together with pypowerwall with a docker-compose file, and get it published to my own github (after I figure out how to have it read/use the environmental variables to get the tesla login data and input a tesla refresh token, lat/lon and py-powerwall_IP variables - I'll refer to your code on the environmental variables stuff), but here's the quick and dirty functional python script:

```
"""
 Adapted by Nate Carroll 03/2023 from TesSense w/ SenseLink  -Randy Spencer 2023 Version 9.7
 Python charge monitoring utility for those who have a Tesla Powerwall.
 Uses jasonacox/pypowerwall docker container to get realtime stats from the local gateway for Production 
 and Grid Utilization of electricity to control
 your main Tesla's AC charging amps to charge only with excess production.
 Simply plug in your car, update your info below, and run this in python.
 
 Added: checking of location of Tesla to be sure it's charging at home
 Added: tracks cabin temp and local chargers, vents the car if it gets too hot 
 - removed due to Tesla removal of Vent option from API
"""

username = 'elon@tesla.com' # Fill in Tesla login email address/account
lat, lon  = ##.###, -###.###        # Fill in Location where charging will occur (shown at startup)
pypowerwall_IP = '10.x.x.x:8675'	# Fill in IP address and port for pypowerwall docker container
    
RedTxt, BluTxt, NormTxt = '\033[31m', '\033[34m', '\033[m'
RedBG, GrnBG, NormBG = '\033[101m', '\033[102m', '\033[0m'

import datetime, asyncio
import logging, sys#, json
# pip3 install teslapy
import teslapy
#/c Set stdout as logging handler
root_log = logging.getLogger()
root_log.setLevel(logging.WARNING) # WARNING or INFO or DEBUG
handler = logging.StreamHandler(sys.stdout)

def printerror(error, err) :                               # Error message with truncated data
    print(str(err).split("}")[0], "}\n", datetime.datetime.now().strftime( "%a %I:%M %p" ), error)

def printmsg(msg) :                                        # Timestamped message
    print( " ", datetime.datetime.now().strftime( "%a %I:%M %p" ), msg )
    
def PrintUpdate(chargedata, fast) :                        # Display stats at every % change
    print( "\nLevel:",
        chargedata['battery_level'], "%, Limit",
        chargedata['charge_limit_soc'], "%,",
        chargedata['charge_rate'], "MPH",
        chargedata['charger_voltage'], "Volts",
        chargedata['charge_energy_added'], "kWh added,")
    if fast : print("Rate:",
        chargedata['charger_power'], "KWs",
        chargedata['conn_charge_cable'],
        chargedata['fast_charger_type'],
        chargedata['minutes_to_full_charge'], "Minutes remaining\n" )
    else : print(chargedata['charger_actual_current'], "of a possible",
        chargedata['charge_current_request_max'], "Amps,",
        chargedata['time_to_full_charge'], "Hours remaining\n" )

def PrintTemp(car) :
    #Tesla removed vent API in USA due to NTHSA ~01/2023 - causes API error
	#if car.get_vehicle_data()['climate_state']['inside_temp'] > 40 : # 104°F
    #    if not car.get_vehicle_data()['vehicle_state']['fd_window'] : # Not Open
    #        Vent(car, 'vent')
    #else :
    #    if car.get_vehicle_data()['vehicle_state']['fd_window'] :    # Open
    #        Vent(car, 'close')
    print(car.temp_units(car.get_vehicle_data()['climate_state']['inside_temp']), end='')
    if car.get_vehicle_data()['climate_state']['fan_status'] : print(car.get_vehicle_data()['climate_state']['fan_status'], end='')
    if car.get_vehicle_data()['climate_state']['cabin_overheat_protection_actively_cooling'] : print(car.get_vehicle_data()['climate_state']['cabin_overheat_protection_actively_cooling'], end='')

        
def SendCmd(car, cmd, err) :                               # Start or Stop charging
[... 181 lines cut ...]
                    newrate = min(rate + int(power_diff/volts), chargedata['charge_current_request_max'])
                                                           
                    print( "Charging at", rate, "amps, with", power_diff, "watts surplus" )

                    if newrate < minrate :                 # Stop charging as there's no free power
                        StopCharging(vehicles[0])
                        newrate = 0
                    elif newrate > rate :                  # Charge faster with any surplus
                        SetCharging(vehicles[0], newrate, "Increasing")
                    elif newrate < rate :                  # Charge slower due to less availablity
                        SetCharging(vehicles[0], newrate, "Slowing")

            if lastime != vehicles[0].get_vehicle_data()['climate_state']['timestamp'] :
                lastime = vehicles[0].get_vehicle_data()['climate_state']['timestamp']
                PrintTemp(vehicles[0])                     # Display cabin temp and fan use
            printmsg(" Wait two minutes...")               # Message after every complete loop
            await asyncio.sleep(120)                       # Could use variable to change frequency of updates, but 2 minutes seems reasonable without hitting Tesla API frequently enough to cause lockout

#run the main program
try:
    await TesSense()
except KeyboardInterrupt:
    print("\n\n Interrupt received, stopping TesSense\n")

```

jasonacox, 2023-04-08:
This is awesome! Thanks @venturanc 🙏  

Do you mind if I add this to our /tools section (also happy to add a link to your github location once you have it)?  Or you could submit a PR youself to get listed as a pypowerwall contributor. 😉

venturanc, 2023-04-10:
I'd be honored to submit a PR to /tools and contribute! Give me just a few days to sort everything out and hopefully will get it in soon.

closed 2023-04-10 by venturanc

jasonacox, 2023-04-13:
Thanks @venturanc !