jasonacox/pypowerwall #44: PVAC_a030_fan_faulted

DasHutch, 2023-07-26:
I believe this alert to be when an inverter fan is failing / failed.
I’m getting this on my system and the corresponding graphs for inverter are showing “clipping” (or drops in power)

alerts
[image]

graphs
[image]
[image]

jasonacox, 2023-07-27:
Wow! Yes.  These two may be related:

[image]

It looks like the first one is cut off, do you have the full alert names?  Was it this:

* PVAC_a019_ambient_overtemperature
* PVAC_a030_fan_faulted

DasHutch, 2023-07-27:
`PVAC_a019_ambient_overtemperature` is the full alert/error

screenshot showing full details of current alerts
[image]

jasonacox-sam, 2026-09-06:
This reported `PVAC_a030_fan_faulted` and `PVAC_a019_ambient_overtemperature` alerts and asked that they be documented. Jason folded these into the pypowerwall Devices and Alerts documentation as part of the follow-up.

Closing as resolved. If new, undocumented alert codes show up in a current release, please open a fresh issue with the alert name and we'll add them.

— Sam ⚡

closed 2026-09-06 by jasonacox-sam