jasonacox/pypowerwall #45: [Feature Request] Proxy - shortcuts to endpoints

dcod3d, 2023-07-31:
Currently, the root path on the proxy only shows the animation. It would be nice to have links to all the proxy and api endpoints as a hyperlink on that page too.

jasonacox, 2023-07-31:
Good feature request.  Help me flesh out a bit of the details you are proposing:

Navigate to http://localhost:8675/ and see an html rendered page that hosts information and links about the pypowerwall connection:

* Link to animation (perhaps even http://localhost:8675/example.html)
* Links to APIs and link to more descriptions (see current http://localhost:8675/help)
* Info about health or stats (see http://localhost:8675/stats)
* Info about Powerwall version (see http://localhost:8675/version)
* others?

dcod3d, 2023-08-01:
I think just having a simple list of links (listed [here](https://github.com/jasonacox/pypowerwall/blob/main/proxy/HELP.md)) is fine, probably without the iframe (unless there's a way to allow 3rd party cookies on a single site on iOS). Bonus would be having the JSON result beautified (formatted/colored).

jasonacox-sam, 2026-04-14:
## Closing — No Activity in 2+ Years

**Issue summary:** Request to add navigation hyperlinks on the proxy root page to easily access all proxy/API endpoints.

This issue has had no activity since 2023-08-01. To keep the tracker focused on actionable work, I'm closing it now.

**Still seeing this?** Please reopen and share:
- Current pypowerwall version (`pip show pypowerwall`)
- Your gateway type (PW1/PW2/PW3, local/TEDAPI/cloud mode)
- What you tried and what happened

We're happy to help, and any updates that may benefit others in the community are always welcome.

---
*— Sam 🌟 (pypowerwall agent)*