jasonacox/pypowerwall #51: New Tesla Pros software update does not work with pypowerwall

gregshep, 2023-10-13:
Tesla's firmware 23.28.2 no longer supports the old way of getting metrics as far as I can see.  Going to my Powerwall's IP address I get "Switch to Tesla Pros for a better commissioning experience.  The current experience is no longer supported"

The pypowerwall app says it connects but there are not any metrics:
10/13/2023 05:13:32 PM [proxy] [INFO] pyPowerwall [0.6.2] Proxy Server [t27] - HTTP Port 8675
10/13/2023 05:13:32 PM [proxy] [INFO] pyPowerwall Proxy Started

Will pypowerwall support the newer Tesla Pros application?

jasonacox, 2023-10-14:
Hi @gregshep  -

I'm on 23.28.2 with no problems.  This was also reported here: https://github.com/jasonacox/Powerwall-Dashboard/discussions/367

* You can bypass the Tesla Pro screen - hit Continue, you end up at the login screen as usual.
* The proxy logs you show are not showing an error (it should be flooding you with connection errors if it couldn't get to the Powerwall) but you can restart it `docker pypowerwall restart`. 
* You can can to make sure it didn't land on a different IP address: `python3 -m pypowerwall scan`
* Check to see if the proxy is giving a response:  http://localhost:8675/version and http://localhost:8675/vitals (change localhost to the address of the machine running the proxy/Dashboard).

From what I can tell, everything is the same, just that Tesla created an Android app for installers to use instead of having to use the web page setup.  I do hope they don't remove the web based version.

gregshep, 2023-10-14:
You're right, pypowerwall is working.  I have grafana permissions issues and that Tesla Pros screen made me think it was that.  Thank you!

closed 2023-10-14 by gregshep

ymilord, 2023-10-18:
> You're right, pypowerwall is working. I have grafana permissions issues and that Tesla Pros screen made me think it was that. Thank you!

@gregshep How did you correct the grafana permission issue? I am having the same issue as you. Thanks

gregshep, 2023-10-18:
> > You're right, pypowerwall is working. I have grafana permissions issues and that Tesla Pros screen made me think it was that. Thank you!
> 
> @gregshep How did you correct the grafana permission issue? I am having the same issue as you. Thanks

My mistake was using native docker-compose commands (e.g. docker-compose pull; docker-compose restart, etc).  This seemed to have messed up the grafana configuration.  Instead I used the provided commands:
upgrade.sh
verify.sh
compose-dash.sh up -d
and this fixed my configuration.  I can't recall the exact combination but lesson is don't use the native docker-compose to upgrade containers..

ymilord, 2023-10-18:
Unfortunately, that won't work for my specific install.

I already have a global Grafana install that has a bunch of different dashboards configured. I wanted to add the Powerwall-Dashboard to it. Everything works except the power flow. I get the same error you described above. If I goto the pypowerwall example page, it displays normally. When I goto my grafana instance with the powerwall-dashboard added- I get the tesla pro splash page. I am assuming the credentials for the gateway are stored somewhere thats not configured/accessible by my separate grafana instance can't get to/or read. Just trying to figure that out to get that working.

jasonacox, 2023-10-21:
Hi @ymilord - If you are getting the "Tesla Pros" screen instead of the animation, it means you still need to log in to the Tesla Powerwall gateway directly and dismiss that popup.  There are some screenshots in this discussion that might help: https://github.com/jasonacox/Powerwall-Dashboard/discussions/367#discussioncomment-7278760 - Also note, some found that they needed to clear their browser cache.

As to your setup, that makes sense, but one caveat, after spending a few days test the animation to work on newer Grafana instances, I have come to the conclusion that the last stable Grafana version is v9.1.2 (technically it will work up to Grafana v9.2.20 but I ran out of time to do full soak testing). I dropped some of my analysis in https://github.com/jasonacox/Powerwall-Dashboard/pull/374.  This may not apply to you, but wanted to raise it in case you are seeing some weird behavior with the animation even after it is working.

ymilord, 2023-10-23:
@jasonacox Thanks for the reply.

I wanted to try and troubleshoot for a couple days before I replied.

I've tried the login trick, but the issue is I can click 'Continue' but it ignores 'Upgrade Later' Click. It just loops back to the splash page with the 'Continue' button. Endlessly.

 I went in and edited the flow panel to something huge thinking the bottom half it being covered by the Firmware info. No dice. I then completely removed the Firmware version display at the bottom and does the same thing. I can click the initial 'Continue' button, But Upgrade later loops back to the beginning splash page with the Continue.

If I goto http://<IP/Hostname>:8675/example.html on any machine on my network it displays normally. I've even tried injecting that example page into the flow panel. Same deal I get the expected 'Tesla Powerwall Power Flows iFrame Example' But then I also get the Tesla Pro splash page and once again it just loops back to the beginning when Continue/Upgrade Later is clicked. Once again Endlessly.

What I don't get is how can I get to that example page from any machine from various OSs and it displays normally, but when I use it for Grafana it craps itself.

jasonacox, 2023-10-28:
That is very odd! I agree. In my testing, if it was misbehaving in Grafana, it would misbehave in the example. The only thing that I can think of is that the browser has cached some bit of code or cookie that is breaking it for one. Try clearing your cache in your bowser and try both again.  You may need to dismiss the Tesla Pros upgrade screens too (see https://github.com/jasonacox/Powerwall-Dashboard/discussions/377#discussioncomment-7409065).