jasonacox/pypowerwall #54: set-reserve ERROR gives Missing access token parameter. Auth token expired?

ma773o, 2023-11-16:
Hi,

`python3 set-reserve.py --read`
fails with this error:

`ERROR: Failed to retrieve PRODUCT_LIST - (missing_token) Missing access token parameter.`

I guess the problem is that the auth token has expired as I have the following in the set-reserve.auth file:
`... "expires_in": 28800, "token_type": "Bearer", "expires_at": 1700028409.1826518}}}`
which can be converted as expires_in 8 hours and expires_at Wednesday, November 15, 2023 6:06:49 AM GMT .

Is there a way to automatically refresh the token without having to login again?
Can it be embedded in set-reserve.py and the other tools?

Thanks

mcbirse, 2023-11-16:
Hi @ma773o - my understanding is that the TeslaPy module should actually automatically update the auth token file... I have used some very old saved auth token files before (however, with the tesla-history script from the Powerwall-Dashboard project) and never had an issue.

Have you checked the permissions of the `set-reserve.auth` file to ensure it is writable by the script perhaps?

ma773o, 2023-11-16:
Hi, 
permissions are fine...
I've now tried to remove .auth and .conf file and re-do the login process to obtain a fresh token and it actually fails. I called it with `-d` to get debug info and I got the following (I've hidden personal data):

```
python3 set-reserve.py -d --login

Config file 'set-reserve.conf' not found

Do you want to create the config now? [Y/n] Y

Tesla Account Setup
-------------------
Email address: hidden@hidden.com
Save auth token to: [set-reserve.auth] 

Config saved to 'set-reserve.conf'

----------------------------------------
Tesla account: hidden@hidden.com
----------------------------------------
Open the below address in your browser to login.

https://auth.tesla.com/oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=HIDDEN&code_challenge=HIDDEN&code_challenge_method=S256&login_hint=hidden%40hidden.com

After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login: https://auth.tesla.com/void/callback?code=HIDDEN&state=HIDDEN&issuer=https%3A%2F%2Fauth.tesla.com%2Foauth2%2Fv3
ERROR: Login failure - (missing_token) Missing access token parameter.
```

This was working a couple of days ago, and of course my the email/password pair works when authenticating to tesla.com

jasonacox, 2023-11-18:
Hi @ma773o - I'm not sure what could be going wrong here. You seem to be doing the right thing (I assume you removed replaced the real code with `code=HIDDEN`). 

Some (likely not very helpful) thoughts:

* Do you have more than one account with Tesla? Is this the right one?
* Try to clear your browser cache / cookies and try again
* Make sure you have the latest [set-reserve.py](https://raw.githubusercontent.com/jasonacox/pypowerwall/main/tools/set-reserve.py)

jasonacox-sam, 2026-08-31:
This reported `set-reserve.py --read` failing with `missing_token` after the auth token expired, with re-login also failing. Suggestions were exchanged — checking permissions on the auth file, verifying the right Tesla account, clearing cache — but no resolution was ever posted.

Closing due to 90+ days of inactivity. The token-expiry pain may still be real — if you or anyone hits this again, re-open with the debug output and we'll dig back in.

— Sam ⚡

closed 2026-08-31 by jasonacox-sam