jasonacox/pypowerwall #56: Ran upgrade.sh but it does not pull all new files

VenomrushCT, 2023-12-05:
Ran the upgrade.sh script to update to latest version, expected tessolarcharge.py script to be available under 'tools' folder but could not find this script after the update.

These are what can be found under tools after running the update:

-rw-r--r--  1 pi pi 3532 Oct  4 20:36 DOCKER.md
drwxr-xr-x  2 pi pi 4096 May  8  2023 ecowitt-weather-history
drwxr-xr-x  2 pi pi 4096 Mar 28  2023 fixmonthtags
drwxr-xr-x  5 pi pi 4096 May  8  2023 influxdb2
drwxr-xr-x  2 pi pi 4096 Mar 28  2023 mysql
drwxr-xr-x  2 pi pi 4096 Mar 28  2023 pvoutput
drwxr-xr-x  2 pi pi 4096 Jul  3 17:11 pwstatus
-rw-r--r--  1 pi pi 4540 Dec  4 23:12 README.md
drwxr-xr-x  2 pi pi 4096 Dec  4 23:12 solar-only
drwxr-xr-x  2 pi pi 4096 Dec  4 23:12 tesla-history
drwxr-xr-x  2 pi pi 4096 Oct 16 22:09 usage-service
drwxr-xr-x  2 pi pi 4096 Jul  3 17:11 weather-history

VenomrushCT, 2023-12-05:
Output from running upgrade.sh

`Upgrade Powerwall-Dashboard from 2.10.0 to 3.0.2
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Upgrade - Proceed? [y/N] Y

Resetting Timezone to Default...

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...

No local changes to save
remote: Enumerating objects: 170, done.
remote: Counting objects: 100% (170/170), done.
remote: Compressing objects: 100% (92/92), done.
remote: Total 170 (delta 103), reused 135 (delta 72), pack-reused 0
Receiving objects: 100% (170/170), 76.25 KiB | 2.93 MiB/s, done.
Resolving deltas: 100% (103/103), completed with 18 local objects.
From https://github.com/jasonacox/Powerwall-Dashboard
   9affc53..2644e62  main       -> origin/main
 * [new branch]      v3.0.3     -> origin/v3.0.3
 * [new tag]         v3.0.2     -> v3.0.2
 * [new tag]         v3.0.1     -> v3.0.1
Updating 9affc53..2644e62
Fast-forward
 .gitignore                                         |    7 +
 README.md                                          |   61 +-
 RELEASE.md                                         |   19 +
 VERSION                                            |    2 +-
 WINDOWS.md                                         |  201 +++
 compose-dash.sh                                    |   54 +-
 compose.env.sample                                 |   11 +-
 dashboards/README.md                               |   16 +-
 .../dashboard-solar-only.json                      |  574 +++----
 dashboards/dashboard.json                          |    4 +-
 .../powerwall.yml => powerwall-v1-solar.yml        |    8 +-
 powerwall-v1.yml                                   |   90 ++
 powerwall.yml                                      |   26 +
 setup.sh                                           |  286 +++-
 tools/README.md                                    |   36 +-
 tools/solar-only/README.md                         |  130 +-
 tools/solar-only/compose-dash.sh                   |   69 -
 tools/solar-only/compose.env.sample                |   27 -
 tools/solar-only/grafana.env.sample                |   34 -
 tools/solar-only/grafana/.keep                     |    1 -
 tools/solar-only/influxdb.conf                     |  154 --
 tools/solar-only/influxdb/dropcq.sql               |   25 -
 tools/solar-only/influxdb/influxdb.sql             |   42 -
 tools/solar-only/influxdb/run-once-2.8.0.sql       |   15 -
 tools/solar-only/influxdb/run-once-2.8.3.sql       |   13 -
 tools/solar-only/influxdb/run-once-2.9.2.sql       |   18 -
 tools/solar-only/setup.sh                          |  178 ---
 tools/solar-only/tesla-history/README.md           |  286 ----
 tools/solar-only/tesla-history/tesla-history.py    | 1560 --------------------
 tools/solar-only/tz                                |    1 -
 tools/solar-only/tz.sh                             |   31 -
 tools/solar-only/weather.sh                        |  149 --
 tools/solar-only/weather/weather411.conf.sample    |   34 -
 tools/{solar-only => }/tesla-history/Dockerfile    |    0
 tools/tesla-history/tesla-history.py               |  793 +++++++---
 tools/{solar-only => }/tesla-history/upload.sh     |    0
 tz.sh                                              |   18 +-
 upgrade.sh                                         |  243 ++-
 verify.sh                                          |  355 +++--
 weather.sh                                         |   86 +-
 40 files changed, 2111 insertions(+), 3546 deletions(-)
 create mode 100644 WINDOWS.md
 rename tools/solar-only/dashboard.json => dashboards/dashboard-solar-only.json (85%)
[... 47 lines cut ...]
telegraf

Deleting old weather411...
weather411
weather411

Restarting Powerwall-Dashboard stack...
Running Docker Compose...
[+] Running 5/5
 ✔ Container pypowerwall  Started                                                                                                                                                                      0.4s 
 ✔ Container influxdb     Running                                                                                                                                                                      0.0s 
 ✔ Container weather411   Started                                                                                                                                                                      0.4s 
 ✔ Container grafana      Running                                                                                                                                                                      0.0s 
 ✔ Container telegraf     Started                                                                                                                                                                      0.2s 

---------------[ Update Dashboard ]---------------
Open Grafana at http://localhost:9000/

From 'Dashboard/Browse', select 'New/Import', and
upload 'dashboard.json' located in the folder
/home/pi/Powerwall-Dashboard/dashboards/

Please note, you may need to select data sources
for 'InfluxDB' and 'Sun and Moon' via the
dropdowns and use 'Import (Overwrite)' button.`

mcbirse, 2023-12-05:
Hi @VenomrushCT - your upgrade output looks good, I don't see any issues?

Please note, the [tessolarcharge.py](https://github.com/jasonacox/pypowerwall/blob/main/tools/tessolarcharge.py) script is part of pypowerwall [tools](https://github.com/jasonacox/pypowerwall/tree/main/tools), not Powerwall-Dashboard.