jasonacox/pypowerwall #69: Missing ? energy_sites/grid_import_export

avecezar, 2024-01-27:
Is it possible to add the following to READ and SET 

https://owner-api.teslamotors.com/api/1/energy_sites/{}/grid_import_export
Payload options:
"customer_preferred_export_rule": "battery_ok"
"customer_preferred_export_rule": "pv_only"
"disallow_charge_from_grid_with_solar_installed": false
"disallow_charge_from_grid_with_solar_installed": true

jasonacox, 2024-01-28:
Hi, @avecezar , I assume you are talking about the [tools](https://github.com/jasonacox/pypowerwall/tree/main/tools) - Yes, these could be added. I often thought it would be good to have a single command line tool to do all these updates or polls. I'll flag that as a TODO and add these.  

If anyone wants to take this on and combine/expand any of the `set-mode` or `set-reserve` tools, please feel free and submit a PR. I won't have time to work on it for a while.

avecezar, 2024-01-29:
Dear Jason, 

The “tools” concept and execution worked for me 100% (I am new to python and linux). 
What I would like to achieve is to stop charging at 70% if I set it this way using cron . 
So battery is charging from 23:30 cron is checking it every 10min and when the battery reached 70% it will disable "grid charging" making space for solar. At 6AM it would enable “grid charging” back again

Do you know any other way to do it ?

Cezar

jasonacox, 2024-01-30:
Yes, that would be a new tool (to turn on/off grid charging). I suspect much of the code is already in one of the existing tools, but someone needs to investigate how to poll and change the "grid charing" setting.