jasonacox/pypowerwall #79: Set Reserve via Proxy API

spoonwzd, 2024-04-12:
Am I blind, or is there no way to set the reserve / change modes using the Proxy?

jasonacox, 2024-04-13:
You are correct, there is no way to set it via the proxy. The Powerwall gateway API no longer allows local control of the Powerwall. To control the Powerwall, you must use the cloud.  

The good news is that you can use the pypowerwall module to set up cloud auth and control via the command line:

```bash
# Setup Connection with Tesla Cloud
python -m pypowerwall setup

# Get Power Levels, Operation Mode, and Battery Reserve Setting
#
# Usage: PyPowerwall get [-h] [-format FORMAT]
#  -h, --help      show this help message and exit
#  -format FORMAT  Output format: text, json, csv
#
python -m pypowerwall get
python -m pypowerwall get -format json
python -m pypowerwall get -format csv

# Set Operation Mode and Battery Reserve Setting
#
# Usage: PyPowerwall set [-h] [-mode MODE] [-reserve RESERVE] [-current]
#  -h, --help        show this help message and exit
#  -mode MODE        Powerwall Mode: self_consumption, backup, or autonomous
#  -reserve RESERVE  Set Battery Reserve Level [Default=20]
#  -current          Set Battery Reserve Level to Current Charge
#
python -m pypowerwall set -mode self_consumption
python -m pypowerwall set -reserve 30
python -m pypowerwall set -current
```

spoonwzd, 2024-04-13:
Understood, and I am able to set reserve etc. using the command line when using cloud auth.

However, it would be massively beneficial to be able to control this from the proxy when using cloud auth. Could this be added?

jasonacox, 2024-04-13:
Yes it could.  I confess, it scares me a bit. I have hesitated adding that since the proxy access point doesn't require authentication. Any malicious actor who happens to gain access to your network could thrash settings on your system, potentially creating damage. I suspect there are some safeguards in the Tesla logic for this but it is an exposure that requires the user to understand and treat with care.

I suppose we could make that write (control) access "default off" that requires specific settings on startup to allow, or better, requires some secret token to execute (e.g. setting `PW_CONTROL_SECRET=****` environmental variable required as a GET/POST payload to perform those changes). I'm open to thoughts around around that and welcome any contributions to start adding. 😉

spoonwzd, 2024-04-13:
Yes, I think you're on the right track there. Any 'set' command requires PW_CONTROL_SECRET to be passed. Anyone concerned about PW_CONTROL_SECRET being passed in plain text can enable HTTPS. At least that way you've done your due diligence.

jasonacox, 2024-04-14:
Ok, I have something for you to try. If you are using Powerwall-Dashboard:

## To Test

1. Update the pypowerwall image setting in `powerwall.yml` to:
```
jasonacox/pypowerwall:0.8.3t54-beta
```
2. Add `PW_CONTROL_SECRET={something}` to `pypowerwall.env`
3. Restart `./compose-dash.sh up -d`

## Usage

Use a browser to try these URLs or use `curl` commands: 

```bash
# Set Mode
curl http://localhost:8675/control/mode?token=$PW_CONTROL_SECRET&value=self_consumption

# Set Reserve
curl http://localhost:8675/control/reserve?token=$PW_CONTROL_SECRET&value=20

# Omit Value to Read Settings
curl http://localhost:8675/control/mode?token=$PW_CONTROL_SECRET
curl http://localhost:8675/control/reserve?token=$PW_CONTROL_SECRET
```

jasonacox, 2024-04-14:
EDIT: I added info logging for any control command executed

```
jasonacox/pypowerwall:0.8.3t54-beta2
```

spoonwzd, 2024-04-14:
Running curl from the command line I get a "{"reserve": 5.0}'value' is not recognized as an internal or external command, operable program or batch file."

...but the same URL string works fine in a browser with the output:

{"set_backup_reserve_percent": {"backup_reserve_percent": 20, "din": "1099752-01-B--T17K000XXXX", "result": "Updated"}, "set_operation": {"real_mode": "self_consumption", "din": "1099752-01-B--T17K000XXXX", "result": "Updated"}}

Seems to work great! Nice one :)

jasonacox, 2024-04-14:
thanks @spoonwzd !

> Running curl from the command line I get a "{"reserve": 5.0}'value' is not recognized as an internal or external command, operable program or batch file."

Ah yes!  What is happening is that the URL needs to be quoted (I omitted that), so anything before `&` the shell is trying to run as a command in the background and omitting the rest of the line. I'll edit my notes above.

```bash
curl "http://localhost:8675/control/reserve?token=$PW_CONTROL_SECRET&value=20"
```

PR #83 (2024-04-14): V0.8.3 - Control APIs
* Added additional error handling logic to clean up exceptions #81
* Proxy: Fix `/pod` API to add `time_remaining_hours` and `backup_reserve_percent` for cloud mode.
* Added control APIs to set backup reserve and operating mode settings. Requires setting `PW_CONTROL_SECRET`. Use with caution! Closes #79

```bash
# Set Mode
curl http://localhost:8675/control/mode?token=$PW_CONTROL_SECRET&value=self_consumption

# Set Reserve
curl http://localhost:8675/control/reserve?token=$PW_CONTROL_SECRET&value=20

# Omit Value to Read Settings
curl http://localhost:8675/control/mode?to [... cut]
closed 2024-04-14 by jasonacox