jasonacox/pypowerwall #81: pyPowerwall [0.8.2] Proxy Server [t53] testing

jgleigh, 2024-04-13:
These aren't unique to 0.8.2, but thought you'd like to see them regardless with all the new error handling.

`2024-04-12 22:01:05 04/12/2024 10:01:05 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:01:05 04/12/2024 10:01:05 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:01:05 04/12/2024 10:01:05 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:01:06 04/12/2024 10:01:06 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:01:08 04/12/2024 10:01:08 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:01:08 04/12/2024 10:01:08 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:31:10 04/12/2024 10:31:10 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:31:10 04/12/2024 10:31:10 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe
2024-04-12 22:31:11 ----------------------------------------
2024-04-12 22:31:11 Exception occurred during processing of request from ('172.18.0.6', 42768)
2024-04-12 22:31:11 Traceback (most recent call last):
2024-04-12 22:31:11   File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
2024-04-12 22:31:11     self.finish_request(request, client_address)
2024-04-12 22:31:11   File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
2024-04-12 22:31:11     self.RequestHandlerClass(request, client_address, self)
2024-04-12 22:31:11   File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
2024-04-12 22:31:11     self.handle()
2024-04-12 22:31:11   File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
2024-04-12 22:31:11     self.handle_one_request()
2024-04-12 22:31:11   File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
2024-04-12 22:31:11     method()
2024-04-12 22:31:11   File "/app/server.py", line 321, in do_GET
2024-04-12 22:31:11     fcv["grid_status"] = pw.grid_status(type="numeric")
2024-04-12 22:31:11   File "/app/pypowerwall/__init__.py", line 616, in grid_status
2024-04-12 22:31:11     grid_status = payload['grid_status']
2024-04-12 22:31:11 TypeError: 'NoneType' object is not subscriptable
2024-04-12 22:31:11 ----------------------------------------
`

jasonacox, 2024-04-13:
Thanks @jgleigh  - I do appreciate these. For some reason, I don't get these but suspect it is due to having my Powerwall hardwired into my network. This is likely caused by something like a network communication error that results in a malformed payload that is missing that key. I'm going to start adding dictionary getters to these elements to help reduce the noise in the logs. However, I will have it continue to log a notice otherwise you won't know that these calls are failing, a possible indication of a weak network link.

jgleigh, 2024-04-13:
Mine is hardwired too. These errors don't seem to cause any issues though which is good.

jgleigh, 2024-04-13:
Are these coming from an external source? I noticed the IP address isn't inside my network.

Exception occurred during processing of request from ('172.18.0.6', 42768)
Exception occurred during processing of request from ('172.18.0.6', 59624)
Exception occurred during processing of request from ('172.18.0.6', 36076)
Exception occurred during processing of request from ('172.18.0.6', 40034)

jgleigh, 2024-04-13:
Might be something inside the Docker instance, since 172.18.x.x is a private IP address. Very strange.

jasonacox, 2024-04-13:
Since you are using the [Powerwall-Dashbaord](https://github.com/jasonacox/Powerwall-Dashboard) setup, that is telegraf.  The Dashboard setup uses docker-compose which create a private network inside docker to manage the communication between the containers. The setup is defined in the compose (powerwall.yml) setup.

Telegraf's job is to poll for data and store it in InfluxDB for use by Grafana.  What happens is this:

telegraf --(polls)--> pypowerwall --(polls)--> Powerwall Gateway

Pypowerwall doesn't proactively poll the Powerwall, it waits for a request. That is why you see "processing a request from" in the log. You will likely see other traffic too. You can directly hit pypowerwall via your browser or other means, and that is exactly what we do for the Power Flow Animation  built in to the proxy (e.g. http://localhost:8675/example.html).

If you want to test the patch to improve the exception errors you are seeing you can update `powerwall.yml` to use this: 

```
jasonacox/pypowerwall:0.8.3t53-beta
```
and restart 

```bash
./compose-dash.sh up -d
```

jgleigh, 2024-04-13:
No change with 0.8.3 :(

```2024-04-13 11:16:11 ----------------------------------------
2024-04-13 11:16:11 Exception occurred during processing of request from ('172.18.0.6', 41366)
2024-04-13 11:16:11 Traceback (most recent call last):
2024-04-13 11:16:11   File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
2024-04-13 11:16:11     self.finish_request(request, client_address)
2024-04-13 11:16:11   File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
2024-04-13 11:16:11     self.RequestHandlerClass(request, client_address, self)
2024-04-13 11:16:11   File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
2024-04-13 11:16:11     self.handle()
2024-04-13 11:16:11   File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
2024-04-13 11:16:11     self.handle_one_request()
2024-04-13 11:16:11   File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
2024-04-13 11:16:11     method()
2024-04-13 11:16:11   File "/app/server.py", line 321, in do_GET
2024-04-13 11:16:11     fcv["grid_status"] = pw.grid_status(type="numeric")
2024-04-13 11:16:11   File "/app/pypowerwall/__init__.py", line 616, in grid_status
2024-04-13 11:16:11     grid_status = payload.get('grid_status')
2024-04-13 11:16:11 AttributeError: 'NoneType' object has no attribute 'get'
2024-04-13 11:16:11 ----------------------------------------```

jgleigh, 2024-04-13:
A quick Google search found similar issues with other Python scripts and people aren't always sure why they happen intermittently.

jasonacox, 2024-04-13:
Hi @jgleigh - can you check http://localhost:8675/stats to make sure it show 0.8.3 as the version? If not, use this container: 

```
jasonacox/pypowerwall:0.8.3t53-beta2
```

jgleigh, 2024-04-13:
{"pypowerwall": "0.8.3 Proxy t53", "gets": 13664, "errors": 0

jasonacox, 2024-04-14:
Ok, another try... I saw where I was missing the empty payload. It will still log an error because this could be indicative of a bigger problem (unable to communicate with the Powerwall), but should not be a exception trace.

```
jasonacox/pypowerwall:0.8.3t54-beta
```

jgleigh, 2024-04-14:
2024-04-13 22:46:25 04/13/2024 10:46:25 PM [pypowerwall.local.pypowerwall_local] [ERROR] 503 Service Unavailable at https://192.168.1.6/api/solar_powerwall - Activating 5 minute API cooldown

I like it!

spoonwzd, 2024-04-15:
```
04/15/2024 06:33:42 AM [proxy] [INFO] Control Command: Set Reserve to 15
04/15/2024 08:00:04 AM [proxy] [INFO] Control Command: Set Reserve to 5
 CANCEL
```

Mine shows 'CANCEL' after setting the reserve this morning.

Seems to be working fine regardless.

jgleigh, 2024-04-18:
Seeing this a lot more now. Not sure what's going on. Also seeing periodic data dropouts of 2min each.

2024-04-18 15:16:07 04/18/2024 03:16:07 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe

jasonacox, 2024-04-22:
> 2024-04-18 15:16:07 04/18/2024 03:16:07 PM [proxy] [ERROR] Socket broken sending API response to client [doGET]: [Errno 32] Broken pipe

This is an indication of a network connection breaking. It is actually on the "client" side (e.g. web browser, telegraf, etc) and is could be due to a timeout of that client waiting for a response from the Powerwall (via the Proxy).

jgleigh, 2024-04-23:
Seeing a lot more errors lately. Anything else I can look at to see what is going on here?

```
Running for period: [2024-04-01 00:00:00-07:00] - [2024-04-23 23:59:59-07:00] (22 days, 23:59:59s)

Searching InfluxDB for data gaps (power usage)
* Found data gap: [2024-04-19 20:31:00-07:00] - [2024-04-19 20:38:00-07:00] (0:07:00s)
* Found data gap: [2024-04-20 04:16:00-07:00] - [2024-04-20 04:34:00-07:00] (0:18:00s)
* Found data gap: [2024-04-20 17:02:00-07:00] - [2024-04-20 17:10:00-07:00] (0:08:00s)
* Found data gap: [2024-04-21 17:17:00-07:00] - [2024-04-21 17:24:00-07:00] (0:07:00s)

Searching InfluxDB for data gaps (grid status)
* Found data gap: [2024-04-19 01:30:00-07:00] - [2024-04-19 01:33:00-07:00] (0:03:00s)
* Found data gap: [2024-04-19 18:15:00-07:00] - [2024-04-19 18:17:00-07:00] (0:02:00s)
* Found data gap: [2024-04-19 20:31:00-07:00] - [2024-04-19 20:38:00-07:00] (0:07:00s)
* Found data gap: [2024-04-20 04:16:00-07:00] - [2024-04-20 04:34:00-07:00] (0:18:00s)
* Found data gap: [2024-04-20 11:32:00-07:00] - [2024-04-20 11:37:00-07:00] (0:05:00s)
* Found data gap: [2024-04-20 17:02:00-07:00] - [2024-04-20 17:10:00-07:00] (0:08:00s)
* Found data gap: [2024-04-21 17:17:00-07:00] - [2024-04-21 17:24:00-07:00] (0:07:00s)

Retrieving data for gap: [2024-04-19 20:32:00-07:00] - [2024-04-19 20:37:59-07:00] (0:05:59s)
* Loading daily history: [2024-04-19] (America/Phoenix)
Retrieving data for gap: [2024-04-20 04:17:00-07:00] - [2024-04-20 04:33:59-07:00] (0:16:59s)
* Loading daily history: [2024-04-20] (America/Phoenix)
Retrieving data for gap: [2024-04-20 17:03:00-07:00] - [2024-04-20 17:09:59-07:00] (0:06:59s)
Retrieving data for gap: [2024-04-21 17:18:00-07:00] - [2024-04-21 17:23:59-07:00] (0:05:59s)
* Loading daily history: [2024-04-21] (America/Phoenix)

Retrieving backup event history
* Creating grid status data: [2024-04-19 01:31:00-07:00] - [2024-04-19 01:32:59-07:00] (0:01:59s)
* Creating grid status data: [2024-04-19 18:16:00-07:00] - [2024-04-19 18:16:59-07:00] (0:00:59s)
* Creating grid status data: [2024-04-19 20:32:00-07:00] - [2024-04-19 20:37:59-07:00] (0:05:59s)
* Creating grid status data: [2024-04-20 04:17:00-07:00] - [2024-04-20 04:33:59-07:00] (0:16:59s)
* Creating grid status data: [2024-04-20 11:33:00-07:00] - [2024-04-20 11:36:59-07:00] (0:03:59s)
* Creating grid status data: [2024-04-20 17:03:00-07:00] - [2024-04-20 17:09:59-07:00] (0:06:59s)
* Creating grid status data: [2024-04-21 17:18:00-07:00] - [2024-04-21 17:23:59-07:00] (0:05:59s)

Writing to InfluxDB
Updating InfluxDB
Done.
```

```
2024-04-21 17:24:45 ----------------------------------------
2024-04-21 17:24:45 Exception occurred during processing of request from ('172.18.0.4', 50070)
2024-04-21 17:24:45 Traceback (most recent call last):
2024-04-21 17:24:45   File "/app/pypowerwall/local/pypowerwall_local.py", line 91, in _get_session
2024-04-21 17:24:45     self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
2024-04-21 17:24:45     return self._find_no_duplicates(name)
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
2024-04-21 17:24:45     raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
2024-04-21 17:24:45 KeyError: "name='AuthCookie', domain=None, path=None"
2024-04-21 17:24:45 
2024-04-21 17:24:45 During handling of the above exception, another exception occurred:
2024-04-21 17:24:45 
2024-04-21 17:24:45 Traceback (most recent call last):
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
2024-04-21 17:24:45     self.finish_request(request, client_address)
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
2024-04-21 17:24:45     self.RequestHandlerClass(request, client_address, self)
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
2024-04-21 17:24:45     self.handle()
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
2024-04-21 17:24:45     self.handle_one_request()
2024-04-21 17:24:45   File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
2024-04-21 17:24:45     method()
2024-04-21 17:24:45   File "/app/server.py", line 293, in do_GET
2024-04-21 17:24:45     alerts = pw.alerts()
2024-04-21 17:24:45   File "/app/pypowerwall/__init__.py", line 484, in alerts
[... 381 lines cut ...]
2024-04-21 17:24:50 
2024-04-21 17:24:50 During handling of the above exception, another exception occurred:
2024-04-21 17:24:50 
2024-04-21 17:24:50 Traceback (most recent call last):
2024-04-21 17:24:50   File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
2024-04-21 17:24:50     self.finish_request(request, client_address)
2024-04-21 17:24:50   File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
2024-04-21 17:24:50     self.RequestHandlerClass(request, client_address, self)
2024-04-21 17:24:50   File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
2024-04-21 17:24:50     self.handle()
2024-04-21 17:24:50   File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
2024-04-21 17:24:50     self.handle_one_request()
[... cut ...]
2024-04-21 17:24:50   File "/app/pypowerwall/local/pypowerwall_local.py", line 182, in poll
2024-04-21 17:24:50     self._get_session()
2024-04-21 17:24:50   File "/app/pypowerwall/local/pypowerwall_local.py", line 83, in _get_session
2024-04-21 17:24:50     raise ConnectionError(err)
2024-04-21 17:24:50 ConnectionError: Unable to connect to Powerwall at https://192.168.1.6: HTTPSConnectionPool(host='192.168.1.6', port=443): Read timed out. (read timeout=5)
2024-04-21 17:24:50 ----------------------------------------
```

jgleigh, 2024-04-23:
Just noticed something...looks like my Gateway rebooted around 17:24 on 4/21. Not sure why. I'm going to keep an eye on that to see if my Gateway is starting to act up.

System Uptime: 38h30m36.182464703s

spoonwzd, 2024-04-23:
I was gonna say, my console looked the same when I was doing some core maintenance at the weekend that affected connectivity. `Read timed out. (read timeout=5)` is the clue.

jgleigh, 2024-05-01:
Another Gateway reboot about 10 hours ago. Starting to wonder if 24.4.0 has some issues.

jasonacox, 2024-05-02:
@jgleigh Yikes... Did it require you to reset your password?

jgleigh, 2024-05-02:
@jasonacox Nope, I only notice from the proxy logs. The system reboots fast enough that nothing goes offline.

jgleigh, 2024-05-03:
And again today...really don't know why it's so unstable these days. Either the Gateway computer is having issues or 24.4.0 has some seriously bad bugs.

  "start_time": "2024-05-02 13:50:57 +0800",
  "up_time_seconds": "5h56m3.175433462s",
  "is_new": false,
  "version": "24.4.0 0fe780c9",

jasonacox, 2024-05-03:
Good idea, @jgleigh - I am on the same Firmware but haven't seen the same thing. However, I do notice that start_time and up_time don't match but that could just be a integer wraparound. 

"start_time": "2024-03-21 21:04:21 +0800",
"up_time_seconds": "1007h8m57.41338074s",
"is_new": false,
"version": "24.4.0 0fe780c9",

jgleigh, 2024-05-03:
Just noticed there might a bug in the API...it's reporting the start time as +0800, but my time zone is actually -07:00 and the time reported is correct for -07:00. Unless that +0800 means something else. It looks like all the API documentation shows it always as +0800.

jgleigh, 2024-05-03:
> However, I do notice that start_time and up_time don't match but that could just be a integer wraparound.

Looks correct. Your start time is 9:04 PM on 3/21 which works out to 1007 hours ago.

jgleigh, 2024-05-08:
I noticed that PowerDashboard is still using proxy 0.8.2. What's the plan to sync up the two projects?

I got two longer dropouts today...very odd.

jasonacox, 2024-05-08:
Hi @jgleigh - Here is the plan to sync them up: https://github.com/jasonacox/Powerwall-Dashboard/pull/474

The tests have gone well and today I heard back from someone in the community that confirmed the new controls.  I'm ready to merge. 

However, I don't believe this update will help your issue. The behavior you are seeing seems very suspicious. It could be the hardware itself, or it could be some unusual load/traffic coming from somewhere on the network. My only suggestion would be based on what I would do. I would change the IP address (change the DHCP pinning to a new address) and power cycle the Powerwall Gateway. I don't know if that would help, but I would have to try. 

I'll merge the PR so you can pull the latest if you want.

jgleigh, 2024-05-08:
Agreed, there's something else going on with my system. I was more just curious since we've been testing out new proxy features. Thanks again.

closed 2024-05-08 by jgleigh

jasonacox, 2024-05-09:
Let me know what you think...  Hopefully it works. :)