jasonacox/pypowerwall #82: Change the path .auth & .site files are saved to

spoonwzd, 2024-04-13:
I'm running pypowerwall in an Unraid docker, but am having to re-perform the setup process each time I make a change to the docker and the image is refreshed.

It would be great to be able to define a custom path for .pypowerwall.auth and other volatile config related files so I can store them outside of the root file structure. Then in Unraid I can redirect a container path to a host path.

How would I achieve this?

jasonacox, 2024-04-13:
Hi @spoonwzd - I'm going to assume you mean the pypowerwall proxy container. There are two envrionmental variables you can set:

```bash
# PW_AUTH_PATH is used to prefix auth and cache files
PW_AUTH_PATH=/home/user/.secret/

# PW_CACHE_FILE is the absolute path to the .powerwall cache file
PW_CACHE_FILE=/home/user/.mypowerwall
```

If you are using something similar to [Powerwall-Dashboard](https://github.com/jasonacox/Powerwall-Dashboard), you can install these configurations in the pypowerwall.env file.

spoonwzd, 2024-04-13:
PW_AUTH_PATH was the key to success! Thanks!

closed 2024-04-13 by spoonwzd