jasonacox/pypowerwall #84: Slow charge speed from grid?

spoonwzd, 2024-04-14:
Back when you could set reserve on the local PowerWall gateway API it would always charge from the grid at the full 5kw.

When that broke I switched to using ToU in the Tesla app, and it always pulled 5kw from the grid when charging.

Now I've switched to pypowerwall and am setting the reserve, it only seems to be charging at ~1850W from the grid.

This may be coincidental of course and I have something else going on.

Any thoughts?

jasonacox, 2024-04-15:
I am on Self-Powered mode and my system has been charging at 1850W per Powerwall for the last 6 months or so. I never switched to TOU to see if it was different. I can't recall which Firmware version upgrade changed it, but it used to do as you say (full 5kW). I suspect Tesla adjusted that  to help save battery life. As I recall, it will still charge at a full 5kW for Storm Watch event mode but will slow charge for regular Self-Powered mode.

spoonwzd, 2024-04-15:
Understood. Filed under "feature"

closed 2024-04-15 by spoonwzd