jasonacox/pypowerwall #87: Changing Battery Configuration State Via GET

KevM, 2024-04-17:
I noticed that you added a control surface for the better reserve. I am very excited about this for my needs. Thank you all for working on this. 💯 

I was curious why you made the request a GET and not a POST? Maybe this has already been debated. My history with creating web APIs points to avoiding using GETs in front of anything that changes the state of the world. 

Feel free to close this if I am being too pedantic. 

Thanks again.

jasonacox, 2024-04-22:
Excellent point, @KevM !  I'll add it to my TODO list to move to POST.

jasonacox, 2024-04-29:
Staged update in #86 to add POST method. Available via beta proxy container:

```
jasonacox/pypowerwall:0.8.4t54-beta5
```

```bash
# Set Reserve to 20
curl -X POST -d "value=20&token=1234" http://localhost:8675/control/reserve

# Set Mode to self_consumption
curl -X POST -d "value=self_consumption&token=1234" http://localhost:8675/control/mode

# Read values by omitting value
curl -X POST -d "token=1234" http://localhost:8675/control/reserve
curl -X POST -d "token=1234" http://localhost:8675/control/mode
```

PR #86 (2024-05-05): v0.8.4
## Library Update for v0.8.4

Update set_reserve(level) logic to handle levels from 0 to 100 
Closes #85

## Proxy Update for t55

Update API to use POST method for `mode` and `reserve` setting changes.
Closes #87 

* Fix `/pod` API to add `time_remaining_hours` and `backup_reserve_percent` for cloud mode.
* Replaced t54 - Move control to POST see https://github.com/jasonacox/pypowerwall/issues/87
* Added GET APIs to retrieve backup reserve and operating mode settings
* Added POST command APIs to set backup reserve and operating mode settings. **Requires setting `PW_CONTROL_SECRET` for  [... cut]
closed 2024-05-05 by jasonacox