jasonacox/pypowerwall #90: Issues with POST

spoonwzd, 2024-05-06:
Things have stopped working for me since you implemented POST.

I can set the reserve correctly using POST using curl, from your example but when I set the method in my node-red function to POST it says my token is bad.

`"{"unauthorized": "Control Command Token Invalid"}"`

I'm not sure what I'm doing wrong. It should be as simple as changing msg.method in my custom function:

```
if (powerwall_reserve!=charge_level && overnight_charge===true) {
    msg.method = "POST"
    msg.url = "http://192.168.78.40:8675/control/reserve?token=MY_TOKEN&value=" + charge_level
return msg;
```

The only thing I can see different is your example sets the value before the token. Should this matter?

Any ideas?

jasonacox, 2024-05-07:
POST sends the payload via a message versus on the URL. I think you would need something like this:

```
if (powerwall_reserve != charge_level && overnight_charge === true) {
    msg.method = "POST";
    msg.url = "http://192.168.78.40:8675/control/reserve";
    msg.payload = "token=MY_TOKEN&value=" + charge_level;
    return msg;
}
```

spoonwzd, 2024-05-07:
Yup that was the secret sauce - thanks!

jasonacox, 2024-05-08:
Thanks for testing, @spoonwzd !  I'll close this but feel free to reopen.

closed 2024-05-08 by jasonacox