jasonacox/pypowerwall #93: Possible FleetAPI bug in __main__.py

darroni, 2024-05-23:
The latest pypowerwall (0.9) appears to have a bug in the #Get Powerwall Mode section.  When running in Cloud Mode with no FleetAPI configured, pypowerwall returns an exception on line 152:
```
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/darroni/myenv/lib/python3.12/site-packages/pypowerwall/__main__.py", line 152
    'site_id': pw.client.fleet.site_id
               ^^^^^^^^^^^^^^^^^^^^^^^
```
It looks like it's trying to force the site.id to the fleet.site_id.  I simply changed the variable to None, per your documentation and it solved the problem.

[image]

jasonacox, 2024-05-24:
Great find @darroni ! It also uncovered another bug. Thank you. I'll push out a patch.

jasonacox, 2024-05-24:
jasonacox/pypowerwall:0.9.1t57-beta

PR #94 (2024-05-24): v0.9.1 Bug Fixes and Updates
* Fix bug in time_remaining_hours() and convert print statements in FleetAPI to log messages.
* Fix CLI bug related to `site_id` as raised by @darroni in https://github.com/jasonacox/pypowerwall/issues/93
* Add CLI option for local mode to get status:

```bash
python -m pypowerwall get -host 10.1.2.3 -password 'myPassword'
```

Closes #93
closed 2024-05-24 by jasonacox