jasonacox/pypowerwall #97: Failed to Connect message for working connection - TEDAPI tool

ttl74ls100, 2024-06-08:
https://github.com/jasonacox/pypowerwall/blob/706afaf5f042913c86637a28f78c78dd899bd889/tools/tedapi/tedapi.py#L111

Ubuntu, python3, Powerwall 3 x 2, using static route to PW

command line
`python3 tedapi.py`
Enter password

Result:
```
Tesla Powerwall Gateway TEDAPI Reader

Connecting to Powerwall Gateway 192.168.91.1
Failed to connect to Powerwall Gateway
Traceback (most recent call last):
  File "/home/jc/pypowerwall/tools/tedapi/tedapi.py", line 305, in <module>
    ted = TEDAPI(gw_pwd)
          ^^^^^^^^^^^^^^
  File "/home/jc/pypowerwall/tools/tedapi/tedapi.py", line 65, in __init__
    raise ValueError("Failed to connect to Powerwall Gateway")
ValueError: Failed to connect to Powerwall Gateway
```

Request returns 404, which is expected.

Added line with comment  `# no exception, so the request was successful` and script ran successfully.

```python
        log.debug(f"Testing Connection to Powerwall Gateway: {GW_IP}")
        url = f'https://{GW_IP}'
        try:
            r = requests.get(url, verify=False, timeout=5)
            r = True # no exception, so the request was successful
        except requests.exceptions.RequestException as e:
            r = False
            log.error("ERROR: Powerwall not Found",
                      f"Try: sudo route add -host <Powerwall_IP> {GW_IP}")

```

jasonacox, 2024-06-08:
Thanks @ttl74ls100 !   I made the update.

Starting with v0.10.0, you don't need to use this tool and can talk to tedapi via the library. I'll create an example and post it as well.

jasonacox, 2024-06-09:
## Command Line Tool

With pyPowerwall v0.10.0, you can now run the TEDAPI tool via the  command line (CLI):

```
# First, upgrade if you haven't
pip install -U pypowerwall

# Run the tool
python3 -m pypowerwall.tedapi
```

It will prompt you for the Powerwall Gateway password (usually printed on the QR code on the gateway), and will then query the Powerwall for config and curren site data. It will display some and write entire payloads to `config.json` and `status.json` in the current directory. Example:

```
Tesla Powerwall Gateway TEDAPI Reader

Enter Powerwall Gateway Password: **********

Connecting to Powerwall Gateway 192.168.91.1

 - Configuration:
   - Site Name: Energy Gateway
   - Battery Commission Date: 2021-09-25T16:05:08-07:00
   - VIN:  123***-**-**-*--TG************
   - Number of Powerwalls: 2

 - Power Data:
   - Battery Charge: 100.0% (25772Wh of 25772Wh)
   - Battery: -10W
   - Site: -337W
   - Load: 1108W
   - Solar: 1460W
   - Solar_Rgm: 1447W
   - Generator: 0W
   - Conductor: 0W

 - Configuration and Status saved to config.json and status.json
 ```

## Using the Library

The following python script will connect to the gateway and display power data:

```python
# Import Class
from pypowerwall.tedapi import TEDAPI

gw_pwd = "THEGWPASS"

# Connect to Gateway
gw = TEDAPI(gw_pwd)

# Grab the Config and Live Status
config = gw.get_config()
status = gw.get_status()

# Print
site_info = config.get('site_info', {})
site_name = site_info.get('site_name', 'Unknown')
print(f"My Site: {site_name}")
meterAggregates = status.get('control', {}).get('meterAggregates', [])
for meter in meterAggregates:
    location = meter.get('location', 'Unknown').title()
    realPowerW = int(meter.get('realPowerW', 0))
    print(f"   - {location}: {realPowerW}W")
```

ttl74ls100, 2024-06-09:
Using a python:3.10-alpine container, I installed `pypowerwall` and ran `python3 -m pypowerwall.tedapi` with the following result:

```
Connecting to Powerwall Gateway 192.168.91.1
Failed to connect to Powerwall Gateway
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/app/jctest-env/lib/python3.10/site-packages/pypowerwall/tedapi/__main__.py", line 27, in <module>
    ted = TEDAPI(gw_pwd)
  File "/app/jctest-env/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 86, in __init__
    raise ValueError("Failed to connect to Powerwall Gateway")
ValueError: Failed to connect to Powerwall Gateway
```

Here are the version numbers from the install:
```
(jctest-env) /app/jctest-env # pip install -U pypowerwall
Requirement already satisfied: pypowerwall in ./lib/python3.10/site-packages (0.10.2)
Requirement already satisfied: requests in ./lib/python3.10/site-packages (from pypowerwall) (2.32.3)
Requirement already satisfied: protobuf in ./lib/python3.10/site-packages (from pypowerwall) (5.27.1)
Requirement already satisfied: teslapy in ./lib/python3.10/site-packages (from pypowerwall) (2.9.0)
Requirement already satisfied: charset-normalizer<4,>=2 in ./lib/python3.10/site-packages (from requests->pypowerwall) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in ./lib/python3.10/site-packages (from requests->pypowerwall) (3.7)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./lib/python3.10/site-packages (from requests->pypowerwall) (2.2.1)
Requirement already satisfied: certifi>=2017.4.17 in ./lib/python3.10/site-packages (from requests->pypowerwall) (2024.6.2)
Requirement already satisfied: requests-oauthlib in ./lib/python3.10/site-packages (from teslapy->pypowerwall) (2.0.0)
Requirement already satisfied: websocket-client>=0.59.0 in ./lib/python3.10/site-packages (from teslapy->pypowerwall) (1.8.0)
Requirement already satisfied: oauthlib>=3.0.0 in ./lib/python3.10/site-packages (from requests-oauthlib->teslapy->pypowerwall) (3.2.2)
```

jasonacox, 2024-06-09:
Thanks @ttl74ls100 - I can only replicate that if there is no route to the Gateway.  I see where I need to adjust the code to help provide a clear message, but can you verify that you are able to ping / curl the Gateway (192.168.91.1) from that container host?

ttl74ls100, 2024-06-09:
I believe the gateway is working.   For example, after the patch to `tedapi.py` in the tools folder, I am able get configuration a power data. 

```
(jctest-env) /app/jctest-env # curl -k https://192.168.91.1
404 page not found
```

Could the issue be related to the 404 response?

ttl74ls100, 2024-06-09:
I should also highlight I am connecting to a Powerwall 3

jasonacox, 2024-06-09:
> I should also highlight I am connecting to a Powerwall 3

@ttl74ls100 , I didn't expect this to work on the PW3! The fact that you got the first to work is a positive step. Thanks for helping test! 🙏 

I updated the logic to give a bit more details to help diagnose:

```bash
# First install the dev version
pip install pypowerwall==0.10.3.dev0

# Run again
python3 -m pypowerwall.tedapi

# Run with debug
python3 -m pypowerwall.tedapi --debug
```

ttl74ls100, 2024-06-09:
It worked.

```
(jctest-env) /app/jctest-env # python3 -m pypowerwall.tedapi
pyPowerwall - Powerwall Gateway TEDAPI Reader
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: <>

Connecting to Powerwall Gateway 192.168.91.1

 - Configuration:
   - Site Name: <>
   - Battery Commission Date: 2024-01-31T15:36:12-05:00
   - VIN: <>
   - Number of Powerwalls: 2

 - Power Data:
   - Battery Charge: 49.32% (14210Wh of 28810Wh)
   - Battery: 2058W
   - Site: -5W
   - Load: 3375W
   - Solar: 1332W
   - Solar_Rgm: 0W
   - Generator: 0W
   - Conductor: 0W

 - Configuration and Status saved to config.json and status.json
```

Thanks for your help,   Let me know if you need any PW3 testing.

jasonacox, 2024-06-09:
Excellent!  Thank you, @ttl74ls100 !

It seems that the PW3 responds with 404 for the `/` URI which means we could use that to detect a PW3 and accommodate with some internal logic. Can you check to see if any of these work or the type of responses you get?

```python
import pypowerwall
pypowerwall.set_debug()

# Set these
PW_HOST="192.168.91.1"
PW_GW_PWD="xxxxxxxx" # Your Gateway Password

pw = pypowerwall.Powerwall(PW_HOST, PW_GW_PWD ,gw_pwd=PW_GW_PWD)

# Get vitals - should work
pw.vitals()

# Get current power, level - likely will not work
pw.power()
pw.level()

# Get alerts - may partially work
pw.alerts()
```

ttl74ls100, 2024-06-10:
**pw.vitals()**
```
DEBUG:Get Status from Powerwall
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/v1 HTTP/1.1" 200 1700
DEBUG:Response Code: 200
DEBUG:Status: {'control': {'alerts': {'active': ...
DEBUG:Get Configuration from Powerwall
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/v1 HTTP/1.1" 200 1310
DEBUG:Response Code: 200
DEBUG:Configuration: {'vin': '<>', 'meters': [{'location': 'site', 'type': 'synchrometerX', 'cts': [True, True, False, False]  ...
{'VITALS': {'text': 'Device vitals generated from Tesla Powerwall Gateway TEDAPI', 'timestamp': 1717978179.1119115, 'gateway': '192.168.91.1', 'pyPowerwall': '0.10.3'}, 
...
```

**pw.power()**
```
DEBUG: -- local: Request Powerwall for /api/meters/aggregates
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET /api/meters/aggregates HTTP/1.1" 404 19
ERROR:404 Powerwall API not found at https://192.168.91.1/api/meters/aggregates
DEBUG:ERROR unable to parse payload 'None': 'NoneType' object is not subscriptable
{'site': 0.0, 'solar': 0.0, 'battery': 0.0, 'load': 0.0}
```
**pw.level()**
```
DEBUG: -- local: Request Powerwall for /api/system_status/soe
DEBUG:https://192.168.91.1:443 "GET /api/system_status/soe HTTP/1.1" 404 19
ERROR:404 Powerwall API not found at https://192.168.91.1/api/system_status/soe
```

**pw.alerts()**
```
DEBUG:Get Status from Powerwall
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/v1 HTTP/1.1" 200 1716
DEBUG:Response Code: 200
DEBUG:Status: {'control': {'alerts': {'active': ...
DEBUG:Using Cached Payload
DEBUG: -- local: Request Powerwall for /api/system_status/grid_status
DEBUG:https://192.168.91.1:443 "GET /api/system_status/grid_status HTTP/1.1" 404 19
ERROR:404 Powerwall API not found at https://192.168.91.1/api/system_status/grid_status
['SelfConsumptionReservedLimit', 'SystemConnectedToGrid', 'FWUpdateSucceeded']
```

jasonacox, 2024-06-10:
Thanks, @ttl74ls100 !  This is very helpful. I suspected we would see 404 for those calls but needed to confirm. I can build mappings from the tedapi to most of those calls. I'll work on something for you to try soon.

jasonacox, 2024-06-10:
Hi @ttl74ls100 - something for you to try:

```bash
# First install the dev version
pip install pypowerwall==0.10.4.dev0
```

Simple Python test:

```python
import pypowerwall

PW_GW_PWD="GWPASSWD"
pw = pypowerwall.Powerwall("192.168.91.1", gw_pwd=PW_GW_PWD)

pw.power()
pw.level()
pw.battery_blocks()
```

You can also try the pypowerwall proxy:

```bash
docker run \
    -p 8675:8675 \
    -e PW_PORT='8675' \
    -e PW_HOST='192.168.91.1' \
    -e PW_GW_PWD="GWPASSWD" \
    --name pypowerwall \
    --restart unless-stopped \
   jasonacox/pypowerwall:0.10.4t61-beta
```

Try these URLs:
* http://localhost:8675/stats
* http://localhost:8675/soe
* http://localhost:8675/vitals
* http://localhost:8675/aggregates
* http://localhost:8675/pod
* http://localhost:8675/freq
* http://localhost:8675/alerts
* http://localhost:8675/example.html - Powerflow Animation

ttl74ls100, 2024-06-11:
All the calls executed without error.   I can post results if you want, I am leaving them out for brevity in case the actual values are not relevant.  

[image]

:tada:

jasonacox, 2024-06-11:
Fantastic!!!  Thanks for testing!  🙏 

* Are you running the [Powerwall-Dashboard](https://github.com/jasonacox/Powerwall-Dashboard)? If so, you can use this container to get the extended local metrics (e.g. alerts/strings). My plan is to roll this out to that project and provide setup instructions for PW3 users.
* How are you getting access to the Gateway IP? Are you connecting via WiFi or did you find it on your LAN and add a route?
* You will notice that the Firmware data is wrong. The TEDAPI payloads for config or status do not contain the firmware number, at least from what I can tell. If anyone finds a way to pull this, let me know. There are a few other missing metrics but nothing too serious. The vitals data is rich with goodness.

jasonacox, 2024-06-11:
Updated: the [Powerwall-Dashboard](https://github.com/jasonacox/Powerwall-Dashboard) will now provide an option to install for Powerwall 3 using the local gateway option. If you install and run `setup.sh` it will walk you through the process (hopefully) :) .

ttl74ls100, 2024-06-11:
I have a static route setup in the gateway/router (UniFi UDM) to the primary PW.  The production Powerwall-Dashboard instance is running on a Proxmox VM and currently the `curl -k https://192.168.91.1` test is timing out...  Need to dig into the networking configs to see what is up there.    

I am using WSL on a Windows machine for `pypowerwall` to PW3 testing. 

I'll stand-up another instance of Powerwall-Dashboard using local access and let you know how it goes.

ttl74ls100, 2024-06-12:
Powerwall Dashboard is working with local PW3   w00t!   Powerwall Capacity panel in Powerwall Vitals has no data. Energy Usage, Grid Status, Frequency, Voltages and Alerts working.

[image]

[image]

jasonacox, 2024-06-12:
Looks great @ttl74ls100 !!

The Powerwall Capacity graph is puzzling. Can you show the results of http://localhost:8675/api/system_status ? It should have the capacity data. If it does check: http://localhost:8675/pod which is what the Dashboard is polling. It should list the different PW data points like `PW2_POD_nom_full_pack_energy`.

ttl74ls100, 2024-06-12:
**/api/system_status**

```json
{
  "command_source": "Configuration",
  "battery_target_power": 0,
  "battery_target_reactive_power": 0,
  "nominal_full_pack_energy": 28840,
  "nominal_energy_remaining": 10570,
  "max_power_energy_remaining": 0,
  "max_power_energy_to_be_charged": 0,
  "max_charge_power": null,
  "max_discharge_power": null,
  "max_apparent_power": null,
  "instantaneous_max_discharge_power": 0,
  "instantaneous_max_charge_power": 0,
  "instantaneous_max_apparent_power": 0,
  "hardware_capability_charge_power": 0,
  "hardware_capability_discharge_power": 0,
  "grid_services_power": null,
  "system_island_state": "SystemGridConnected",
  "available_blocks": 0,
  "available_charger_blocks": 0,
  "battery_blocks": [],
  "ffr_power_availability_high": 0,
  "ffr_power_availability_low": 0,
  "load_charge_constraint": 0,
  "max_sustained_ramp_rate": 0,
  "grid_faults": [],
  "can_reboot": "Yes",
  "smart_inv_delta_p": 0,
  "smart_inv_delta_q": 0,
  "last_toggle_timestamp": "2023-10-13T04:08:05.957195-07:00",
  "solar_real_power_limit": null,
  "score": 10000,
  "blocks_controlled": 0,
  "primary": true,
  "auxiliary_load": 0,
  "all_enable_lines_high": true,
  "inverter_nominal_usable_power": 0,
  "expected_energy_remaining": 0
}
```
**/pod**
```json
{
  "time_remaining_hours": 11.0975932293044,
  "backup_reserve_percent": 33.5
}
```

jasonacox, 2024-06-13:
Thanks @ttl74ls100 !

> "battery_blocks": [],

Ugh, there is the problem. For previous Powerwalls, we have an entry there for each Powerwall. Can you look at http://localhost:8675/tedapi/status and see if you have entries for 'esCan' -> 'bus' -> 'THC'?  That is what builds the battery_block data. I wonder if it is missing data or a code problem.

Also, the latest beta release I'm testing adds api locking which should help with overloading the /teadpi:

```
jasonacox/pypowerwall:0.10.5t61-beta8
```

It may also be good to see what you have in your logs:

```
docker logs pypowerwall
```

ttl74ls100, 2024-06-13:
**Status has a `batteryBlocks` array:**

```
 "batteryBlocks": [
      {
        "din": "1707000-25-G--TG1232151234VR",
        "disableReasons": null
      },
      {
        "din": "1707000-25-G--TG1232151234RL",
        "disableReasons": null
      }
    ]
```

**Here is `esCan.bus.THC`:**
```
        "THC": [
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          },
          {
            "THC_InfoMsg": {
              "THC_appGitHash": [0, 0, 0, 0, 0, 0, 0],
              "isComplete": false,
              "isMIA": true
            },
            "THC_Logging": {
              "THC_LOG_PW_2_0_EnableLineState": "ENABLE_LINE_LOW"
            },
            "packagePartNumber": "",
            "packageSerialNumber": ""
          }
        ]
```

**docker logs pypowerwall**
```
06/13/2024 06:54:04 PM [proxy] [INFO] pyPowerwall [0.10.5] Proxy Server [t61] - HTTP Port 8675
06/13/2024 06:54:04 PM [proxy] [INFO] pyPowerwall Proxy Started
06/13/2024 06:54:04 PM [proxy] [INFO] pyPowerwall Proxy Server - Local Mode
06/13/2024 06:54:04 PM [proxy] [INFO] Connected to Energy Gateway 192.168.91.1 (JC)
06/13/2024 06:54:04 PM [proxy] [INFO] TEDAPI Mode Enabled for Device Vitals (full)
```

jasonacox, 2024-06-14:
Thank you @ttl74ls100  - Sadly that is what I feared. The energy levels are not showing for Powerwall 3s. However, I can pull the aggregates and include them in the /pod API. Update to:

```
jasonacox/pypowerwall:0.10.5t62-beta
```

Edit the "Powerwall Capacity" graph you can add the total battery capacity and remaining metrics, you can create these queries: 

[image]

If you have a chance to try it, let me know how it work. Thanks again for all your help!

jasonacox, 2024-06-16:
Hi @ttl74ls100 - One thing we are missing for PW3 is the Firmware Version. It is not included in the TEDAPI payloads. Could you run this test for us? I suspect it is still a 404, but I would like to verify.

```
curl -ki https://192.168.91.1/api/status
```

If anyone else finds an API to get the firmware version of a PW3, let us us know.

ttl74ls100, 2024-06-16:
404

jasonacox, 2024-06-16:
> 404

Thanks @ttl74ls100 !  Bummer, but as I expected.  I'll keep hunting... :)

welbymcroberts, 2024-06-27:
Just throwing my hat into the ring here as well, I have 4 PW3's (1 leader, 3 followers) and a AC Coupled Solar (Enphase) system, if any different requests are of use. Also happy to setup a shell if you want to get access to poke about.

jasonacox, 2024-06-27:
Thanks @welbymcroberts - There is also a discussion here where a new API has been found: https://github.com/jasonacox/Powerwall-Dashboard/discussions/392#discussioncomment-9864364

I'm traveling this week so limited on what I can explore but it does look promising if we can find a way to unlock it.

jasonacox, 2024-06-28:
@welbymcroberts If you are interested in helping test, we are exploring a query that may give more PW3 data mentioned in the link above. You can test the script: https://github.com/jasonacox/pypowerwall/blob/main/tools/tedapi/ComponentsQuery.py (also requires downloading [tedapi_pb2.py](https://github.com/jasonacox/pypowerwall/blob/main/tools/tedapi/tedapi_pb2.py) )

geptto, 2024-07-03:
> Hi @ttl74ls100 - One thing we are missing for PW3 is the Firmware Version. It is not included in the TEDAPI payloads. Could you run this test for us? I suspect it is still a 404, but I would like to [...]

Found the PW3 firmware version!
It is returned with the following pb (raw) request to /tedapi/v1.

```
1 {
  1: 1
  2 {
    3: 1
  }
  3 {
    1: "1707000-00-J--TG9999999999XP"
  }
  4 {
    2: ""
  }
}
2 {
  1: 1
}`

Sample response.

`1 {
  1: 1
  2 {
    1: "1707000-00-J--TG9999999999XP"
  }
  3 {
    3: 1
  }
  4 {
    3 {
      1 {
        1: "1707000-00-J"
        2: "TG9999999999XP"
      }
      2: "1707000-00-J--TG9999999999XP"
      3 {
        1: "24.12.6-PW3-AFCI 008bf6ff"    <--- PW3 firmware version
        2: "\000\213\366\...Redacted..."
      }
      5 {
        2: 1
      }
      6: 4
      7 {
        1 {
          1 {
            1: "Quectel"
          }
          2 {
            1: "BG95-M2"
          }
          3 {
            1: "XMR2020BG95M2"
          }
          4 {
            1: "10224A-2020BG95M2"
          }
        }
        1 {
          1 {
            1: "Texas Instruments"
          }
          2 {
            1: "WL18MODGI"
          }
          3 {
            1: "Z64-WL18DBMOD"
          }
          4 {
            1: "451I-WL18DBMOD"
          }
        }
      }
      8: "\370!s\306\212...Redacted..."
      9: "\373U\353\322...Redacted..."
    }
  }
}
2 {
  1: 1
}
```

jasonacox, 2024-07-06:
This is awesome @geptto !  Thank you! 🙏 

I created a test to get the firmware version:

```bash'
git clone https://github.com/jasonacox/pypowerwall.git
cd pypowerwall/tools/tedapi
# or git pull if you have it already

python3 ComponentsQuery.py GW_PASSWD
```

My Powerwall 2/+ doesn't have some of the PW3 elments so I will be missing (404) some items, but I am getting the Firmware version:

```
Tesla Powerwall Gateway API Decoder
 - Testing Connection to Powerwall Gateway...
 - Using Powerwall Gateway Password: GATEWAY_PWD
 - Fetching DIN from Powerwall...
 - Connected: Powerwall Gateway DIN: 1232100-00-E--REDACTED
 - Fetching Configuration from Powerwall...
 - Config Written to config.json
 - Battery Blocks:
   - Battery Block: 2012170-25-E--REDACTED (SolarPowerwall)
   - Battery Block: 3012170-05-B--REDACTED (ACPW)
 - Fetching /tedapi/v1 PW3 Firmware from Powerwall (1232100-00-E--REDACTED)...
Response Code: 200
 - Firmware Written to firmware.raw
Firmware version (len=15): 24.4.0 0fe780c9
 - Fetching /tedapi/v1 PW3 ComponentsQuery from Powerwall (1232100-00-E--REDACTED)...
Response Code: 200
Payload (len=128): {"components":{"baggr":[],"bms":[],"hvp":[],"pch":[],"pws":[]},"pw3Can":{"firmwareUpdate":{"isUpdating":false,"progress":null}}}
 - Components Written to components.json
 - Fetching /tedapi/device/<din>/v1 PW3 ComponentsQuery from Powerwall (2012170-25-E--REDACTED SolarPowerwall)...
Response Code: 404
 - No Components Found
 - Fetching /tedapi/device/<din>/v1 PW3 ComponentsQuery from Powerwall (3012170-05-B--REDACTED ACPW)...
Response Code: 404
 - No Components Found
 ```

ttl74ls100, 2024-07-27:
I updated my Production Powerwall-Dashboard to use local access for the Powerwall 3.    It works great.  Thanks for all the work.

I am having a strange occurrence.   About once a day, data collection stops.   Here is a sample from the `pypowerwall` log:

```
07/26/2024 04:04:25 PM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
07/26/2024 04:04:29 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port
=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection obje
ct at 0x7eb7c7235150>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
07/26/2024 04:04:29 PM [pypowerwall.tedapi] [ERROR] Error fetching config: HTTPSConnectionPool(host='192.168.91.1', port
=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection obje
ct at 0x7eb7c7237010>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
07/26/2024 04:04:34 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port
=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection obje
ct at 0x7eb7c7236dd0>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
07/26/2024 04:04:34 PM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
07/26/2024 04:04:34 PM [pypowerwall.tedapi] [ERROR] Error fetching config: HTTPSConnectionPool(host='192.168.91.1', port
=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection obje
ct at 0x7eb7c72143d0>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
```

So here is the strange thing....    The following three commands resolves the problem and data collection resumes (without restarting any containers)

```
jc@pwdb2:~$ curl -k https://192.168.91.1
^C
jc@pwdb2:~$ curl -k https://10.20.20.116
404 page not found
jc@pwdb2:~$ curl -k https://192.168.91.1
404 page not found
```

`10.20.20.116` is the address of the primary PW3 on the LAN.  I used curl to test 192.168.91.1 and it was non responsive.   Then used curl to see if I could access the PW3 from its internal address, and it 'worked' by returning 404.  After the request on the internal IP, curl on the 192.168.91.1 address starts working again.    Checking the Dashboard shows data collection resumed.

I have a static route setup in my Unifi router for 192.168.91.0/24 to 10.20.20.116.   No other special network configuration other than the machine is a VM in Proxmox. 

I'm not sure how to diagnose this.  I did have the idea to try to set `PW_HOST=10.20.20.116` in `pypowerwall.env` but that did not work.   It would be interesting to see if switching to the LAN address resolves the issue.

jasonacox, 2024-07-27:
Hi @ttl74ls100 , thanks for the report!  I can't think of an easy fix, but see a few things here that might help with troubleshooting:

* First, keep in mind that you **must** use the 192.168.91.1 address to get the TEDAPI data (the local metrics). The gateway will give a 40x error if you try from any other address (as you saw in your curl example).

>  Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection obje
ct at 0x7eb7c72143d0>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))

This is helpful. This is indicating either a network issue or the Gateway itself is in a bad state (like a server over loaded and can't respond). The fact that you can get a 404 on the LAN address seems to indicate that the Gateway web server is fine. That would point me towards this being a network issue (which includes everything from the VM, host, switch, router, firewall, etc).  A few thoughts:
* By setting up the 192.168.91.1 to be routable from anything on your network, you could be exposing the TEDAPI to a lot of broadcast or other traffic that could cause issues. That seems unlikely, due to my point above but putting it out there.
* Is the VM host having any issues in processing traffic? Are you able to ping the 192.168.91.1 from the host when the above occurs?

>  The following three commands resolves the problem and data collection resumes (without restarting any containers)

I agree, this is odd. Again, could it be the host? I would try:
* Install the Dashboard on another host to see if there are issues with the VM setup
* Add a cron job to curl the 192.168.91.1 endpoint every 5m or so to help identify when it occurs
* Check to make sure host isn't running out of file handles or ephemeral ports (unlikely) 
* Set up the static route on the host instead of at the router level to see if this helps (this is my setup by the way, I don't have a router that can do static routes)

I would love to hear if anyone else has ideas or is seeing the same thing.

leesamtao, 2024-11-03:
I had local access to PW3 working fine for about a week and a few days ago I started to run into the same issue mentioned here where it will randomly stop logging for a while.  pypowerwall log shows:

Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f26e7ac6380>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))

If I notice it happening and just run a ping to 192.168.91.1 it will seem to make it start working again, or else it will randomly start working by itself a few hours later. 

@ttl74ls100 - did you ever get a resolution to this?

leesamtao, 2024-11-03:
Just to add more info, I have a Windows 11 host and I have a permanent route setup from the host. When I notice it occurring and run a ping to 192.168.91.1 the first few requests will time out until it suddenly starts working again.

jasonacox, 2024-11-03:
Thanks @leesamtao - The ping timeout does indicate a network issue. It is good that it eventually recovers, but why does it happen? I had issues were my PW would disappear from my network for 1-5m randomly throughout the day. I was using WiFi and figured my wireless router was just having trouble reaching the PW as it is the furthest device on our network, even though the signal strength looked good. Anyway, I switched to using a hardwired ethernet cable to the Powerwall and all those problems went away. Others report solid performance with WiFi so I suspect it has something to do with my WLAN, house geometry or the wifi lottery I got with the PW. ;)

leesamtao, 2024-11-03:
I'll keep looking into the network issues, though everything seems to function fine the whole time, just stops responding on 192.168.91.1 randomly, unfortunately hardwired ethernet isn't an option for my setup. 

Could there be an option for powerwall dashboard to pull the main energy data from the cloud since that seems to always work for me, but still pull the string data/vitals from the local access? If the string/vitals data goes out occasionally that's less an issue than missing hours of energy data randomly.

jasonacox, 2024-11-04:
This could be an arp issue if a simple ping resolves it. Are you setting the route to 192.168.91.1 on the host or via your router? If via the host, you might try refreshing the arp with `arp -d x.x.x.x` where x.x.x.x is the LAN IP address for the Powerwall.

leesamtao, 2024-11-07:
Thanks for the additional info. Still happening typically once or twice a day, no pattern on the timing that I can identify. I did confirm it seems to only occur from my host windows machine. I can ping the powerwall from other machines without issue when it occurs. Pinging the powerall ip from the host machine has always fixed it once it occurs. Refreshing the arp table doesn't seem to change anything. Also tried a new static ip for the powerwall, same thing occurs. So it seems to be something with my windows host networking, but just at a loss for what it could be. Possible something with my router, as my host is the only ethernet connected device on my network.

jasonacox, 2024-11-08:
Thanks for the update @leesamtao . I don't use Windows, so I'm not going to be any help unfortunately. We do have some Windows gurus in the community that may chime in with some suggestions.

jakekeeys, 2025-01-20:
can anyone confirm if this works with the "gateway 3"?
also be super grateful if someone could post the "gateway 3" responses for https://github.com/jasonacox/pypowerwall/issues/97#issuecomment-2157451511

jasonacox, 2025-01-20:
> can anyone confirm if this works with the "gateway 3"?

I don't recall anyone posting results with the GW3, but can you give us your experience? If it impacts pypowerwall let's open a new Issue too.

> also be super grateful if someone could post the "gateway 2/3" responses for https://github.com/jasonacox/pypowerwall/issues/97#issuecomment-2157451511

Hi @jakekeeys  - AG7GN posted some from his system here: https://github.com/jasonacox/Powerwall-Dashboard/discussions/572#discussioncomment-11857649  - Key learning is that with PW3, you connect to the "leader" PW3 and not the gateway.

kylebrowning, 2025-02-21:
I can confirm that `arp` works correctly when the pinging dies :( I am hardwired ethernet 
```
ocy@raspberrypi:~ $ ping 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
^C
--- 192.168.91.1 ping statistics ---
8 packets transmitted, 0 received, 100% packet loss, time 7166ms

ocy@raspberrypi:~ $ ping 192.168.1.56
PING 192.168.1.56 (192.168.1.56) 56(84) bytes of data.
^C
--- 192.168.1.56 ping statistics ---
66 packets transmitted, 0 received, 100% packet loss, time 66562ms

ocy@raspberrypi:~ $ arp -a 192.168.1.56
TeslaPW_LZLJTV.localdomain (192.168.1.56) at 58:2b:0a:6d:72:25 [ether] on eth0
```

jakekeeys, 2025-05-03:
> I can confirm that `arp` works correctly when the pinging dies :( I am hardwired ethernet
@kylebrowning did you get this working over ethernet, mine is being installed on friday, wondering if i should bother to run a cable or not

JohnJ9ml, 2025-05-03:
I would always run a cable. I'd never give up the option.

kylebrowning, 2025-05-08:
> > I can confirm that `arp` works correctly when the pinging dies :( I am hardwired ethernet
> 
> @kylebrowning did you get this working over ethernet? mine is being installed on friday, wondering if i should bother to run a cable or not
> 
> 

No, my connection to the TEDApi is fully dead and I havnt been able to restore.

Considering asking a few friends at Tesla HQ what's up

jasonacox, 2025-05-08:
Are you on Firmware 25.10?  https://github.com/jasonacox/pypowerwall/issues/165

kylebrowning, 2025-11-09:
> > > > I can confirm that `arp` works correctly when the pinging dies :( I am hardwired ethernet
> > > 
> > > 
> > > [@kylebrowning](https://github.com/kylebrowning) did you get this working over ethernet? mine is being installed on friday, wondering if i should bother to run a cable or not
> > 
> > No, my connection to the TEDApi is fully dead and I havnt been able to restore.
> > 
> > Considering asking a few friends at Tesla HQ what's up
> 
> Try https://github.com/jasonacox/pypowerwall/issues/165#issuecomment-3507207979 and let me know if the PoC works.

Wish I could, my brain powerwall has been dead for 2 months and I dont get service until Dec 1

jasonacox-sam, 2026-09-06:
This epic thread established local TEDAPI access for Powerwall 3 (via 192.168.91.1) — a major community effort. Much of what was discovered and troubleshot here (connection dropouts, ARP/routing issues, gateway password quirks, firmware 25.10 changes) has since been productized: recent releases include [v0.16.5 – PW3 Lifetime Energy Accumulators via Gateway Local API](https://github.com/jasonacox/pypowerwall/releases/tag/v0.16.5), [v0.16.3/v0.16.4 TEDAPI robustness fixes](https://github.com/jasonacox/pypowerwall/releases/tag/v0.16.4), and [v0.17.0 – TEDAPI Bearer Authentication Mode](https://github.com/jasonacox/pypowerwall/releases/tag/v0.17.0).

Closing as historically-resolved — PW3 local access is now a supported, documented feature rather than an open investigation. For current PW3 connection issues, please open a new issue so it gets fresh attention.

— Sam ⚡

closed 2026-09-06 by jasonacox-sam