jasonacox/pypowerwall #111: PyPowerwallFleetAPI() needs to handle non-energy products

gregrahn, 2024-09-05:
In trying to get pypowerwall to use the fleet API I experienced several errors that I've tracked down.

When pypowerwall calls `/api/1/products` it returns both a car and power product for me like below (shorted and redacted).  Because the car details do not contain a key `energy_site_id`, code that assumes this key exists throws a KeyError exception, and never gets to check `sites[1]` so an additional check is needed for that.

Also, I noticed that in trying to get the setup completed through some hacking, the `site_id` in `.pypowerwall.fleetapi` was stored as a string, not a number.  This causes comparison checks to fail given the API for products returns `energy_site_id` as a number. 

Once I changed `site_id` in `.pypowerwall.fleetapi` to be a number and added a key check, it worked as expected.
 

Shorted  `/api/1/products` response:
```json
{
  "response": [
    {
      "id": 12345,
      "user_id": 321,
      "vehicle_id": 12345,
      "vin": "VIN3E8MFJK147",
      "color": null,
      "access_type": "OWNER",
      "display_name": "Tesla"
    },
    {
      "energy_site_id": 2255994400559966,
      "resource_type": "battery",
      "site_name": "1 E Main",
      "id": "STE9999-9999",
      "gateway_id": "0000000-00-E--TG000000000FYN",
      "asset_site_id": "",
      "grid_installation_id": "",
      "warp_site_number": "",
      "percentage_charged": 49.69920997318257,
      "battery_type": "ac_powerwall",
      "battery_power": 10   
    }
  ],
  "count": 2
}
```

Key check
```diff
diff --git a/pypowerwall/fleetapi/pypowerwall_fleetapi.py b/pypowerwall/fleetapi/pypowerwall_fleetapi.py
index 091cab3..60ae0ce 100644
--- a/pypowerwall/fleetapi/pypowerwall_fleetapi.py
+++ b/pypowerwall/fleetapi/pypowerwall_fleetapi.py
@@ -154,10 +154,11 @@ class PyPowerwallFleetAPI(PyPowerwallBase):
         else:
             found = False
             for idx, site in enumerate(sites):
-                if site['energy_site_id'] == self.siteid:
-                    self.siteindex = idx
-                    found = True
-                    break
+                if 'energy_site_id' in site:
+                    if site['energy_site_id'] == self.siteid:
+                        self.siteindex = idx
+                        found = True
+                        break
```

jasonacox, 2024-09-06:
Great find @gregrahn !!! Wow, that is bad.  I don't have a car, so  my tests were passing. Would you be willing to submit a PR? I want you to get credit for this great work.

jasonacox, 2024-09-07:
I'll combine this with #112 for a hotfix release.

jasonacox, 2024-09-07:
Thanks for your help with this @gregrahn !

* New version: https://pypi.org/project/pypowerwall/0.11.1/
* New proxy: jasonacox/pypowerwall:0.11.1t64

jasonacox, 2024-11-27:
Please reopen if the issue persists.

closed 2024-11-27 by jasonacox