jasonacox/pypowerwall #120: Proxy Server TEDAPI MODE

erikgieseler, 2024-12-11:
Hello,
does the proxy already support the new TEDAPI MODE?
If so, how is it activated?
Thanks for the feedback!
Erik

jasonacox, 2024-12-12:
Hi @erikgieseler - Yes, here is how - make sure you update PW_GW_PWD to match your gateway password:

Proxy Server Example

```bash
# Bash startup for pypowerwall proxy
echo "TEDAPI Mode Full (Powerwall 3)"
export PW_HOST="192.168.91.1"
export PW_GW_PWD="XXXXXXXX"

echo "Running Proxy..."
python3 proxy/server.py
```

Proxy Server Docker Example

```bash
# Docker startup for pypowerwall proxy
docker run \
    -d \
    -p 8675:8675 \
    -e PW_PORT='8675' \
    -e PW_HOST='192.168.91.1' \
    -e PW_GW_PWD='XXXXXXXX' \
    --name pypowerwall \
    --restart unless-stopped \
    jasonacox/pypowerwall
```

And just for reference:

Command Line

```bash
# Run Test
python3 -m pypowerwall.tedapi
```

```python
# Python example

import pypowerwall

PW_GW_PWD="XXXXXXXX"
PW_HOST="192.168.91.1"

pw = pypowerwall.Powerwall(PW_HOST, gw_pwd=PW_GW_PWD)

print(f"mode: {pw.mode} - tedapi: {pw.tedapi_mode}")
din = pw.tedapi.get_din()
power = pw.power()
firmware = pw.tedapi.get_firmware_version()

print(f"din = {din}")
print(f"power = {power}")
print(f"firmware = {firmware}")
```

erikgieseler, 2024-12-12:
Thank you, it worked!