jasonacox/pypowerwall #122: Is PW_CACHE_EXPIRE=0 not possible? (Proxy)

erikgieseler, 2024-12-23:
Is it not possible to permanently deactivate the cache with PW_CACHE_EXPIRE=0?
I have set this in the environment variables as a test, but the debug log still shows that cached values have been taken.
It also looks as if 5 seconds are cached anyway, even with a lower value.
My docker-compose.yml looks like this:

```
version: "3"

services:
  pypowerwall:
    image: jasonacox/pypowerwall:latest
    container_name: pypowerwall
    hostname: pypowerwall
    ports:
      - 8675:8675
    environment:
      - PW_PORT=8675
      - PW_HOST=192.168.91.1
      - PW_GW_PWD=123
      - PW_CACHE_EXPIRE=0
      - PW_TIMEZONE=Europe/Berlin
      - TZ=Europe/Berlin      
      - PW_DEBUG=yes
    restart: always
```

jasonacox, 2024-12-24:
Hum... There may be a bug in the proxy or the pypowerwall library. Can you check if the env variable is setting the cache config by looking at http://localhost:8675/help and find the cache setting?

Having said that, it looks like you are using TEDAPI endpoint. I highly recommend against using no-cache. You are very likely to trigger the Powerwall rate limiting, high CPU backoff error or cause other issues.

erikgieseler, 2024-12-24:
Don't worry, I don't want to deactivate the cache, but set it to 3 seconds. 
0 was just to test whether the setting is adopted or not. 
With 3 seconds, however, 5 seconds are also cached. 
2-3 seconds is not a problem for the local API.

The env variable is applied correctly, see screenshot.

[image]

jasonacox, 2024-12-26:
Thanks! Are you seeing the cache working via logs (via PW_DEBUG=yes) or is there some other way you see it not expiring the cache?  

The main code for the TEDAPI caching that could be relevant:

https://github.com/jasonacox/pypowerwall/blob/2c93ce9327d97020cb1ff78862ab9c38c8006e09/pypowerwall/tedapi/__init__.py#L306-L310

erikgieseler, 2024-12-26:
I could see this in the MQTT log.

The cache expire was set to 2 seconds, the api was called every 3 seconds. 
The MQTT log always showed 2 identical values, only from the 3rd value onwards was it different. 
The debug log also showed “Using Cached Payload”.

I have now manually changed all occurrences in the container in the file `__init__.py` from `pwcacheexpire: int = 5` to `pwcacheexpire: int = 2`. Now no more values are output twice.

It seems that the environment variable is not taken over correctly.

jasonacox, 2024-12-26:
Thanks @erikgieseler - There has to be a bug in the constructor logic. I'll work on a fix...

erikgieseler, 2024-12-27:
I just tested the changes, looks good.
Thank you!

erikgieseler, 2024-12-27:
Is there also a fixed cache for `backup_reserve_percent` from the API endpoint `/pod` or `/api/operation` somewhere in the code?
When I change the value via the app, it takes about 4-5 minutes until the set reserve value is displayed.

jasonacox, 2024-12-27:
Thanks for testing!

Yes, there was a separate config cache that needed to link to the expire settings. Hopefully latest commit fixes that.

erikgieseler, 2024-12-27:
Yes, it fixes the problem, the new `backup_reserve_percent` value now appears after 5 seconds.
Thank you!

jasonacox, 2024-12-27:
Thanks! I can't recall why we added the separate 300s cache for the config data/payload. I will run this a while to ensure it doesn't introduce some instability before merging.

jasonacox, 2024-12-27:
Test docker image: jasonacox/pypowerwall:0.12.2t67-beta

erikgieseler, 2024-12-29:
> Test docker image: jasonacox/pypowerwall:0.12.2t67-beta

Looks good, no problems for me so far

PR #123 (2024-12-31): v0.12.2 - TEDAPI Cache Expiration Fix
## v0.12.2 - Cache Expiration Fix

* Fix bug in cache expiration timeout code that was not honring pwcacheexpire setting. Raised by @erikgiesele in https://github.com/jasonacox/pypowerwall/issues/122 - PW_CACHE_EXPIRE=0 not possible? (Proxy)
* Add WARNING log in proxy for settings below 5s.
* Change TEDAPI config default timeout from 300s to 5s and link to pwcacheexpire setting.

Closes #122
closed 2024-12-31 by jasonacox