jasonacox/pypowerwall #125: Missing time_of_use_settings in Fleet API?

Matthew1471, 2024-12-28:
Hi,

Unless I missed it when I did a code search there's no way to set the tariff in this API.. but there is in the Tesla Fleet API by using `POST /api/1/energy_sites/{energy_site_id}/time_of_use_settings` (Source: https://developer.tesla.com/docs/fleet-api/endpoints/energy)

Are some endpoints missing?

I also could not find it in the Cloud API either (although of course the app does support this)

Kind Regards (and happy holidays!),
Matthew

jasonacox, 2024-12-29:
Hi @Matthew1471 you are correct. This API is not implemented, and likely many others. I would absolutely love and welcome the enhancement if you  have time to code it. Feel free to submit a PR. I will tag this as an enhancement request.

jasonacox-sam, 2026-09-06:
This was a request to expose `time_of_use_settings` (tariff/rate schedule data) from the Tesla Fleet API in pyPowerwall. Jason confirmed it's unimplemented and welcomed a PR, and @Matthew1471 later shared proof-of-concept code including Octopus Energy tariff calculation: https://github.com/Matthew1471/Tesla-API/blob/main/Python/examples/cloud_octopus_energy_set_tariff.py

Closing due to inactivity — but the PoC is a great starting point. If anyone wants to pick this up as a PR, please do, and we'll happily review.

— Sam ⚡

closed 2026-09-06 by jasonacox-sam