jasonacox/pypowerwall #127: Environmental Settings on Readme needs listing for PW_GW_PWD

psmith3, 2024-12-30:
Took a while to figure out that the PW_PASSWORD was not the same as PW_GW_PWD. When setting up Powerwall 3 TEDAPI in local mode, you need the docker command to look like this: The docker example needs an example reference to PW_GW_PWD

```
sudo docker run \
    -d \
    -p 8675:8675 \
    -e PW_PORT='8675' \
    -e PW_HOST='192.168.91.1' \
    -e PW_GW_PWD='Gateway_Password' \
    -e PW_TIMEZONE='America/Chicago' \
    -e TZ='America/Chicago' \
    -e PW_CACHE_EXPIRE='5' \
    -e PW_DEBUG='no' \
    -e PW_HTTPS='no' \
    -e PW_STYLE='clear' \
    --name pypowerwall \
    --restart unless-stopped \
    jasonacox/pypowerwall
```

jasonacox, 2024-12-30:
Thanks @psmith3 ! Yes, these were a bit outdated. I updated the docs. Let me know if this makes more sense. Feel free to suggest more or submit a PR. Thank you! 🙏 

https://github.com/jasonacox/pypowerwall/tree/main/proxy

psmith3, 2024-12-31:
Will do. Thanks!

jasonacox-sam, 2026-08-31:
This pointed out that the proxy docs didn't distinguish `PW_PASSWORD` from `PW_GW_PWD` (the gateway password needed for Powerwall 3 TEDAPI local mode). The docs were updated in response, and the reporter confirmed it made sense.

Closing since the documentation fix landed. Always happy to take further doc suggestions or PRs.

— Sam ⚡

closed 2026-08-31 by jasonacox-sam