jasonacox/pypowerwall #128: New alert: PVAC_a041_virtual_clamper_triggered

Nexarian, 2025-01-01:
Saw this in my monitoring of my system, not sure what it means or where it comes from though.

jasonacox, 2025-01-01:
Interesting. It looks like we have had `PVAC_a041_excess_PV_clamp_triggered` reported before:

https://github.com/jasonacox/pypowerwall?tab=readme-ov-file#pvac---photovoltaic-ac---solar-inverter

But not this one, so I will add it. Can you describe yours system? Also, do you see any odd power metrics happening at the time of the alert?

Nexarian, 2025-01-01:
System:
- 2x 7.6 kW Tesla Standalone Inverters (No Powerwalls)
- 20.5 kW (50x 410 Rec Alpha panels)
- 8 strings of various sizes, the smallest being 3 panels (the minimum)

As to odd metrics: Yes. The 3-panel string is dodgy right now. Sometimes the MCI works properly, sometimes it shuts down the string entirely for no apparent reason, and sometimes it does what I call "grey but producing" meaning in Tesla One the string is greyed out but power is still being reported on it. When that happens I get:

```
alerts = [
    "PVS_a018_MciStringB",
    "PVS_a027_Mci2PvVoltage",
    "PVS_a043_InactiveUnsafePvStrings"
]
```

Presumably (not a Tesla engineer and not privy to extra info): Power is still being shoved through the string even though the MCI should be open, but because it's not, it means control of the MCI on string B has been lost; ergo, "unsafe."

I have noticed if I go into Tesla One and "disable" and "enable" the inverter again, sometimes the strings will come back in a better state, it is during this cycling that I saw the alert in the issue title. So I suspect it's a startup alert related to initialization of the system, sometime that doesn't last long. HOWEVER, it might not be normal, as I also saw `PVAC_a014_PVS_disabled_relay` at the same time.

Nexarian, 2025-01-01:
One more note: When I set the timestamp to Tesla (they usually require that you send your installer a timestamp and the error codes your seeing, and it has to be a very sunny day at close to solar noon for them to even look at your case), they said MCIs getting into this state may be related to a "surge" of some sort. They didn't elaborate.

The techs are scheduled to come out sometime within the next month or two. They are slow :/

jasonacox, 2025-01-01:
Do you have the [Dashboard](https://github.com/jasonacox/Powerwall-Dashboard) graphing your alerts? I noticed that for my system, the "PVS_a027_Mci*PvVoltage" alerts happen at sundown right before "PVS_a059_MciOpen". The "PVAC_a014_PVS_disabled_relay " alert happens at sunrise. Here is my system and alerts over time:

[image]

Nexarian, 2025-01-01:
@jasonacox See discussion here: https://gist.github.com/jasonacox/91479957d0605248d7eadb919585616c#gistcomment-5364463

Without an intelligent way to query BOTH inverters, I'll only ever at best see half the picture, I'm blocked.

I've gone to war trying to figure out the proper settings to make this query mechanism possible, but alas, no dice.

jasonacox, 2025-01-01:
Ah, thanks for the reminder.  Understood... If it were me, I believe I would set up the Dashboard for at least the problematic string inverter. I have a few RPIs around so I would probably set up a dashboard for each inverter (I suppose two VMs would also work).  With some customization, the two pypowerwall containers could be set up to write to the same InfluxDB, or use Grafana to pull from two sources to provide a combined view. Ideally there is some networking wizardry that would allow it to work on the same host, but it is beyond my skill or ability to test.

Nexarian, 2025-01-01:
@jasonacox You're clearly a manager :). I will eventually give up/in on that and follow that approach, but for now I want my cake and eat it, too!

jasonacox, 2025-01-02:
Ha! That usually requires two cakes. 😜

Nexarian, 2025-01-10:
I solved it; though, it only works on Linux here's the script. Regardless, this issue was about logging the new error code, and I still don't have the proxy working to be able to do that just yet. We can follow up on the rest here: https://github.com/jasonacox/pypowerwall/issues/119

<details>
<summary>Bash Version</summary>

```
#!/bin/bash

set -ex

# Function to check if a command succeeded
check_command() {
    if [ $? -ne 0 ]; then
        echo "Error: $1"
        exit 1
    fi
}

# Function to clean up existing rules and configurations
cleanup() {
    echo "Cleaning up existing rules and configurations..."
    
    # Remove virtual IP addresses
    ip addr del 192.168.92.100/24 dev eth0 2>/dev/null || true
    ip addr del 192.168.92.101/24 dev eth0 2>/dev/null || true

    # Flush NAT table
    iptables -t nat -F
    check_command "Failed to flush NAT table"

    # Flush mangle table
    iptables -t mangle -F
    check_command "Failed to flush mangle table"

    # Remove all rules in filter table
    iptables -F
    check_command "Failed to flush filter table"

    # Remove non-default chains
    iptables -X
    check_command "Failed to delete non-default chains"

    # Remove all ip rules (except default)
    ip rule show | grep -v "from all lookup" | cut -d: -f1 | xargs -r -n1 ip rule del prio
    check_command "Failed to remove ip rules"

    # Remove routing tables
    ip route flush table 100 || true
    ip route flush table 101 || true
    check_command "Failed to flush routing tables"

    echo "Cleanup completed successfully."
}

# Main setup function
setup() {
    echo "Setting up routing and NAT rules..."

    # Enable IP forwarding
    echo 1 > /proc/sys/net/ipv4/ip_forward
    check_command "Failed to enable IP forwarding"

    # Add virtual IP addresses
    ip addr add 192.168.92.100/24 dev eth0
    check_command "Failed to add virtual IP 192.168.92.100"
    ip addr add 192.168.92.101/24 dev eth0
    check_command "Failed to add virtual IP 192.168.92.101"

    # Set up routing tables
    ip route add 192.168.91.1/32 via 192.168.1.67 dev eth0 onlink table 100
[... 228 lines cut ...]

            print("Setup completed successfully.")

        except Exception as e:
            print(f"Error during setup: {e}")

if __name__ == "__main__":
    check_root()
    
    devices = scan.scan(max_threads=256)

    inverters: List[Dict[str, str | int]] = [
        {
            "ip": f"192.168.92.{100 + i}",
            "table_id": BASE_TABLE_ID + i,
            "gateway": ip,
            "mark": BASE_MARK + i
        }
        for i, ip in enumerate([i['ip'] for i in devices])
    ]

    cleanup(inverters)
    setup(inverters)
```
</details>

closed 2025-01-11 by Nexarian