jasonacox/pypowerwall #131: Missing battery_blocks data on PW3 with Multiple Powerwalls in Local Mode

JohnJ9ml, 2025-01-13:
Version 0.12.1:

My local API calls for system_status and anything else that should show battery_blocks returns a count of zero and an empty array.

My installation has three PW3  units. 

The information was missing previously when I was running in cloud mode as well.

` "rawData" : {
    "all_enable_lines_high" : true,
    "auxiliary_load" : 0,
    "available_blocks" : 0,
    "available_charger_blocks" : 0,
    "battery_blocks" : [ ],
    "battery_target_power" : 0,
    "battery_target_reactive_power" : 0,
    "blocks_controlled" : 0,
    "can_reboot" : "Yes",
    "command_source" : "Configuration",
    "expected_energy_remaining" : 0,
    "ffr_power_availability_high" : 0,
    "ffr_power_availability_low" : 0,
    "grid_faults" : [ ],
    "grid_services_power" : null,
    "hardware_capability_charge_power" : 0,
    "hardware_capability_discharge_power" : 0,
    "instantaneous_max_apparent_power" : 0,
    "instantaneous_max_charge_power" : 0,
    "instantaneous_max_discharge_power" : 0,
    "inverter_nominal_usable_power" : 0,
    "last_toggle_timestamp" : "2023-10-13T04:08:05.957195-07:00",
    "load_charge_constraint" : 0,
    "max_apparent_power" : null,
    "max_charge_power" : null,
    "max_discharge_power" : null,
    "max_power_energy_remaining" : 0,
    "max_power_energy_to_be_charged" : 0,
    "max_sustained_ramp_rate" : 0,
    "nominal_energy_remaining" : 13150,
    "nominal_full_pack_energy" : 42450,
    "primary" : true,
    "score" : 10000,
    "smart_inv_delta_p" : 0,
    "smart_inv_delta_q" : 0,
    "solar_real_power_limit" : null,
    "system_island_state" : "SystemGridConnected"
  }
`

jasonacox, 2025-01-14:
Hi @JohnJ9ml - I suspect we need to add some logic to detect PW3 installs and pull the relevant data from tedapi.  Can you help by running a few things for me?  I don't have a PW3 to test. 

```python
# TEDAPI (full)
#
import pypowerwall
PW_GW_PWD="XXXXXXXXX"
PW_HOST="192.168.91.1"
pw = pypowerwall.Powerwall(PW_HOST, gw_pwd=PW_GW_PWD)

# Flags
print(f"mode: {pw.mode} - tedapi: {pw.tedapi_mode} - pw3: {pw.tedapi.pw3} - vitals: {pw.vitals_api}")

# Battery Blocks
print("pw.battery_blocks():")
print(pw.battery_blocks())
print()

# I believe this is what you posted above - please confirm

# System Status
print("system_status():")
print(pw.system_status()) # Is this the one missing?
print()

# TEDAPI Reported Battery blocks
print("pw.tedapi.get_battery_blocks():")
print(pw.tedapi.get_battery_blocks())
```

JohnJ9ml, 2025-01-14:
I'll confirm all of your items above:  

Yes, pw.battery_blocks() is empty.

Yes, pw.system_status() says I have zero battery blocks.

Yes, pw.tedapi.get_battery_blocks() sees them:

`python batteryBlockDebug.py
mode: local - tedapi: full - pw3: True - vitals: True
pw.battery_blocks():
{}

system_status():
{'command_source': 'Configuration', 'battery_target_power': 0, 'battery_target_reactive_power': 0, 'nominal_full_pack_energy': 42450, 'nominal_energy_remaining': 10700, 'max_power_energy_remaining': 0, 'max_power_energy_to_be_charged': 0, 'max_charge_power': None, 'max_discharge_power': None, 'max_apparent_power': None, 'instantaneous_max_discharge_power': 0, 'instantaneous_max_charge_power': 0, 'instantaneous_max_apparent_power': 0, 'hardware_capability_charge_power': 0, 'hardware_capability_discharge_power': 0, 'grid_services_power': None, 'system_island_state': 'SystemGridConnected', 'available_blocks': 0, 'available_charger_blocks': 0, 'battery_blocks': [], 'ffr_power_availability_high': 0, 'ffr_power_availability_low': 0, 'load_charge_constraint': 0, 'max_sustained_ramp_rate': 0, 'grid_faults': [], 'can_reboot': 'Yes', 'smart_inv_delta_p': 0, 'smart_inv_delta_q': 0, 'last_toggle_timestamp': '2023-10-13T04:08:05.957195-07:00', 'solar_real_power_limit': None, 'score': 10000, 'blocks_controlled': 0, 'primary': True, 'auxiliary_load': 0, 'all_enable_lines_high': True, 'inverter_nominal_usable_power': 0, 'expected_energy_remaining': 0}

pw.tedapi.get_battery_blocks():
[{'vin': '1707000-21-K--TG124300003JZH', 'min_soe': 0, 'max_soe': 100, 'type': 'Powerwall3', 'can_connection': 'can1', 'pvi_power_status': 'on', 'battery_power_status': 'on', 'enable_inverter_battery_meter': True, 'enable_inverter_solar_meter': True}, {'vin': '1707000-21-K--TG124300003DZE', 'min_soe': 0, 'max_soe': 100, 'type': 'Powerwall3Follower', 'pvi_power_status': 'on', 'battery_power_status': 'on', 'enable_inverter_battery_meter': True, 'enable_inverter_solar_meter': True}, {'vin': '1707000-21-K--TG12430000309Y', 'min_soe': 0, 'max_soe': 100, 'type': 'Powerwall3Follower', 'pvi_power_status': 'on', 'battery_power_status': 'on', 'enable_inverter_battery_meter': True, 'enable_inverter_solar_meter': True}]
`

JohnJ9ml, 2025-04-28:
I wandered through the code a bit, seeing if I could figure out how to do this one myself, but alas it escapes me at the moment.

jasonacox, 2025-06-08:
Hi @JohnJ9ml , that is possible.  Are there any other PW3 owners out there that can chime in and help?

In the meantime, can you run this? I'm curious to see if the data you are looking for is in this raw data.

```python
import pypowerwall
PW_GW_PWD="xxxxxxxxxx" # replace with your password
PW_HOST="192.168.91.1"

# Connect
pw = pypowerwall.Powerwall(PW_HOST, gw_pwd=PW_GW_PWD)
print(f"mode: {pw.mode} - tedapi: {pw.tedapi_mode}")
status = pw.tedapi.get_status()
pw3_data = pw.tedapi.get_pw3_vitals()
print(f"pw3_data: {pw3_data}")
```

JohnJ9ml, 2025-06-08:
`python issue_131.py
mode: local - tedapi: full
pw3_data: {'TEPOD--1707000-21-K--TG124300003JZH': {'alerts': ['PWS_a003_MciSelfTestCD', 'PWS_a004_MciSelfTestEF', 'PWS_a008_MciPvIrrationalDEF'], 'POD_nom_energy_remaining': 4620, 'POD_nom_energy_to_be_charged': 9590, 'POD_nom_full_pack_energy': 14210}, 'PVAC--1707000-21-K--TG124300003JZH': {'PVAC_Fout': 59.98, 'PVAC_Vout': 239.60000000000002, 'PVAC_VL1Ground': 119.80000000000001, 'PVAC_VL2Ground': 119.80000000000001, 'PVAC_Pout': 5800, 'PVAC_State': 'AcMode_GridFollowing', 'PVAC_PvState_A': 'Pv_Active', 'PVAC_PVMeasuredVoltage_A': 260, 'PVAC_PVCurrent_A': 9.4, 'PVAC_PVMeasuredPower_A': 2444.0, 'manufacturer': 'TESLA', 'partNumber': '1707000-21-K', 'serialNumber': 'TG124300003JZH', 'PVAC_PvState_B': 'Pv_Active_Parallel', 'PVAC_PVMeasuredVoltage_B': 260, 'PVAC_PVCurrent_B': 9.3, 'PVAC_PVMeasuredPower_B': 2418.0, 'PVAC_PvState_C': 'Pv_Active', 'PVAC_PVMeasuredVoltage_C': 192, 'PVAC_PVCurrent_C': 9.6, 'PVAC_PVMeasuredPower_C': 1843.1999999999998, 'PVAC_PvState_D': 'Pv_Active', 'PVAC_PVMeasuredVoltage_D': 4, 'PVAC_PVCurrent_D': 0, 'PVAC_PVMeasuredPower_D': 0, 'PVAC_PvState_E': 'Pv_Active', 'PVAC_PVMeasuredVoltage_E': 0, 'PVAC_PVCurrent_E': 0, 'PVAC_PVMeasuredPower_E': 0, 'PVAC_PvState_F': 'Pv_Active_Parallel', 'PVAC_PVMeasuredVoltage_F': 0, 'PVAC_PVCurrent_F': 0, 'PVAC_PVMeasuredPower_F': 0}, 'PVS--1707000-21-K--TG124300003JZH': {'PVS_StringA_Connected': True, 'PVS_StringB_Connected': True, 'PVS_StringC_Connected': True, 'PVS_StringD_Connected': True, 'PVS_StringE_Connected': True, 'PVS_StringF_Connected': True}, 'TEPINV--1707000-21-K--TG124300003JZH': {'PINV_Fout': 59.98, 'PINV_Vout': 239.60000000000002, 'PINV_VSplit1': 119.80000000000001, 'PINV_VSplit2': 119.80000000000001, 'PINV_Pout': 5.8, 'PINV_State': 'AcMode_GridFollowing'}, 'TEPOD--1707000-21-K--TG124300003DZE': {'alerts': ['PCH_a054_pwsDisabledMpptEnableLine'], 'POD_nom_energy_remaining': 4710, 'POD_nom_energy_to_be_charged': 9450, 'POD_nom_full_pack_energy': 14160}, 'PVAC--1707000-21-K--TG124300003DZE': {'PVAC_Fout': 59.99, 'PVAC_Vout': 239.8, 'PVAC_VL1Ground': 120, 'PVAC_VL2Ground': 119.80000000000001, 'PVAC_Pout': -920, 'PVAC_State': 'AcMode_GridFollowing', 'PVAC_PvState_A': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_A': 18, 'PVAC_PVCurrent_A': 0, 'PVAC_PVMeasuredPower_A': 0, 'manufacturer': 'TESLA', 'partNumber': '1707000-21-K', 'serialNumber': 'TG124300003DZE', 'PVAC_PvState_B': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_B': 18, 'PVAC_PVCurrent_B': 0, 'PVAC_PVMeasuredPower_B': 0, 'PVAC_PvState_C': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_C': 18, 'PVAC_PVCurrent_C': 0, 'PVAC_PVMeasuredPower_C': 0, 'PVAC_PvState_D': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_D': 16, 'PVAC_PVCurrent_D': 0, 'PVAC_PVMeasuredPower_D': 0, 'PVAC_PvState_E': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_E': 16, 'PVAC_PVCurrent_E': 0, 'PVAC_PVMeasuredPower_E': 0, 'PVAC_PvState_F': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_F': 16, 'PVAC_PVCurrent_F': 0, 'PVAC_PVMeasuredPower_F': 0}, 'PVS--1707000-21-K--TG124300003DZE': {'PVS_StringA_Connected': False, 'PVS_StringB_Connected': False, 'PVS_StringC_Connected': False, 'PVS_StringD_Connected': False, 'PVS_StringE_Connected': False, 'PVS_StringF_Connected': False}, 'TEPINV--1707000-21-K--TG124300003DZE': {'PINV_Fout': 59.99, 'PINV_Vout': 239.8, 'PINV_VSplit1': 120, 'PINV_VSplit2': 119.80000000000001, 'PINV_Pout': -0.92, 'PINV_State': 'AcMode_GridFollowing'}, 'TEPOD--1707000-21-K--TG12430000309Y': {'alerts': ['PCH_a054_pwsDisabledMpptEnableLine'], 'POD_nom_energy_remaining': 4660, 'POD_nom_energy_to_be_charged': 9290, 'POD_nom_full_pack_energy': 13950}, 'PVAC--1707000-21-K--TG12430000309Y': {'PVAC_Fout': 59.98, 'PVAC_Vout': 240.20000000000002, 'PVAC_VL1Ground': 120.2, 'PVAC_VL2Ground': 120, 'PVAC_Pout': -880, 'PVAC_State': 'AcMode_GridFollowing', 'PVAC_PvState_A': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_A': 14, 'PVAC_PVCurrent_A': 0, 'PVAC_PVMeasuredPower_A': 0, 'manufacturer': 'TESLA', 'partNumber': '1707000-21-K', 'serialNumber': 'TG12430000309Y', 'PVAC_PvState_B': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_B': 14, 'PVAC_PVCurrent_B': 0, 'PVAC_PVMeasuredPower_B': 0, 'PVAC_PvState_C': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_C': 14, 'PVAC_PVCurrent_C': 0, 'PVAC_PVMeasuredPower_C': 0, 'PVAC_PvState_D': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_D': 14, 'PVAC_PVCurrent_D': 0, 'PVAC_PVMeasuredPower_D': 0, 'PVAC_PvState_E': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_E': 14, 'PVAC_PVCurrent_E': 0, 'PVAC_PVMeasuredPower_E': 0, 'PVAC_PvState_F': 'Pv_Standby', 'PVAC_PVMeasuredVoltage_F': 14, 'PVAC_PVCurrent_F': 0, 'PVAC_PVMeasuredPower_F': 0}, 'PVS--1707000-21-K--TG12430000309Y': {'PVS_StringA_Connected': False, 'PVS_StringB_Connected': False, 'PVS_StringC_Connected': False, 'PVS_StringD_Connected': False, 'PVS_StringE_Connected': False, 'PVS_StringF_Connected': False}, 'TEPINV--1707000-21-K--TG12430000309Y': {'PINV_Fout': 59.98, 'PINV_Vout': 240.20000000000002, 'PINV_VSplit1': 120.2, 'PINV_VSplit2': 120, 'PINV_Pout': -0.88, 'PINV_State': 'AcMode_GridFollowing'}}
`

jasonacox, 2025-06-09:
Hi @JohnJ9ml I updated the battery block assembler to use vitals. Since your battery blocks do show up there, hopefully this will address the issue and all the tests above should start to work.

Would you be able to test the code? You will need to checkout the v0.13.1 branch to run the tests.

```bash
git clone https://github.com/jasonacox/pypowerwall.git
cd pypowerwall
git checkout v0.13.1
python
```

Or you can use the proxy container: jasonacox/pypowerwall:0.13.1t74-beta

JohnJ9ml, 2025-06-09:
I can confirm this is now working:  battery_blocks now returns data:

`{}jjames@crewchief:energy$ curl http://powerwall:8678/pw/battery_blocks
{"TG124300003JZH": {"Type": "", "PackagePartNumber": "1707000-21-K", "disabled_reasons": [], "pinv_state": "AcMode_GridFollowing", "pinv_grid_state": null, "nominal_energy_remaining": 3240, "nominal_full_pack_energy": 14210, "p_out": 0.72, "q_out": null, "v_out": 242.4, "f_out": 59.99, "i_out": null, "energy_charged": null, "energy_discharged": null, "off_grid": null, "vf_mode": null, "wobble_detected": null, "charge_power_clamped": null, "backup_ready": null, "OpSeqState": null, "version": null}, "TG124300003DZE": {"Type": "", "PackagePartNumber": "1707000-21-K", "disabled_reasons": [], "pinv_state": "AcMode_GridFollowing", "pinv_grid_state": null, "nominal_energy_remaining": 3370, "nominal_full_pack_energy": 14160, "p_out": 0.0, "q_out": null, "v_out": 243.8, "f_out": 59.985, "i_out": null, "energy_charged": null, "energy_discharged": null, "off_grid": null, "vf_mode": null, "wobble_detected": null, "charge_power_clamped": null, "backup_ready": null, "OpSeqState": null, "version": null}, "TG12430000309Y": {"Type": "", "PackagePartNumber": "1707000-21-K", "disabled_reasons": [], "pinv_state": "AcMode_GridFollowing", "pinv_grid_state": null, "nominal_energy_remaining": 3370, "nominal_full_pack_energy": 13950, "p_out": 0.0, "q_out": null, "v_out": 243.4, "f_out": 59.995, "i_out": null, "energy_charged": null, "energy_discharged": null, "off_grid": null, "vf_mode": null, "wobble_detected": null, "charge_power_clamped": null, "backup_ready": null, "OpSeqState": null, "version": null}} 
`