jasonacox/pypowerwall #135: New error seen in Tesla Solar Inverters: HighControlLoopLatency

Nexarian, 2025-01-23:
Title says it all. I think it means the system is lagging.

jasonacox, 2025-01-24:
One interesting thing about Tesla Powerwalls is that they use a control loop similar (or maybe identical since it is Tesla after all) to automotive loops. They use a Controller Area Network (CAN bus) which is a message-based protocol designed to allow the Electronic Control Units (ECUs) to communicate with each other in a reliable, priority-driven fashion. I'm guessing this error indicates that some heartbeat message is arriving late. It may be interesting to do some Google searching to see what causes that (presumably even in vehicles or robotics) and speculate why you are seeing it.

Fun fact, if you look at vitals (e.g. http://localhost:8675/vitals) you will see ECU type numbers (ecuType) which identify the different components in the loop.

Nexarian, 2025-01-24:
The dumb answer: This is on my inverter that is known to be dodgy and that Tesla and my installer are dragging their feet on fixing. It's probably simply a symptom of pre-failure. Part of the entire reason I'm doing this project is to prove to them that they actually need to replace it. The most obvious symptom I have of this inverter is that the Neurio meter only lasts about 30 minutes before the inverter's internal wifi card gives up and the Neurio starts beeping (This inverter nearly continually has the `SolarRGMComms alert) and Tesla One will no longer connect. When the sun is strong and the panels are not snow covered, 3 of the 4 strings connected to it are at risk of not activating (InactiveUnsafePVStrings and other alerts). However, I need to PROVE to them that I have this data.

Other potential answers:
- It occurred at 10:02 AM and lasted ~2 minutes. It is possible that snow was falling off my panels and there was a momentary surge or a new string came online (Unfortunately I only got per-string monitoring working shortly after this). It's also possible that the querying my Raspberry PI was doing was getting too intense (though unlikely).
- Transient error. Every dev's favorite way of saying "nothing to see here."

Nexarian, 2025-01-24:
From ChatGPT:

```
Failing or Misconfigured Node

In a CAN network, a single faulty node (e.g., one whose firmware has crashed or is repeatedly sending out-of-spec messages) can degrade bus performance for everyone. If one module is “babbling” on the bus or not respecting arbitration rules, it can block or delay legitimate messages.
Similarly, if one node’s CAN interface is misconfigured (e.g., wrong bitrate or sample point settings), it can introduce error frames on the bus.
```

jasonacox-sam, 2026-08-31:
This surfaced a `HighControlLoopLatency` error on Tesla solar inverters. The discussion covered the CAN-bus control loop architecture these systems use, and the reporter's conclusion was the most likely explanation: it appeared on an inverter already known to be flaky (pre-failure hardware the installer was dragging feet on), with the monitoring data helping make the case for replacement.

Closing due to inactivity. If this error shows up on healthy hardware, please re-open — it would be interesting to compare notes.

— Sam ⚡

closed 2026-08-31 by jasonacox-sam