jasonacox/pypowerwall #138: Cannot Connect to GW with Routed IP address

aesculus, 2025-02-04:
I added this to the route table where 192.168.1.117 is the physical IP address of my GW1 on my lan.

ip route add 192.168.91.1 via 192.168.1.117

But I get this error when

python -m pypowerwall tedapi

pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.12.4]
 - Connecting to https://192.168.91.1... FAILED

ERROR: Unable to connect to Powerwall Gateway 192.168.91.1 on port 443.
Please verify your your host has a route to the Gateway.

Error details: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: / (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7fbe22c59610>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))

Thoughts?

jasonacox, 2025-02-04:
> ConnectTimeoutError

That sounds like it is getting blocked by a firewall. What happens if you try this:

```bash
curl -ik https://192.168.91.1/                                                 
```

aesculus, 2025-02-04:
to show link
192.168.91.1 via 192.168.1.117 dev eth1
[~] # curl -ik https://192.168.91.1/

curl: (28) Failed to connect to 192.168.91.1 port 443 after 129295 ms: Couldn't connect to server

slyglif, 2025-02-06:
I'm seeing similar, and it was working yesterday.  Maybe they pushed a new firmware that disables it?

aesculus, 2025-02-07:
I am now on 24.44.0

slyglif, 2025-02-07:
ok, so it isn't firmware version, as I'm on 24.44.3.  I am seeing intermittent problems that don't make sense though.  I'm hoping it is just a flaky VPN in my case.  In your case, have you recently made any network changes (AP firmware updates / reboots)?

aesculus, 2025-02-07:
No changes. It's on the internal network. No VPN involved.

jasonacox, 2025-02-08:
> [~] # curl -ik https://192.168.91.1/
> curl: (28) Failed to connect to 192.168.91.1 port 443 after 129295 ms: Couldn't connect to server

There is definitely something off about your network. This curl is a basic TCP request. Something is blocking it or the route isn't set up correctly.  I wish I could help more. I'm not sure what to try.

aesculus, 2025-02-08:
Success?

`[~] # curl -ik https://192.168.91.1/
HTTP/1.1 200 OK
Cache-Control: no-cache, no-store
Content-Type: text/html
Date: Sat, 08 Feb 2025 22:47:41 GMT
Content-Length: 1085

<!doctype html><html lang="en"><head><title>Tesla Energy - Setup</title><link rel="icon" type="image/png" href="favicon-32x32.png" sizes="32x32"><link rel="icon" type="image/png" href="favicon-16x16.png" sizes="16x16"><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,minimum-scale=1,maximum-scale=1,user-scalable=no"><script>window.__INITIAL_STATE__={"configuration":{"isNew":false,"version":"24.44.0 510adfd5","hash":"510adfd585b8bd2d897b59af48d0f08e024c1aec","deviceType":"hec","isSolarPowerwall":false,"bootstrapped":true},"sitemaster":{"running":true,"connectedToTesla":true},"system":{"isFirmwareUpdating":false,"Version":""},"authentication":{"loginType":"kiosk","selectedLoginType":"kiosk","lastLoginAt":1734082813000,"toggle_auth_supported":true}}
</script><link href="/app.e111f50697fec651e57e.css" rel="stylesheet"></head><body><div id="root" style="height:100%"></div><script type="text/javascript" src="/vendor.e111f50697fec651e57e.js"></script><script type="text/javascript" src="/app.e111f50697fec651e57e.js"></script></body></html>[~] #
`

I have a crazy QNAP NAS and the route stuff from command line did not work. But went into the control panel networking and choose Route. Then added a Static Route

[image]

jasonacox, 2025-02-08:
Yes, success! You should be able to get pypowerwall working again. That's good to know about the QNAP. Thanks for updating the issue name so others can find it. I can add a note to the troubleshooting section that references this.

aesculus, 2025-02-09:
Well I can connect to the gw but when I try the 
python -m pypowerwall tedapi

I get a pw error: 
_Connecting to Powerwall Gateway 192.168.91.1
Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway_

My pw is my Tesla account pw and works fine for the old api and using the non upgraded web UI.

slyglif, 2025-02-09:
It sounds like you're using a PW2, not a PW3, so I'm not positive, but for a PW3 it's not your tesla pw that is needed, is't the password on the PW itself.  For the 2, I'm guessing it's the password printed in the gateway, or the last X digits of the serial number.-Chris

aesculus, 2025-02-09:
Yeah, 2-PW2 and GW1.

No PW printed and its supposed to be the SN of the GW (STG1....) but that did not work either. Will hunt around a bit more.

jasonacox, 2025-02-09:
@aesculus What do you use to get the Tesla One app to connect to your system? That would be the same thing for tedapi. 

I seem to recall the GW1 systems sometimes require a power toggle. I hope not.

aesculus, 2025-02-09:
I have not tried the Tesla 1 app because I think that requires an update to the API in the GW? When I access the GW from the web it asks me to upgrade and I just scroll to the bottom and load the old screen.

My PW is good now and I am afraid to step into the mire of all the PW issues people are having. 

So the TedAPI stuff only works if you update the GW to Tesla One?

aesculus, 2025-02-09:
OK. Thanks  for clearing that up before I went down the rabbit hole.

I don't have the extended vitals but otherwise rock solid now. Not going the mess with it anymore until once again, Tesla shoots itself in the foot in the name of progress.

closed 2025-02-09 by aesculus