jasonacox/pypowerwall #140: Is there a way to get site currents in the aggregate api?

felberch, 2025-02-05:
In the aggregate api for site, battery and solar currents are for all 2 phases 0ed.
      
      "i_a_current": 0,
      "i_b_current": 0,
      "i_c_current": 0,

Is the api / powerwall 3 not delivering this values?

jasonacox, 2025-02-08:
Good question @felberch  - For PW3, all data is coming from these APIs (change localhost to the host running Powerwall Dashboard):

* http://localhost:8675/tedapi/status
* http://localhost:8675/tedapi/config
* http://localhost:8675/tedapi/components

See if you can see anything that would map to those. If so, we can wire that in to aggregates. I didn't see any in the test data I was given (I don't have a PW3).

felberch, 2025-02-09:
Hi @jasonacox,

Sry it took me a while to get the TEDAPI to work. With the PW 3 it is a pain in the b… .
Running my env. With evcc and pypowerwall in docker to create the correct ip route wasn’t to easy from inside the container.

Nevertheless, it is working now. I’ll attach my PW 3 output for the above URLs.

From my point of view the currents can only be these 3 values:

- METER_X_CTA_I - current for phase A/1
- METER_X_CTB_I - current for phase B/2
- METER_X_CTC_I - current for phase C/3

[image]

I added 3 versions of status.json i created them with different load profiles.
The status_11kw.json shows the numbers with my ev charging at 11kw + other consuming devices.

[config.json](https://github.com/user-attachments/files/18725222/config.json)
[status_11kw.json](https://github.com/user-attachments/files/18725221/status_11kw.json)
[components.json](https://github.com/user-attachments/files/18725224/components.json)
[status.json](https://github.com/user-attachments/files/18725223/status.json)
[status_lowPV.json](https://github.com/user-attachments/files/18725220/status_lowPV.json)

jasonacox, 2025-02-09:
Excellent!  I'll work on getting those added.

jasonacox, 2025-02-09:
Based on my system, the METER_X represents site (grid) and METER_Y is always zero (I don't have an external pv inverter). I'm not sure where to add this information in the aggregates data. Solar?

jasonacox, 2025-02-09:
Made code changes - you can test via pulling PR #142 or using this proxy: jasonacox/pypowerwall:0.12.6t68-beta

felberch, 2025-02-09:
Hi METER_Y can be Solar yes. If the PV is directly connected at PW3 the numbers may look different.
Let me try that changes.

felberch, 2025-02-09:
I tested your beta, looks good:

  "site": {
    "last_communication_time": "2025-02-09T21:07:28+01:00",
    "instant_power": 2897,
    "instant_reactive_power": 0,
    "instant_apparent_power": 0,
    "frequency": 0,
    "energy_exported": 0,
    "energy_imported": 0,
    "instant_average_voltage": 400.3924636965382,
    "instant_average_current": 18.3665,
    "i_a_current": 6.394,
    "i_b_current": 5.745,
    "i_c_current": 6.2275,
    "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
    "timeout": 1500000000,
    "num_meters_aggregated": 1,
    "instant_total_current": 18.3665,
    "disclaimer": "voltage and current calculated from tedapi data"
  },

Can you add the the Solar currents as well?

jasonacox, 2025-02-09:
> Can you add the the Solar currents as well?

Yes! I thought I had but used the wrong variable (meter_x vs meter_y).  

Try this new one: jasonacox/pypowerwall:0.12.6t68-beta2

jasonacox, 2025-02-11:
Excellent! Thank you, @felberch

PR #142 (2025-02-11): Update Aggregates API
## v0.12.6 - Aggregates Data

* Updated aggregates call to include site current (METER_X) and external PV inverter data in solar (METER_Y). 
* Reported in Issue #140.

Closes #140
closed 2025-02-11 by jasonacox