jasonacox/pypowerwall #144: Unminified source for animation JS?

mccahan, 2025-02-15:
Hello! I'd like to make some modifications to the animation, but I can only seem to find the processed files in https://github.com/jasonacox/pypowerwall/tree/main/proxy/web . Is there a place to find the underlying source for this JS with the variable names intact?

Nexarian, 2025-03-22:
Claude can't fit the whole thing in its context window (Claude Max might be able to). I've been trying this in multiple forms and I haven't gotten anything super useful yet. I will keep trying.

mccahan, 2025-03-22:
Doesn't seem worth the effort to try to reverse-engineer it, it's an entire Tesla app just for the power flow visualization. There's all kinds of code in there for provisioning, strings in every language you can think of, etc etc. I've just been using one I made with the same visual appearance https://github.com/mccahan/power-flow

Nexarian, 2025-03-22:
Generally I'd agree with you, but what worries me is the edge cases. I'm sure it's highly optimized for all sorts of situations we wouldn't think of until we encounter it live. I think it's worth trying to pull it out and compare, if not fully reverse engineer.

mccahan, 2025-03-22:
Yeah by all means. I'm not really intending to merge in my version, in large part because it's a different build process that probably doesn't make sense in this repo, and I've got what I wanted out of it (better scaling, readable code, it being smart about requests if there's a failure or delay, an indicator that it's loading data, etc).

I did have a little scenario system to catch the cases I could think of here: https://github.com/mccahan/pypowerwall-simulator-control when I was rebuilding it. No promises it catches all the edge cases, but the juice isn't worth the squeeze for me to stress about potential edge cases in a non-critical diagram, so hopefully you can get what you need out of Claude.

Nexarian, 2025-03-22:
Depending on whether I can glean anything from the Tesla version, we should probably eventually combine forces to simplify the whole thing!

mccahan, 2025-03-22:
Whatever works! I feel like it was just lifted from some old installer app (maybe even a web server on the PWs themselves?), I saw you had mentioned an issue with session stuff elsewhere which is just a side-effect of the cruft in there that was designed to handle actual provisioning and logins, and I had extra issues if I had multiple tabs with it open. In that app.js file you'll find all kinds of things:

Scanning for WiFi networks:
[image]

Agreeing to the warranty and saving customer info:
[image]

Firmware update checks:
[image]

Nexarian, 2025-03-22:
Yeah I want to pull it out, see what's useful in the actual animation, then combine with what you have and then update.

Gene-splicing and removing the junk dna, so to speak :)

mccahan, 2025-03-22:
Out of curiosity, are the edge cases you're thinking of something besides the numbers/connections between nodes? Right now the way I have it it's just showing numbers from the csv2 endpoint, and which flows to show/hide are [computed here](https://github.com/mccahan/power-flow/blob/main/components/power-flow.tsx#L222-L228), but I'm wondering if it would meet your needs if there was an endpoint in the proxy to calculate all that so you could more easily test/tweak?

E.g. if there was some hypothetical `/dataviz` endpoint with
```json
{
    "flows": {
        "solar_to_home": true,
        "solar_to_grid": true,
        "solar_to_battery": false,

        "grid_to_battery": false,
        "grid_to_home": false,

        "battery_to_home": false,
        "battery_to_grid": false
    },
    "batteries": [0.95],
    "power": {
        "solar": 5.6,
        "grid": -5.1,
        "home": 0.5,
        "battery": 0.0
    },
  "solar_only":  false
}
```

Nexarian, 2025-03-23:
Edge cases: I'm not trying to think of them at present. Having cut my early career as a SDET at Microsoft, I know that many edge cases are the ones you don't think of! Still, it would be worthwhile to fill out a matrix of what we can think of.

Also: Great minds think alike :) I'm actually already thinking about that endpoint! It's going to be essential for my other other other side project (do I have too many?) Of using cloudflared to expose my dashboard to a public Internet url, so I can show my solar installer the errors I'm seeing, access it remotely when I'm not at home, and well, brag a little bit to friends and family :)

Right now the animation doesn't work at all when I do this (everything else does work) because it's the only part of the Grafana dashboard that requires a direct connection back to the rendering server.

mccahan, 2025-03-23:
I love looking at my dashboard way too much (@jasonacox has done a great job with so many cool things), so I'm using Cloudflare Zero Trust for SAML logins alongside Cloudflared, and Nginx Proxy Manager (which I was running anyway) to expose it to the internet. I have Grafana (with [anonymous access](https://github.com/jasonacox/Powerwall-Dashboard/pull/583) so you don't have to log in twice) on one subdomain, and then pypowerwall on another subdomain, which is what I point the iframe at. I have a third subdomain because I run my visualization without merging it into the proxy, but that's a me thing lol

With Traefik/Caddy/Nginx Proxy Manager, whatever your preference is, it's pretty straightforward to just proxy pypowerwall with the viz to a separate subdomain behind https and then it'll just work. It wouldn't be much effort to rewrite to a subdirectory at the HTTP proxy level, but it's fine to just have it on its own subdomain.

Ditto, way too many side projects. I've ended up using xbar to put live stats up in my macOS toolbar:
[image]

And then pypowerwall + a library  I found for the API of our car charger [with a custom UI](https://github.com/mccahan/chargepoint-monitor) so I can adjust the EV charge rate to just use excess solar.

I have a problem.

mccahan, 2025-03-23:
@Nexarian if you wanted a demo of how you might hook things up through cloudflared with one subdomain, check out https://github.com/mccahan/Powerwall-Dashboard/commit/e5a907bcd60bb620919bcaf957c64f55f2a643f4 with Traefik. I used my dataviz because the built-in one is absolute URLs and I'm lazy, but...

- Routes Grafana to http://localhost:8090/
- Pypowerwall proxied to e.g. http://localhost:8090/pypowerwall/vitals
- Visualization at http://localhost:8090/power-flow/ (this is what you would point your iframe to, though with your external domain you're using)

Nexarian, 2025-03-25:
@mccahan I ended up doing this: https://github.com/Nexarian/Powerwall-Dashboard/commit/380ab4aed52c9ccf36d7eab73e411d6055f04b8e

mccahan, 2025-03-25:
@Nexarian funny enough, earlier I was working on moving the static asset URLs and allowing a prefix for the animation, which might make your life easier: https://github.com/jasonacox/pypowerwall/compare/main...mccahan:pypowerwall:feature/prefix-animation-assets?expand=1

Instead of jumping through the extra hoops to separate out your static assets for Grafana vs the animation and specify all the API endpoints that the animation uses, you could `export PROXY_BASE_URL=/api8675/`

Nexarian, 2025-03-25:
Probably. The fact that I have two inverters (Ergo: Two fully independent gateways, which makes me grumpy) that I have to aggregate together complicates things.

Nexarian, 2025-03-27:
@mccahan I have some more information on the animation JS.

The "core" part of the animation comes from line 18,673 in app.js. This is the first part of it:

```
        const ht = [et.h.SOLAR, et.h.BATTERY, et.h.GRID, et.h.USAGE],
            Et = Object(u.defineMessages)({ gridButtonSpinnerAltLabel: { id: "power_flow_view_grid_spinner_alt_label", defaultMessage: "System Transitioning" } });
        class bt extends c.Component {
            static getActiveIcons(e) {
```

 I was able to use Claude to de-obfuscate a lot of it, but it still has dependencies. I think your wizardry might be able to make short work of hooking up the rest: https://github.com/Nexarian/pypowerwall/commit/dbfb331e60872d1151a047d42a5d831ea10a85fd#diff-5b9f7495d720dce1ccd1176cc634e07832413ac5568dc2a7c93b225364ade655

jasonacox-sam, 2026-08-31:
This asked for the unminified source behind the proxy web animation JS. Through the thread, the "core" animation logic was traced to around line 18,673 of `app.js`, and a feature branch adding an asset URL prefix was started — but the upstream source itself was never published, since Tesla's minified bundle is what's mirrored.

Closing due to 90+ days of inactivity. If there's renewed interest in maintaining a forkable/prefixed version of the animation assets, re-open and we'll pick it back up.

— Sam ⚡

closed 2026-08-31 by jasonacox-sam