jasonacox/pypowerwall #160: pypowerwall doesn't give error if it can't connect

snarl817, 2025-03-31:
A few weeks ago, I noticed that my login to the gateway was no longer working when I pointed a browser directly at it, but the dashboard still had stats. I figured that Tesla had updated the firmware and nuked the custom password, but the token used by the proxy was still valid. I was going to try resetting it, but there was a bad storm and I was on backup power at the time, so I really didn't want to mess with it until the grid was back up.

Today, I noticed that I no longer had stats, so I went through the procedure to connect to the gateway's internal AP with my phone, and reset the password. NOTE: The password I used is the one specified in the pypowerwall.env file.  Looking at the pypowerwall logs, they're filled with: 
[pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
A restart of the container doesn't fix it. 

I logged in with a browser, then changed the URI to https://GATEWAY-HOSTNAME/api/system_status/grid_status, and I get a JSON payload:
{"grid_status":"SystemGridConnected","grid_services_active":false}

I pulled the latest pypowerwall container image and used it, but the error persists. Note that setting PW_DEBUG=yes in the env file doesn't change anything...no debug information gets printed. Which got me curious. The env file was pointing at 172.16.1.72. I pointed my browser at that IP, and got a connection timeout. A check of DNS revealed that the gateway's IP address had changed (my fault, really...I should have pinned it to a static address in the dhcpd.conf). I ALSO noticed that the gateway had switched from Ethernet to WiFi.

The container SHOULD have logged that it couldn't connect. Instead, on startup, it logged that it was connected to 172.16.1.72 (unknown). NOW it says:
Connected to Energy Gateway 172.16.0.237 (Home Energy Gateway)
However, it DOES look like my previously reported issue where the proxy couldn't connect to the hostname is now fixed (as long as I use the FQDN...docker containers on my network can't resolve short names for some reason...probably missing a DNS suffix in the docker config), so this shouldn't be a problem going forward. I changed the hostname to the FQDN of the gateway and the proxy is able to connect...but connection errors should probably be reported as a connection timeout/connection refused/hey, this isn't a home energy gateway.

jasonacox, 2025-04-05:
Great observation. We need to address this to make it easier to troubleshoot.

jasonacox-sam, 2026-05-25:
@jasonacox — I reviewed the current code. The diagnostics exist but are all at `log.debug` level, which means users running at default (ERROR) logging never see them:

**Current behavior:**
- `ConnectionError` in `poll()` → `log.debug('ERROR Unable to connect...')` — invisible to users
- `Timeout` in `poll()` → `log.debug('ERROR Timeout...')` — invisible to users  
- `connect()` failures → `log.debug('Failed to connect using Local mode...')` — invisible to users
- The only user-visible error is the generic `log.error("Unable to connect to Powerwall.")` which says nothing about why

**What users see:** `[ERROR] Failed to get /api/system_status/grid_status` repeated endlessly, with no explanation of whether it's a network issue, password issue, token expiry, etc.

**What needs fixing:**
1. Promote connection errors in `poll()` from `debug` to `error` (at least the first occurrence, or periodically)
2. Add specific diagnostic messages: "check network connectivity", "check password", "session expired"
3. The `connect()` method should log at `warning`/`error` level, not `debug`, when modes fail

I'll put together a PR with these improvements.

— Sam 🌊