jasonacox/pypowerwall #165: pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.12.7] PW2 PASSWORD Issue

maischrot, 2025-04-16:
Hi and sorry for a potential double Question.
I used Tedapi Reader for some weeks, since a few days i cannot connect. i ran the tedapi test.

Connecting to Powerwall Gateway 192.168.91.1
Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again

I found a Post/Information to just use the last 5 digits. But that does work

Thanks for a tip.
Armin

maischrot, 2025-04-16:
Tesla Gateway
 Software 25.10

maischrot, 2025-04-16:
PW2 is Firmware exactly is 25.10.1, 
I tried pypowerwall Version 11.1, 12.1 to 12.9.

maischrot, 2025-04-16:
404 Powerwall API not found at https://192.168.179.4/api/devices/vitals
Firmware 251001 detected - Does not support vitals API - disabling.

jasonacox, 2025-04-17:
With Firmware version 25.10, Tesla has removed the ability to LAN route to TEDAPI.  The only way you can get access to that end point is to use the Powerwall WiFi access point directly (e.g. TEG-xxx or TeslaPW-xxx). 

You can set up a host with WiFi to access this for you. Here is an example: https://github.com/jasonacox/Powerwall-Dashboard/discussions/607

There is a bit more information about this issue here: https://github.com/jasonacox/Powerwall-Dashboard/issues/606

psmith3, 2025-05-23:
@jasonacox, I just updated to version 0.12.12 from 0.12.1, and my Powerwall 3 has been running 25.10.4 for a while, working just fine through TEDAPI over the ethernet, using a static route onboard my host. My communications via TEDAPI broke today, right after I pulled the latest version and upgraded. I am running on Docker on a Synology NAS. The issue of Tesla blocking access through the Ethernet using TEDAPI is not accurate, and I am now seeing the same bad Gateway password issue as everyone else. Has there been any progress made since these discussions started on April 17th? Has any of the syntax for the parameters changed with 0.12.12? It appears that the proxy documentation is behind schedule regarding the latest release.

jasonacox, 2025-07-04:
> Is it possible to connect to TEDAPI using the ethernet interface?

Unfortunately there does not appear to be a way after the Firmware 25.10.x upgrade. It specifically blocks this access.

Nexarian, 2025-11-09:
@Matthew1471 If you've cracked this you're my hero and a hero a huge number of Tesla users.

@jasonacox Red Alert.

jasonacox, 2025-11-09:
Thanks @Matthew1471 ! I'll take a look. I sure thought we tried this before for the other URIs (e.g. /tedapi/config, /tedapi/status, /tedapi/components, /tedapi/battery, /tedapi/controller) but 100% worth trying this! Thank you! 🙏 

@Nexarian precisely ❤️ ... if you or anyone else has time to work on this, please let us know what you discover. Unfortunately I have a lot of travel this week so my ability to test will be spotty.

Matthew1471, 2025-11-09:
> [@Matthew1471](https://github.com/Matthew1471) If you've cracked this you're my hero and a hero a huge number of Tesla users.
> 
> [@jasonacox](https://github.com/jasonacox) Red Alert.

Too kind 😂

> Thanks [@Matthew1471](https://github.com/Matthew1471) ! I'll take a look. I sure thought we tried this before for the other URIs (e.g. /tedapi/config, /tedapi/status, /tedapi/components, /tedapi/battery, /tedapi/controller) but 100% worth trying this! Thank you! 🙏

The official app uses `/tedapi/v1r` (Version 1 Routable Message perhaps?) to get a Common API Get Networking Status Response (and I've given it a Tesla Energy Gateway Message / TEG Message containing a TEG API Schedule Manual Backup Event Request successfully). I don't believe the individual URLs will need to exist as a Routable Message will contain what type of data is being requested.

> [@Nexarian](https://github.com/Nexarian) precisely ❤️ ... if you or anyone else has time to work on this, please let us know what you discover. Unfortunately I have a lot of travel this week so my ability to test will be spotty.

I appreciate none of you will end up using my library long term or be familiar with it (just use it as a PoC to write your own code) so I'll outline the steps:

Authentication Preparation:

1. Generate (and store!) a RSA private and public key (Key Size = 4096, Exponent = 65537) other key types may be valid but this is how the Tesla app on iOS does it.
2. Obtain JWT from Owner API (my code has a clumsy open the web browser then use something like Web Developer tools to sniff the tesla:// redirection and paste it into the app - you should probably be using Fleet API).
3. Use Owner API to work out `energy_site_id`.
4. Use Owner API (LAN route may exist but this how Tesla app does it) to give `/api/1/energy_sites/{energy_site_id}/command?language=en_GB` an `add_authorized_client_request` message with the DER encoded PKCS1 formatted public key in Base64 format.
5. Toggle a Powerwall for the key to be accepted (you can send a `list_authorized_clients_request` message to check the `state` of the key going from `1` to `3`).

API Usage:

1. Create a valid Protobuf command payload applicable to your type of message (currently known are `CommonMessages`, `TEGMessages` (Tesla Energy Gateway Messages), `WCMessages` (Wall Connector Messages), `NeurioMeterMessages`. `EnergySiteNetMessages`, `AuthorizationMessages` (probably how we will be able to later remove the Cloud component), `FileStoreMessages` and `GraphQLMessages`). For example let's pick a `CommonAPIGetNetworkingStatusRequest`, it has no fields.
2. Wrap the command object in the applicable type, e.g. a `CommonMessages`.
3. Wrap the type of message into a `MessageEnvelope`, setting `DELIVERY_CHANNEL_HERMES_COMMAND` as your `DeliveryChannel` and adding a `Participant` to the sender field of a `authorized_client` of `AUTHORIZED_CLIENT_TYPE_CUSTOMER_MOBILE_APP`, set the recipient as a `Participant` but this time with a [Device Identification Number (DIN)](https://energylibrary.tesla.com/docs/Public/More/Powerhub/Residential/UserManual/en-us/GUID-A0E703A8-D151-4323-9EEE-B1DEC2D29619.html) of the Gateway you wish to address (you may be able to address batteries too). Serialise this to bytes.
4. Wrap the bytes of the `MessageEnvelope` in a `RoutableMessage`, set the `domain` as `DOMAIN_ENERGY_DEVICE`. Set a random UUID so the API router(?) can respond back with it. Currently this is a `uuid.uuid4()` converted to bytes. Serialise this to bytes.
5. You will need to sign the `RoutableMessage` (although I am suspecting some commands might not require signature), while Tesla vehicles use other types of signature data, Tesla app on IOS uses RSA. Generate a `expires_at` epoch, Tesla app on IOS currently sets that as 12 seconds in the future. If your platform provides milliseconds (i.e. a float) convert it to an `int` as it needs to be a `fixed32`.. the Tesla app on IOS rounds up any milliseconds to the nearest second (i.e. `ceil` not `int(`). The signature data will not be just the message envelope payload - it will be a combination of fields encoded as Type Length Value (TLV). Generate the bytes of TAG_SIGNATURE_TYPE=SIGNATURE_TYPE_RSA, TAG_DOMAIN=DOMAIN_ENERGY_DEVICE, TAG_PERSONALIZATION=din, TAG_EXPIRES_AT=expires_at,TAG_END=message_envelope_bytes and use this as the data to sign. Sign a SHA512 hash of this data with PKCS1v15 padding (you may need to hash the data first yourself if your library doesn't support automatically hashing the data for you on request). Then append to the `RoutableMessage` a `signature_data` of type `SignatureData` with a `signer_identity` of `KeyIdentity` and with the field `public_key` containing the previously authorised public key (DER encoding, PKCS1 format) and `rsa_data` of `RsaSignatureData` containing the fields `expires_at` and `signature`.
6. HTTP POST to `/tedapi/v1r` e.g. `https://192.168.0.100/tedapi/v1r` with the raw serialised Protobuf bytes or you can also send using Owner API URL `/api/1/energy_sites/{energy_site_id}/device_command` with a JSON object of `{"data": {"routable_message": "<<BASE64_SERIALISED_PROTOBUFF>>", "identifier_type": 1, "target_id": "<<DIN>>"}}`. The response that comes back will be a RoutableMessage but it will not be signed.

I spotted in my code I was adding the public key twice so I'll correct that. My code will probably undergo a lot of cleaning up over the next few days.. and I also will need to add another couple of hundred of Protobuf definition files.. including for vitals.. at the moment I've been focusing on "Max Backup" (aka. Storm Watch). Some of this was explained at the [vehicle-command](https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md) project.. but I didn't find the explanation straightforward and they use different keys and different signatures.

jasonacox, 2025-11-10:
Thank you @Matthew1471 !  I'll have a PW2 too so I will need to mock up a stand alone version for our PW3 owners to test.  High level thinking of a standalone script that....

Registers a Public Key via Tesla Owner API

1. Generate RSA keys or load existing ones
2. Tesla OAuth login
3. Find your energy sites/Powerwalls
4. Register the public key
5. Save configuration for later use

Physical Pairing

1. Power cycle (turn off/on) your Tesla Powerwall
2. The key will transition from status 1 (pending) to 3 (authorized)
3. Check the status using `list_authorized_clients_request`

Uses A Protobuf Client

```python
from protobuf_demo import TeslaPowerwallProtobufClient

client = TeslaPowerwallProtobufClient(
    host="192.168.1.100",
    din="1152100-13-J--AA123456B7C89D", 
    private_key_pem="<base64_encoded_private_key>",
    public_key_der="<base64_encoded_public_key>"
)

result = client.get_system_info()
```

Matthew1471, 2025-11-14:
There's been a lot of refactoring so my code should be easier to understand now.

It's also looking good for being able to pair locally:

- [protobuf/energy_device/v1/authorization_messages.proto](https://github.com/Matthew1471/Tesla-API/blob/main/Documentation/Protobuf/tesla_api/protobuf/energy_device/v1/authorization_messages.proto)
- [protobuf/energy_device/v1/authorized_state.proto](https://github.com/Matthew1471/Tesla-API/blob/main/Documentation/Protobuf/tesla_api/protobuf/energy_device/v1/authorized_state.proto)

```
def generate_sample_message3(private_key, public_key_bytes, gateway_din):
    # Build a Authorization API List Authorized Clients Request.
    list_authorized_clients_request = authorization_api_list_authorized_clients_request_pb2.AuthorizationAPIListAuthorizedClientsRequest()

    # Build a AuthorizationMessages containing the AuthorizationAPIListAuthorizedClientsRequest.
    authorization_message = authorization_messages_pb2.AuthorizationMessages(
        list_authorized_clients_request=list_authorized_clients_request
    )

    # Build the MessageEnvelope containing the AuthorizationMessage.
    message_envelope = message_envelope_pb2.MessageEnvelope(
        delivery_channel=delivery_channel_pb2.DELIVERY_CHANNEL_HERMES_COMMAND,
        sender=participant_pb2.Participant(
            authorized_client=authorized_client_type_pb2.AUTHORIZED_CLIENT_TYPE_CUSTOMER_MOBILE_APP
        ),
        recipient=participant_pb2.Participant(
            din=gateway_din
        ),
        authorization=authorization_message
    )

    # Build the RoutableMessage containing the MessageEnvelope.
    routable_message = routable_message_pb2.RoutableMessage(
        to_destination=destination_pb2.Destination(
            domain=domain_pb2.DOMAIN_ENERGY_DEVICE
        ),
        protobuf_message_as_bytes=message_envelope.SerializeToString(),
        uuid=str(uuid.uuid4()).encode()
    )

    # Sign the message.
    sign_message(private_key, public_key_bytes, gateway_din, routable_message)

    # Return the routable message.
    return routable_message
```

Matthew1471, 2025-11-15:
> [@Matthew1471](https://github.com/Matthew1471) I started trying this tonight, and I can get a paired device and a key, my credentials file looks like this after running `tesla_manage_paired_devices.py`
> 
> ```
>     "tesla": {
>         "token": {
>             "current": "...",
>             "refresh": "...",
>             "refresh_expiry": 1111111
>         },
>         "energy_site_id": "...",
>         "gateway_din": "...",
>         "paired_device": {
>             "private_key": "...",
>             "public_key": "..."
>         }
>     }
> ```
> 
> However, it doesn't seem to be working for me after running `tesla_managed_paired_devices.py`
> 
> ```
>   File "~/Tesla-API/Python/src/tesla_api/cloud/owner_api.py", line 98, in api_call
>     raise ValueError(f'{response.status_code}: {response.reason}')
> ValueError: 400: Bad Request
> ```
> 
> After running `protobuf_demo.py`
> 
> ```
> {
>   "common": {
>     "error_response": {
>       "status": {
>         "code": 7,
>         "message": "v1r: client not authorized"
>       }
>     }
>   }
> }
> ```
> 
> This is continuing to make me worried that the problem is that my regular solar inverters are locked out of this feature :/

Interesting you're getting a proper protobuf response back (and a HTTP 400 status code) from the v1r endpoint too with the error correctly structured.

Stupid question but did the key change to "state" "3" when you did the pairing (for us powerwalls we have to do something on the device within a certain time for it to take it otherwise it will change to a timeout state)? If you can't remember the easiest way to check would be in the Tesla app does your key show up in the list of paired phones (Settings->Phone Pairing, although I think you said you don't get that screen?). 

If you don't have the Phone Pairing screen in your Tesla app the other option could be to add my "generate_sample_message3" code above to "protobuf_demo.py" and instead of calling generate_sample_message2.. tell it to send generate_sample_message3 (you'll need to set the constants at the top FAKE_DATA = False and SEND_DEMO = True.. do not share any of the request or DIN of those as it will contain your real IDs).

Remember the pairing app doesn't (currently) bother to check if the pairing was successful before storing it, it just issues the command.

Matthew1471, 2025-12-15:
Just an update I have made significant usability, performance and readability improvements.

- [cloud_manage_paired_devices.py](https://github.com/Matthew1471/Tesla-API/blob/main/Python/examples/cloud_manage_paired_devices.py) : Allows you to pair/list/unpair a device (just like the official app).
- [gateway_octopus_energy_max_backup.py](https://github.com/Matthew1471/Tesla-API/blob/main/Python/examples/gateway_octopus_energy_max_backup.py) : Lets you trigger via LAN API a Max Backup session based off Octopus Energy's Planned Dispatch times (if you're not with Octopus Energy still useful to see the code).
- [protobuf_demo.py](https://github.com/Matthew1471/Tesla-API/blob/main/Python/examples/protobuf_demo.py) : Simple Protobuf demo (with optional LAN API demo).

I need to flesh out more protobuf definitions but I am running out of things I can change to my Max Backup code.. that's about as production ready as I can get it now.

As for being able to pair locally, I haven't figured out how to initiate the initial pairing soley via LAN (I'd need to watch the One app a bit more to figure out how it's doing auth - you can see all the functionality I can get it to do locally at [gateway_manage_paired_devices.py](https://github.com/Matthew1471/Tesla-API/blob/main/Python/examples/gateway_manage_paired_devices.py)), however once you have a paired key all other commands can be done over LAN.

There's also [cloud_octopus_energy_set_tariff.py](https://github.com/Matthew1471/Tesla-API/blob/main/Python/examples/cloud_octopus_energy_set_tariff.py) to show how to set tariffs and some cloud_* versions of the LAN based gateway_* versions above.

Nexarian, 2026-02-15:
@Matthew1471 Your core insight: That Tesla did not explicitly block the LAN static route and this was more an auth change, appears true. I could not figure out a way to get the device registration in your workflow to work with my solar inverters, it gets stuck on waiting for a "HERMES" command from the cloud. HOWEVER

@jasonacox I believe I have found a workaround to this issue that should *fingers crossed* work on everything. We have LAN back :). To prove this, I disabled the wifi adapters on my raspberry pi and I was still able to send and receive TEDAPI commands.

The change was from "basic" auth to a more comprehensive "bearer" auth on the 192.168.91.1 endpoint. I believe this was an unintentional regression from the addition of offline mode for the Powerwalls.

There is going to be some work to integrate this into pypowerwall, but if you understand what's necessary to apply this change, I'd appreciate an affirmative/negative as to whether it works for you or not (Such as you @erikgieseler). I will be following up with PRs to integrate this within the next few days.

<details>

<summary> Relevant code snippets </summary>

```
# IP here is the NAT'ed IP to 192.168.91.1 or that IP itself.
session = requests.Session()
session.verify = False
resp = session.post(
    f"https://{ip}/api/login/Basic",
    json={
        "username": "installer",
        "password": pwd,
        "email": "installer@tesla.com",
        "clientInfo": {"timezone": "America/Los_Angeles"},
    },
    timeout=1,
)
if resp.ok:
    token = resp.json().get("token", "")
```

```
def send_message(self, pb_message: tedapi.Message, _retried: bool = False) -> tedapi.Message:
    """Send TEDAPI message wrapped in AuthEnvelope, authenticated with Bearer token."""
    if not self.token:
        self.login()

    url = f"{self.base_url}/tedapi/v1"

    # Wrap in AuthEnvelope (required even with Bearer token)
    auth_env = tedapi.AuthEnvelope()
    auth_env.payload = pb_message.message.SerializeToString()
    auth_env.externalAuth.type = tedapi.EXTERNAL_AUTH_TYPE_PRESENCE
    data = auth_env.SerializeToString()

    response = self.session.post(
        url,
        data=data,
        headers={
            "Authorization": f"Bearer {token}",
            "Content-Type": "application/octet-stream",
        },
        timeout=self.timeout,
    )
```
</details>

<details>

<summary> And you may need this updated protobuf definition </summary>

```
// Tesla TEDAPI Protocol Buffer definition
//
// Create Python bindings:
//     protoc --python_out=. tedapi.proto
// Or:
//     python -m grpc_tools.protoc --python_out=. --proto_path=. tedapi.proto
//
// Date: 12 Feb 2026

[... 555 lines cut ...]
  FileStoreAPIFile file = 1;
  bytes hash = 2;
}

message FileStoreAPIUpdateFileRequest {
  FileStoreAPIDomain domain = 1;
  FileStoreAPIFile file = 2;
  bytes hash = 3;                      // Previous hash for optimistic concurrency
}

message FileStoreAPIUpdateFileResponse {
  FileStoreAPIFile file = 1;
  bytes hash = 2;
}

message FileStoreAPIForceWriteFileRequest {
  FileStoreAPIDomain domain = 1;
  FileStoreAPIFile file = 2;
}

message FileStoreAPIForceWriteFileResponse {
  bytes hash = 1;
}
```
</summary>

jasonacox, 2026-02-15:
Thanks @Nexarian !

I created a test script I stored in https://github.com/jasonacox/pypowerwall/tree/main/tools/tedapi-lan :

SETUP PHASE (Cloud - one time):

1. Authenticate with Tesla via OAuth (using pypowerwall's TeslaPy)
2. Query Tesla Cloud API to discover your Powerwall gateway details
3. Generate RSA-4096 key pair locally
4. Register public key with gateway via Tesla Cloud API
5. Physical confirmation required: toggle any Powerwall switch OFF/ON
   within 30 seconds to approve the pairing
    
OPERATION PHASE (LAN - ongoing):

6. Build TEDAPI protobuf messages locally
7. Sign messages with RSA private key
8. Send signed requests directly to Powerwall's local IP via HTTPS
9. Receive and parse responses

The problem is that while I get 200 responses back for the /tedapi/v1r calls to my Powerwall 3 (indicating the auth was valid), I can't seem to get the config payload. I'll try others, but feel free to play wit this.

```
$ python lan_tedapi.py

======================================================================
Tesla Powerwall LAN-Based TEDAPI Access
======================================================================
Setup: OAuth → Cloud Pairing → Physical Confirmation
Operation: RSA-Signed Messages → Direct LAN Communication
======================================================================

Tesla account email: jason@jasonacox.com
Powerwall IP address (e.g., 192.168.91.1) [192.168.91.1]: 

[Phase 1] Tesla OAuth Authentication
----------------------------------------------------------------------
Authenticating with Tesla account: email@email.com

======================================================================
Open this URL in your browser:
======================================================================
https://auth.tesla.com/{xxxxxxxxxxxxxxxxxx}
======================================================================

Paste the 'Page Not Found' URL:  {xxxxxxxxxxxxxxxxxx}
✓ Successfully authenticated!
✓ Token refreshed (valid for 8 hours)

[Phase 2] Device Pairing
----------------------------------------------------------------------
✓ Loaded credentials from credentials.json
  DIN: 1700000-00-A--TG1234567890TB
  Keys: RSA-4096
✓ Found existing paired device
Use existing keys? [Y/n]: 

[Phase 3] Send Signed TEDAPI Request
----------------------------------------------------------------------

Building signed config request...
✓ MessageEnvelope: 57 bytes
✓ RoutableMessage UUID: 5d3330c3-ba3b-4a35-9000-cda1abbd7cf2
✓ Signed (512 byte signature)
✓ Payload: 1160 bytes

Sending to https://192.168.91.1/tedapi/v1r...
✓ Status: 200
✓ Parsed RoutableMessage (0 bytes)
✓ Parsed MessageEnvelope
  - No filestore in response

✗ No config data found in response

======================================================================
✓ Complete! TEDAPI access working with RSA signatures
======================================================================
```

jasonacox, 2026-02-16:
Thanks @Nexarian  - I tried routed and direct. Unfortunately it seems that for the Powerwall 3, it doesn't accept that login (401 login failed). At first, it would timeout after 10s. I changed the timeout in your code to 30s and finally got a response, but it was the 401, bad credentials. I'm using the QR password from the Powerwall 3 itself (same I use for WiFi access point login).

For anyone else who wants to try - you will need all 3 files:

```bash
curl -O https://raw.githubusercontent.com/Nexarian/tesla-solar-reporter/refs/heads/main/pvi_status/pvi_control_bearer.py
curl -O https://raw.githubusercontent.com/Nexarian/tesla-solar-reporter/refs/heads/main/pvi_status/pvi_static_rules.py
curl -O https://raw.githubusercontent.com/Nexarian/tesla-solar-reporter/refs/heads/main/pvi_status/tedapi_pb2.py
```

Example run (replace password with your PW QR / WiFi password):

```
$ python pvi_control_bearer.py --host 192.168.91.1 --password ABCDEFGHIJK device-status
Traceback (most recent call last):
  File "/Users/jason/Code/pypowerwall/tools/tedapi-lan/pvi_control_bearer.py", line 1122, in <module>
    main()
  File "/Users/jason/Code/pypowerwall/tools/tedapi-lan/pvi_control_bearer.py", line 1118, in main
    args.func(args)
  File "/Users/jason/Code/pypowerwall/tools/tedapi-lan/pvi_control_bearer.py", line 945, in cmd_device_status
    api = make_api(args)
  File "/Users/jason/Code/pypowerwall/tools/tedapi-lan/pvi_control_bearer.py", line 822, in make_api
    api.login()
  File "/Users/jason/Code/pypowerwall/tools/tedapi-lan/pvi_control_bearer.py", line 95, in login
    raise Exception(f"Login failed: HTTP {resp.status_code} - {resp.text[:200]}")
Exception: Login failed: HTTP 401 - {"code":401,"error":"bad credentials","message":"Login Error"}
```

Nexarian, 2026-02-16:
@jimmyfitz You have to recompile your protobuf for your GRPC version. You can update it with this: `python3 -m grpc_tools.protoc --python_out=. --proto_path=. tedapi.proto`

@jasonacox Ugh, that's frustrating! From NetZero though it looks like perhaps the firmware paths have started to diverge. It's possible that the Powerwall 3 requires a device to be paired for offline mode as @Matthew1471 described and doesn't accept the basic auth as my inverters do.

jasonacox, 2026-02-16:
Hi @jimmyfitz , what @Nexarian  said, but  you can also just upgrade your python lib for protobuf.

```bash
pip install -U "protobuf>=6,<7"
```

@Nexarian  I know. 😦  Was hoping. The code in lan_tedapi.py test that does the signing and RSA process, works in the sense of letting me in (fast response) accepting the HTTP request and parsing the signed message, but it is rejecting it at the authorization layer (hence “200 + empty payload” with signed_message_status.message_fault = 2).

Nexarian, 2026-02-16:
One of the biggest changes to the pvi_control_bearer.py script versus what PyPowerwall does is the AuthEnvelope wrapper. I might play with that and see what AI assistants can come up with with the APK

What appears to be clear: Tesla didn't outright block LAN, they fiddled with the auth protocols. Knowing that, I suspect it's possible to claw this back, but I don't have a PW, so it's hard for me to debug this from my side.

jimmyfitz, 2026-02-16:
```
(.venv) dashboard@power:~$ python3 pvi_control_bearer.py --host 192.168.91.1 --password ABC123456 device-status
Traceback (most recent call last):
  File "/home/dashboard/pvi_control_bearer.py", line 1116, in <module>
    main()
  File "/home/dashboard/pvi_control_bearer.py", line 1112, in main
    args.func(args)
  File "/home/dashboard/pvi_control_bearer.py", line 945, in cmd_device_status
    api = make_api(args)
          ^^^^^^^^^^^^^^
  File "/home/dashboard/pvi_control_bearer.py", line 822, in make_api
    api.login()
  File "/home/dashboard/pvi_control_bearer.py", line 95, in login
    raise Exception(f"Login failed: HTTP {resp.status_code} - {resp.text[:200]}")
Exception: Login failed: HTTP 401 - {"code":401,"error":"roles [Provider_Engineer] require physical access","message":"Login Error"}

```

Nexarian, 2026-02-16:
Right. I'm saying in addition to the ethernet, connect it to the wifi. Tesla is obsessed with "local presence" and a wifi connection might provide that

jasonacox, 2026-02-16:
@Nexarian  - yes, we need you to get a PW. 🤣 

FWIW - My Powerwalls are not connected to WiFi, I also have a hardline ethernet cable to them. The error I get is different than @jimmyfitz  but he has PW2 and I have PW3.

jimmyfitz, 2026-02-16:
I tried connecting to Wifi with no success either.  Same error.  I can get to the web page on both IPs but the script still fails.

I'm not sure if it's relevant, but right at the moment my Powerwall is currently in "Remote Service Access" mode as I recently had them investigating something relating to my solar.

jasonacox, 2026-02-16:
Thanks for helping us test this @jimmyfitz - I tried toggling the Remote Service Access too.  No difference.

Also, I tried connecting directly to the Powerwall WiFi access point (no routing, direct) and I get the same "login error".

Nexarian, 2026-02-16:
@jasonacox I pushed another script from my experimenting. [register_local_client.py](https://github.com/Nexarian/tesla-solar-reporter/blob/main/pvi_status/register_local_client.py) -- It's a different attempt to do what your script did. Instructions are in the comments at the top.

I asked Claude Opus 4.5 what differences there might be. I told it your name, I hope you don't mind :)

I could never get my key out of "PENDING" (Waiting on HERMES command) status, which is what drove me to find the bearer token solution. You may have more luck because your powerwall is meant to do this.

Hopefully a breadcrumb or two in here to try to get it working?

```
Jason's Code vs Ours — Key Differences
1. CRITICAL: Not checking signed_message_status (THE BUG)
Jason's parse_tedapi_response only checks protobuf_message_as_bytes:

# Jason's code - ONLY checks payload
resp_routable = pb.RoutableMessage()
resp_routable.ParseFromString(response_content)
print(f"✓ Parsed RoutableMessage ({len(resp_routable.protobuf_message_as_bytes)} bytes)")
# ← never checks signed_message_status!
Our code checks for errors first:

# Our code - checks error status
if routable.signed_message_status.operation_status == vc.OPERATIONSTATUS_ERROR:
    fault = vc.MessageFault_E.Name(routable.signed_message_status.signed_message_fault)
    raise Exception(f"Signed message error: {fault}")
The 0-byte protobuf_message_as_bytes means the gateway returned an error in signed_message_status, not a payload. Jason's code silently discards the error.

2. CRITICAL: Incomplete MessageStatus proto
Jason's proto:

message MessageStatus {
  MessageFault_E message_fault = 1;  // WRONG - only 1 field, wrong type
}
Our proto (correct):

message MessageStatus {
    OperationStatus_E operation_status     = 1;  // 0=OK, 1=WAIT, 2=ERROR
    MessageFault_E    signed_message_fault = 2;  // actual error code
}
Jason is missing field 2 entirely — the signed_message_fault that tells you what went wrong (e.g., UNKNOWN_KEY_ID, INVALID_SIGNATURE, TIME_EXPIRED, INCORRECT_EPOCH).

3. CRITICAL: MessageFault_E only has 1 value
Jason's proto:

enum MessageFault_E {
  MESSAGEFAULT_ERROR_NONE = 0;
}
Our proto has 24 values including the critical diagnostic ones:

MESSAGEFAULT_ERROR_UNKNOWN_KEY_ID = 3 — key not recognized/not VERIFIED
MESSAGEFAULT_ERROR_INVALID_SIGNATURE = 5 — signing error
MESSAGEFAULT_ERROR_INCORRECT_EPOCH = 15 — time sync issue
MESSAGEFAULT_ERROR_TIME_EXPIRED = 17 — signature expired
MESSAGEFAULT_ERROR_WRONG_PERSONALIZATION = 12 — wrong DIN in TLV
4. MINOR: Missing gzip decompression
Jason doesn't decompress gzip responses. Our code does:

if len(raw) >= 2 and raw[0] == 0x1f and raw[1] == 0x8b:
    raw = gzip.decompress(raw)
5. Everything else matches
The core logic is identical:

Same MessageEnvelope construction (field numbers, delivery channel, sender/recipient)
Same TLV payload format for signing
Same RSA SHA-512 PKCS1v15 algorithm
Same RoutableMessage field layout (to_destination=6, payload=10, signature_data=13, uuid=51)
Same fixed32 expires_at in RsaSignatureData
The Fix for Jason
The error is almost certainly visible in signed_message_status. He needs to add this before the payload check:

# Check for signed message errors FIRST
status = resp_routable.signed_message_status
if status.operation_status == 2:  # OPERATIONSTATUS_ERROR
    print(f"✗ Signed message ERROR!")
    print(f"  operation_status: {status.operation_status}")
    print(f"  signed_message_fault: {status.signed_message_fault}")
    # Fault codes: 3=UNKNOWN_KEY, 5=INVALID_SIG, 15=INCORRECT_EPOCH, 17=TIME_EXPIRED
And update his MessageStatus proto to have both fields + the full MessageFault_E enum. The fault code will tell us exactly why the gateway rejected the signed message — most likely the key isn't VERIFIED yet or there's a time sync issue.
```

Nexarian, 2026-02-16:
@Matthew1471 Have you been able to run other queries like status or get config?

Matthew1471, 2026-02-17:
> Thanks @Matthew1471 - it would be good to know if you have been able to pull any local read-only config or status data. I'm wondering if the pairing is only used for actions (setting mode for "disconnect" scenarios) or if it can also read data.
> 
> > also suggest we break out the proto into their own files
> 
> I agree. However, at this point, I'm looking for simple tests to see if there is anything here worth pursuing and don't want to build a proto build pipeline just to test. 😉 

My example includes reading the IP address of the Gateway - I will pull more stats and config though.

When the app is paired it allows stats access when offline and the ability to trigger go off grid.. I've also noticed when the app is paired it bypassed Tesla cloud for local stats. The app makes a GraphQL query to the Powerwall for its current stats.. I dumped it out into a text file but haven't yet incorporated the GraphQL samples in the repo yet..

It's just a case of making my repo list all the proto files and document common queries.. I'm not expecting there to be any limitations that either the app or installer app doesn't have.

jasonacox, 2026-02-20:
> Then run `python3 ./example.py` from the repo clone directory (you may have to set up a virtualenv to install some dependencies)

Thanks, @Nexarian  - Sadly, doesn't work for the Powerwall 3.  I then connected direct to access point using the edited example.py and it does work (as expected).  I'm gald your use case works, but I suspect the Powerwall users will need to do the pairing as @Matthew1471 mentions.

@Matthew1471 I'll try to MiM capture some of those. Will be key to test. I'm still debating if this is easier or more complicated for community users. Of course, those with weak or fragile WiFi would prefer it. :)

Nexarian, 2026-02-24:
@arothian There are several descriptions and code samples from @Matthew1471 and myself earlier in the thread, try them! I think for the RSA key approach Jason was really close.

Looking at my bearer solution approach, it appears that this might be what the "local" algorithm was already using.

kentsimon50, 2026-02-25:
@Nexarian  I have a powerwall 2, gateway 2, and  tesla inverter. I have put in a router in client mode on my lan to connect to the powerwall 2/gateway and a route to 192.168.91.1. I have pypowerwall-server running and working beautifully with powerwall-dashboard. I also have a the powerwall/gateway connected on my lan and the inverter connected by wifi on my lan not the AP.

I can connect to the inverter with pvi_control_bearer.py on the lan ip and the wifi lan ip (see below)

I have tried to connect to the inverter with example.py to the wifi lan ip, to lan ip (I connected it up), and on the AP (192.168.91.1). It's always denied.(see below)

I can connect to the powerwall with example.py to the lan ip.(see below)

I really would like to be able to get the string-level data (individual MPPT voltages and currents) out of the inverter. Is this possible?
I'm trying to understand all this but I think you are saying that with this merge I can get rid of the client mode router. I this correct?

pi@oowerwall-desktop:~/pypowerwal$ python3 pvi_control_bearer.py --host 192.168.3.164 --password ABCDEFG device-status
Token acquired (88 chars, opaque)
DIN: 1538000-45-A--GF22301500011A

System time: 2026-02-25T08:18:16-07:00
Sitemanager running: True

Grid: OK  Contactor: CLOSED  MicroGrid: OK
Island mode: BACKUP

Meter Aggregates:
     BATTERY:      0.0 W
        SITE:      0.0 W
        LOAD:      0.0 W
       SOLAR:    190.0 W
   SOLAR_RGM:      0.0 W
   GENERATOR:      0.0 W
   CONDUCTOR:      0.0 W

PV Inverters: 1
  1538100-01-F--ADU22362L00049: OK

Active Alerts (5):
  - OpticasterRetry
  - SystemConnectedToGrid
  - FWUpdateSucceeded
  - GridCodesWrite
  - IslandChecksFailed

**INVERTER**
pi@oowerwall-desktop:~/pypowerwal$ python3 ./example.py                         
Connecting to Powerwall using tedapi mode...
Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway
Access Denied: Check your Gateway Password
Traceback (most recent call last):
  File "/home/pi/pypowerwal/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/lib/python3/dist-packages/requests/cookies.py", line 328, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/lib/python3/dist-packages/requests/cookies.py", line 399, in _find_no_duplicates
    raise KeyError('name=%r, domain=%r, path=%r' % (name, domain, path))
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/pypowerwal/./example.py", line 74, in <module>
    print("Site Name: %s - Firmware: %s - DIN: %s" % (pw.site_name(), pw.version(), pw.din()))
  File "/home/pi/pypowerwal/pypowerwall/__init__.py", line 511, in site_name
    payload = self.poll('/api/s

pi@oowerwall-desktop:~/pypowerwal$ python3 ./example.py
Connecting to Powerwall using tedapi mode...
Site Name: My Home - Firmware: 26.2.0 87adb89e - DIN: 1232100-00-H--GF222351000123
System Uptime: 151h43m48.000072807s

Battery power level: 44%
Combined power metrics: {'site': -8, 'solar': 129.75999450683594, 'battery': 390, 'load': 513.2799987792969}
[... 12 lines cut ...]
Vitals: None

String Data: {}

System Status: {'command_source': 'Configuration', 'battery_target_power': 376.25, 'battery_target_reactive_power': 0, 'nominal_full_pack_energy': 13898, 'nominal_energy_remaining': 6080, 'max_power_energy_remaining': 0, 'max_power_energy_to_be_charged': 0, 'max_charge_power': 5000, 'max_discharge_power': 5000, 'max_apparent_power': 5000, 'instantaneous_max_discharge_power': 8840, 'instantaneous_max_charge_power': 7000, 'instantaneous_max_apparent_power': 5760, 'hardware_capability_charge_power': 0, 'hardware_capability_discharge_power': 0, 'hardware_capability_adjusted_charge_power': 0, 'grid_services_power': 0, 'system_island_state': 'SystemGridConnected', 'available_blocks': 1, 'available_charger_blocks': 0, 'battery_blocks': [{'Type': '', 'PackagePartNumber': '3012170-10-C', 'PackageSerialNumber': 'TG1220010005WH', 'disabled_reasons': [], 'pinv_state': 'PINV_GridFollowing', 'pinv_grid_state': 'Grid_Compliant', 'nominal_energy_remaining': 6080, 'nominal_full_pack_energy': 13898, 'p_out': 390, 'q_out': -10, 'v_out': 246.70000000000002, 'f_out': 60.021, 'i_out': -9.5, 'energy_charged': 9120646, 'energy_discharged': 8044388, 'off_grid': False, 'vf_mode': False, 'wobble_detected': False, 'charge_power_clamped': False, 'backup_ready': True, 'OpSeqState': 'Active', 'version': '87adb89e000f2a'}], 'ffr_power_availability_high': 5800, 'ffr_power_availability_low': 5800, 'load_charge_constraint': 0, 'max_sustained_ramp_rate': 1250000, 'grid_faults': [], 'can_reboot': 'Yes', 'smart_inv_delta_p': 0, 'smart_inv_delta_q': 0, 'last_toggle_timestamp': '2026-02-19T00:11:21.675165-07:00', 'solar_real_power_limit': 1.7976931348623157e+308, 'score': 10000, 'blocks_controlled': 1, 'primary': True, 'auxiliary_load': 0, 'all_enable_lines_high': True, 'inverter_nominal_usable_power': 5800, 'system_available_charge_power_design_pf': 5000, 'system_available_discharge_power_design_pf': 5000, 'system_available_charge_power_unity_pf': 5000, 'system_available_discharge_power_unity_pf': 5000, 'system_charge_power_capability_design_pf': 0, 'system_discharge_power_capability_design_pf': 0, 'system_charge_power_capability_unity_pf': 0, 'system_discharge_power_capability_uni [... cut ...]
Battery Capacity: 13.90 kWh
Battery Energy Remaining: 6.08 kWh
Max Charge Power: 5.00 kW
Max Discharge Power: 5.00 kW
Grid Status: SystemGridConnected
Available Battery Blocks: 1

Battery Blocks:
 Block 1:
   Serial: TG1220010005WH
   Nominal Full Pack Energy: 13.90 kWh
   Nominal Energy Remaining: 6.08 kWh
   Power Output: 390 W
   Voltage Output: 246.70000000000002 V
   Frequency Output: 60.021 Hz
   Inverter State: PINV_GridFollowing
   Grid State: Grid_Compliant
   Backup Ready: True
   Version: 87adb89e000f2a

Nexarian, 2026-02-25:
> I really would like to be able to get the string-level data (individual MPPT voltages and currents) out of the inverter. Is this possible?

@kentsimon50 This thread is mostly discussing ways to get access to Tesla solar products using the hardwired ethernet/LAN network connection (not wifi). It sounds like the inverter is connected only with wifi?

Regardless, it might depend on what type of inverter you have. There are two types of Tesla Solar Inverters (non-Powerwall) that I'm aware of. One with site controller and one without. To determine which you have see this [Tesla reference](https://energylibrary.tesla.com/docs/Public/Solar/Inverter/InstallManual/en-us/GUID-15CA6519-C179-434D-A25C-1039FF4EE46B.html)

If you have the one without Site Controller, I'm not 100% sure that string-level metrics are available with it through pypowerwall. The easiest way to check would be to sign into the inverter using the Tesla One app and examine the metrics it reports. If you see string level data reported in that app, then it is theoretically possible.

Nexarian, 2026-02-26:
@kentsimon50 192.168.91.1 only works as a valid IP if (and only if) one of the following is true:
- You have a static route set
- You are connected to the inverter's wifi (and there's only one wifi connection) -- If you're connected to the wifi of the inverter AND the powerwall or gateway, it'll get confused. You have to use a static route to override that as well.

I'm nearly certain that the example.py from my branch will work on the site controller inverter. I just switched my entire monitoring stack back to hardwired only (and unplugged the wifi cards), and I'm using the IP addresses that my router sees, no more 192.168.91.1 or static routes needed!

You are going to have an issue with needing to monitor multiple devices though. If we get one working, we can investigate modifying powerwall-server to get it so you can see all your devices in a single pane of glass.

Nexarian, 2026-02-27:
@kentsimon50 Try `nexarian/pypowerwall-nexarian:0.0.1` -- It's based on [this branch](https://github.com/Nexarian/pypowerwall/tree/re-work-proxy-next)

This is my custom version of pypowerwall that is based on an older version. I don't want this to compete with the mainline pypowerwall (I'd rather we invest in pypowerwall-server), but this also has the bearer token login feature and it can handle multiple inverters if you set up the environment variables.

```
PW_DEBUG=no
TZ=America/New_York

# Inverter 1
PW_EMAIL_1=[Tesla Acct Email]
PW_PASSWORD_1=
PW_GW_PWD_1=[QR Code Password]
PW_HOST_1=[IP Address]
PW_PORT_1=8685
PW_TIMEZONE_1=America/New_York
PW_STYLE_1=solar
PW_NEG_SOLAR_1=yes
PW_CACHE_EXPIRE_1=30
PW_TIMEOUT_1=15
PW_POOL_MAXSIZE_1=10

# Inverter 2
PW_EMAIL_2=[Tesla Acct Email]
PW_PASSWORD_2=
PW_GW_PWD_2=[QR Code Password]
PW_HOST_2=[IP Address]
PW_PORT_2=8675
PW_TIMEZONE_2=America/New_York
PW_STYLE_2=solar
PW_NEG_SOLAR_2=yes
PW_CACHE_EXPIRE_2=30
PW_TIMEOUT_2=15
PW_POOL_MAXSIZE_2=10
```

kentsimon50, 2026-02-27:
Could you build the nexarian/pypowerwall-nexarian:0.0.1 to work on amd64?

jasonacox, 2026-03-01:
Thanks @nalditopr  - I did get it working. I do wonder if the 10.42.1.x subnet path will stick long term. I assume the third party integrations require that so it should have a longer life. :-)  Have you tried other subnets?  I may do that to see what else I can find.

For any of you who already ahve a FleetAPI account, please help us test. I provided detailed information on steps I took to get it to work here: https://github.com/jasonacox/pypowerwall/pull/265#issuecomment-3978757154

nalditopr, 2026-03-01:
@rick-davis-39 Great to hear! Thanks for testing. 

For using in the dashboard I had to rebuild the docker locally using a local python wheel for pypowerwall. I can share the process later if you need it. 

Also, I hit a bug with getting data from the followers but I think is related to not flipping the followers breakers when enrolling the fleetapi key. I cannot test for about a week but will try to recreate the resgistration and flip all the breakers.

Tracking issue here. 

https://gist.github.com/nalditopr/129c811172c3e4e1d633308da05267e4

Nexarian, 2026-03-01:
Not giving up on my bearer approach :)

@kentsimon50 Try this docker image: `docker run --rm docker.io/nexarian/pvi_status:0.0.1 --host [LAN IP] --password [QR Code Password] config`

Built from https://github.com/Nexarian/tesla-solar-reporter/blob/main/pvi_status/Dockerfile.status using:

`docker buildx build --platform linux/amd64,linux/arm64 -f Dockerfile.status -t pvi-status . -t nexarian/pvi_status:0.0.1`

Nexarian, 2026-03-01:
> Did you want to add din as an option?

Don't fully grok what you mean here?

rick-davis-39, 2026-03-01:
"Thanks @rick-davis-39 ! I see your are using the 10.42.1.x subnet. Can you describe how you built your network?"

First of all, I already had a Tesla Fleet app registration for Home Assistant. I just used this and everything seems happy.
Question, will I have to regenerate tedapi_rsa_private.pem at any point now that this is operational?

Anyway, now to the build info you @jasonacox requested.

I use a virtualized (hyper-v) pfsense as my main firewall/router. Fortunately, with more good luck than good management, I had a spare unused physical interface on the host server. I created another virtual switch using this interface and then added it to the pfsense VM as another interface.

[image]

[image]

[image]

Configured this with a static IP (10.42.1.100/24)

[image]

Next, I created a VLAN in my switch using the ports that connect this interface and the leader Powerwall, plus a testing port.

Back in pfsense, configured & enabled a DHCP server only on the PW_LAN interface with the following settings:

[image]

Successful DHCP registration after using Tesla One to recommit DHCP. Probably didn't need to do this as previous addresses for leader and follower had no comms beyond the PW_LAN subnet and possibly would have attempted registration by themselves? Not sure. Just wanted to speed things up! 

[image]

Could probably reduce this as pypowerwall host(s) are on my primary network subnet (192.168.5.0/24)

I then created a LAN firewall rule that only allows nominated clients (hosts) to access the PW_LAN subnet, rejecting all others. pypowerwall hosts are members of this alias (TEDAPI_Clients) as well as a mangement/testing host

[image]

Created a simple firewall rule that allows the PW_LAN subnet hosts to communicate externally beyond this subnet.

[image]

Next, created an outbound NAT rule for the PW_LAN interface that rewrites the source to the interface address (10.42.1.100). Tested TCP/ICMP with another host plugged in to Powerwall VLAN and confirmed that the source is indeed logged as 10.42.1.100.

[image]

End result with all of the above plus your guidance on how to use this with the Powerwall Dashboard.

[image]

One thing, even though the solar generation is correct I see NO power for Solar Strings C1, D1, E1, F1. Strange.

Anyway, thanks again for your help and the great work you guys are doing. It is very much appreciated.

Thx,

Rick

jasonacox, 2026-03-01:
Thanks @rick-davis-39 !

> One thing, even though the solar generation is correct I see NO power for Solar Strings C1, D1, E1, F1. Strange.

Yes, that's the bug that @nalditopr mentions - the Follower's data is just a mirror the the Leader.  Hopefully we can figure that out.

Did you happen to try other subnets (besides 10.42.1.x/24)? I suspect that is a reserved vendor subnet but won't have time to try others today unfortunately.

Nexarian, 2026-03-02:
Idea: What subnet was the device used to create the RSA key on when it was created?

rick-davis-39, 2026-03-02:
@Nexarian , in my case 192.168.5.0/24

kentsimon50, 2026-03-02:
@Nexarian Yes and Yes

kentsimon50, 2026-03-02:
@Nexarian Thank You I would love to try it.

nalditopr, 2026-03-03:
@rick-davis-39 Thanks for testing!

I'm out of ideas on how to get the followers registered. I might need to sniff the eth traffic between leader and followers and see if I there are any other api endpoints we don't see yet.

nalditopr, 2026-03-03:
> @nalditopr What I did:
> - Download Tesla One & Tesla App APK
> - Open files in VS code
> - Use Claude Code to analyze.

I'm going to try the SPAN Installer app APK as well.

Nexarian, 2026-03-05:
@kentsimon50 Actually the PW_STYLE_2 parameter is probably meaningless -- As I mentioned this is a bit of a hacky branch.

If you connect to your Powerwall 2 using wifi and put in 192.168.91.1 as the IP for it, it should aggregate both properly, but I will concede I've only tested this with my scenario: 2x vanilla inverters.

I think the RSA token approach that others above have mentioned would work if we can integrate my inverter approach, but we should probably do that with pypowerwall-server.

Mind if I reach out by email so we don't have to bother literally everyone online? Then we can iterate until we get a working solution and share with the community.

jasonacox-sam, 2026-09-06:
This thread covered the Powerwall firmware 25.10 changes that blocked TEDAPI access for many users — including follower-gateway RSA registration, subnet requirements, and standalone inverter access. A major outcome: the bearer-token approach validated by @Nexarian and @kentsimon50 in this thread has now shipped as [v0.17.0 – TEDAPI Bearer Authentication Mode](https://github.com/jasonacox/pypowerwall/releases/tag/v0.17.0), giving inverter-only users a supported local auth path.

Closing as resolved in v0.17.0 and follow-up releases. If you're still blocked on a current version, please open a new issue with your firmware version and setup details.

— Sam ⚡

closed 2026-09-06 by jasonacox-sam