jasonacox/pypowerwall #172: Problem with string data in v4.6.7 (2 PW3's)

heynorm, 2025-04-27:
I have 2 Powerwall 3's and after upgrading from  v4.6.6 to v4.6.7 and I am no longer seeing string data from my second Powerwall.  This was working fine with v4.6.6.  I run the Dashboard on a Pi that is wirelessly connected directly my primary PW3 (I switched to this setup a few weeks ago after being upgraded to fw 25.10.1).  I ran PW3_Strings.py and it appears to me that it is not seeing my second PW3 at all.   Output of verify.sh is below.  Let me know what additional information would be helpful.   Thanks - Earl

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.12.11 Proxy t72
 - Powerwall State: CONNECTED - Firmware: 25.10.1
 - Site Name: Stahl HH
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Connected - Mode: full
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 35.8}
 - Version: 0.2.3

All tests succeeded.

jasonacox, 2025-04-30:
Hi @heynorm  - I rolled back the changes of #169 related to the device API. I wonder if this is the cause of what you are seeing. Can you test by editing your powerwall.yml file and change the pypowerwall image to this:

```
jasonacox/pypowerwall:0.12.11t72-beta172
```

Restart with:

```bash
./compose-dash.sh up -d
```

Let me know that helps.

heynorm, 2025-04-30:
@jasonacox - I made the change you recommended and it does solve the issue - thanks!.  Let me know how I can be of further help.

jasonacox, 2025-05-11:
Hi @heynorm  - I added logic that should work for single or multiple PW3 installations. Can you test on you system by editing your powerwall.yml file and change the pypowerwall image to this:

```
jasonacox/pypowerwall:0.12.12t72-beta
```

Restart with:

```bash
./compose-dash.sh up -d
```

Also @johncuthbertuk could you do the same for your single PW3 installation?

I always get nervous about editing any of the PW3 code because I can't test it on my system. 🤞

heynorm, 2025-05-12:
@jasonacox - I make the change to the powerwall.yml file you requested.  It all looks good on my system.

Thanks!

jasonacox, 2025-05-12:
Thanks @heynorm ! 🙏   I'll proceed with releasing this as the next version.

PR #175 (2025-05-12): Fix logic for multiple PW3 battery blocks
## v0.12.12 - Multiple PW3 Battery Blocks

* Bug Fix - Logic added in https://github.com/jasonacox/pypowerwall/pull/169 does not iterate through all PW3 strings. This adds logic to handle multiple PW3 string sets. 

Reported in https://github.com/jasonacox/pypowerwall/issues/172. 

Closes #172
closed 2025-05-12 by jasonacox