jasonacox/pypowerwall #179: Firmware 25.10.4: New warnings for missing POD data

JohnJ9ml, 2025-05-23:
pypowerwall 0.12.12:

I was brought forward to 25.10.4 last night. As of this morning, I'm seeing the following from the proxy:

`05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ActiveHeating]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ChargeComplete]
05/23/2025 08:24:26 AM [pypowerwall.tedapi.pypowerwall_tedapi] [DEBUG]  -- tedapi: Request for /api/system_status/grid_status
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ChargeRequest]
05/23/2025 08:24:26 AM [pypowerwall.tedapi] [DEBUG] Using Cached Payload
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_DischargeComplete]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_PermanentlyFaulted]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_PersistentlyFaulted]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_enable_line]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_available_charge_power]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_available_dischg_power]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ActiveHeating]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ChargeComplete]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ChargeRequest]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_DischargeComplete]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_PermanentlyFaulted]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_PersistentlyFaulted]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_enable_line]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_available_charge_power]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_available_dischg_power]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ActiveHeating]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ChargeComplete]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_ChargeRequest]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_DischargeComplete]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_PermanentlyFaulted]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_PersistentlyFaulted]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_enable_line]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_available_charge_power]
05/23/2025 08:24:26 AM [proxy] [DEBUG] Missing key in payload [POD_available_dischg_power]
05/23/2025 08:24:26 AM [pypowerwall.tedapi] [DEBUG] Using Cached Payload
05/23/2025 08:24:26 AM [pypowerwall.tedapi] [DEBUG] Using Cached Payload
05/23/2025 08:24:26 AM [pypowerwall.tedapi.pypowerwall_tedapi] [DEBUG]  -- tedapi: Request for /api/operation
05/23/2025 08:24:26 AM [pypowerwall.tedapi] [DEBUG] Using Cached Payload
05/23/2025 08:24:26 AM [proxy] [DEBUG] 192.168.1.37 "GET /pw/power HTTP/1.1" 200 -
`

I felt like I saw something along these general lines on another topic, but haven't located it right now.

clonea1, 2026-11-09:
Checked on 0.24.0 (proxy t113), Backup Gateway 2 + 2x Powerwall 2, firmware 26.34.2, v1r, with `PW_DEBUG=yes`: no "Missing key in payload" lines at all, so the log noise in this report is gone and this could probably be closed.

One related thing for whoever touches POD next: in TEDAPI mode the vitals set `POD_ActiveHeating`, `ChargeComplete`, `ChargeRequest`, `DischargeComplete`, `PermanentlyFaulted`, `PersistentlyFaulted` and `POD_enable_line` to `None` (no source for them; `tedapi/__init__.py` around the TEPOD block), and the proxy's `/pod` turns that into `0` via `int(get_value(...) or 0)`. Dashboards therefore show "not faulted / enable_line 0" as if it were measured. Here `enable_line` reads 0 on both units while they're discharging. `POD_available_charge_power` / `POD_available_dischg_power` (7000 W, from msa) and the energy values are real. I didn't send a PR since passing `None` through would change the `/pod` shape that Telegraf configs rely on. Flagging it in case you'd rather mark those as unavailable.

jasonacox, 2026-11-12:
Closing this one. Thanks @clonea1 for re-checking it on 0.24.0 with `PW_DEBUG=yes` and current firmware. No more "Missing key in payload" lines, so the log noise in this report is gone.

The `/pod` point you raised is a fair catch. In TEDAPI mode the vitals have no source for `POD_ActiveHeating`, `ChargeComplete`, `ChargeRequest`, `DischargeComplete`, `PermanentlyFaulted`, `PersistentlyFaulted` or `POD_enable_line`, so they come through as `None`. Then `/pod` turns each one into `0` with `int(... or 0)`, and that reads like a measured "not faulted". The pypowerwall-server `/pod` does the same thing. You're right that we can't just pass `None` through, because existing Telegraf configs depend on that shape. That's a separate change from this issue, and I'll handle it on its own.

Thanks @JohnJ9ml for the original report and @hiSandog for the input.

closed 2026-11-12 by jasonacox