jasonacox/pypowerwall #197: Issues setting Powerwall Mode since recent Firmware upgrade

DaveCudd, 2025-07-21:
My Powerwall has recently been upgraded to version 25.18.4 b6b41ca8.
Soon after this, the set_mode() function has not been successful.
I'm not having errors reported, but the command doesn't change the mode between "self_consumption" and "autonomous"

When I use the following calls in my script, they fail to change the mode.
pw.set_mode('autonomous')
pw.set_mode('self_consumption')
But they used to work.

If I make the call from the command line
python -m pypowerwall set -mode self_consumption
It changes the state.

What have I done to break the operation of the program/script I've been running?

jasonacox, 2025-07-21:
This command uses the cloud mode (via teslapy) or fleetapi mode to make these changes. I'm seeing problems with the cloud mode to set operations too. Are others seeing this on FleetAPI mode? There haven't been any changes to teslapy in a while so I suspect an API change on the Tesla side. I don't have time to investigate further due to travel but if anyone has time, code for cloude-mode is here: https://github.com/jasonacox/pypowerwall/blob/ca510b7a8c061054d47001bc13c28aee1afacd72/pypowerwall/cloud/pypowerwall_cloud.py#L1058

MikeScavell, 2025-07-22:
Hi all, I believe I have identified the issue and the fix. The issue appears to be occurring here:
https://github.com/jasonacox/pypowerwall/blob/ca510b7a8c061054d47001bc13c28aee1afacd72/pypowerwall/cloud/pypowerwall_cloud.py#L1075
This function sets the backup reserve correctly but then silently throws an error and fails to continue. So actually, the problem seems to be in the TeslaPy package rather than this one. Luckily it appears to be a very simple fix. I will open an issue and/or pull request in TeslaPy to sort it out.
If you are interested in the answer on how to fix TeslaPy yourself right now, see here:

https://github.com/tdorssers/TeslaPy/blob/fde2a33bf8653055f6cd8312259a038d1163913d/teslapy/__init__.py#L816

This line must be changed to `if response['Code'] == 200:` to match what I believe are some updates to the Tesla Cloud API. After that, things seem to work as expected!

Might be worth leaving this issue open here for the time being though, just in case other people come across the same issue.

DaveCudd, 2025-07-24:
Sorry folks, but I'm very new to Python and this library.
How can I get the functions start working again?

ayshfm1, 2025-07-24:
Get the disclaimer out the way -  I didn't do a proper job, just got back to the UK from three weeks in the US to find my powerwall not force charging which was costing me money, so tired AND then cross, professionalism out the window.  

Ran the command command manually to find out why and got the back trace (read back traces from bottom up so the last lines are the ones that are where the problem starts).  In this case it was a bit cryptic to my addled brain.

What I should have done next was search the web with the back trace and check to see what was thrown up.   What I actually did was rather stupid.   
 
  File "  File "/home/ayshfm1/powerwall/src/teslapy/teslapy/__init__.py", line 770, in command
    if response['code'] == 201:
       ~~~~~~~~^^^^^^^^

So edited that file and went to line 770, which  contains the error.   That checking code is looking at the response values from the api call above it.  I put a print statement to display what was in the returned structure.  Re-ran the command looked at the output and then changed the checking code to align with what was actually returned.

Reran and it and the ugly stack trace was gone.

After all that pre-amble what do you need to do.   Run the command manually

In my case manually looks like this:-

source /home/ayshfm1/powerwall/bin/activate
 /home/ayshfm1/powerwall/pypowerwall/tools/
python3 set-reserve.py --set 10

YMMV, basically your directory structure will be different

Edit the file which has the error thrown

vi /home/ayshfm1/powerwall/src/teslapy/teslapy/__init__.py

And make the changes I outlined in my post.

When I did a search I came across this.    

Additional points 

Mike beat me to the punch, however Message also needs to change plus the line numbers are different in his patch from mine, so I likely have a different version.   What really needs to happen is the underlying library has to be fixed and given the response structure has changed this could be rather more involved than the point fix being hacked in here.

Will the library be automatically updated, probably not, I imagine some form of pip update will be needed when it's been fixed.  Moot for now.

I don't actually think this was a problem....  After all this my powerwall still wasn't fricking well charging!   I had to reset my gateway 2 to actually get it work.

ayshfm1, 2025-07-24:
Sigh.   This problem is not your problem MikeCudd.  Post the change the setmode command generates a stack trace it did not before, so you don't want to hack the library.

python3 set-mode.py --set autonomous
{
    "code": 201,
    "message": "Updated"
}
[Stack trace error]

This call returns exactly what the original expects without the uppercase first letter change joy-joy.

So I've been no help to you.

Vikings39, 2025-07-24:
Thanks very much for the responses. I'm still a bit confused now as to what is being suggested. Should I amend the __init__.py code to look for 200 instead of 201? Or should it be 'Code' instead of 'code'? Or is there something more I need to do?

Sorry to be a pain but it's just there seem to be a number of different options listed above and I couldn't pick out what was the final conclusion.

Thanks for the help.

MikeScavell, 2025-07-24:
There seems to have been a bit more work than what I originally did here. All the points are very correct in that it seems to return a slightly different message depending on whether you do a set-mode or a set-reserve. I'll make some changes in the public TeslaPy repo that I forked to cover both cases right now. Stand by for updates!

https://github.com/MikeScavell/TeslaPy/tree/tesla-api-fix

DaveCudd, 2025-07-25:
Sorry guys to be a complete newby here.
I think I deployed the pypowerwall functions using the pip installer and I can't find any trace of the code on my machine.
Do I need to download the installation and save it to a folder and then edit the code suggested?

jasonacox, 2025-07-26:
Great work @MikeScavell  on the PR https://github.com/tdorssers/TeslaPy/pull/171

Once @tdorssers publishes a release, pypowerwall will get the patch. In the meantime, a test:

```bash
# install latest from git
pip install git+https://github.com/tdorssers/TeslaPy.git --force

# set to backup
python -m pypowerwall set -mode backup
python -m pypowerwall get
```
```
pyPowerwall [0.13.2] - Get Powerwall Mode and Power Levels using cloud mode.

  Site              Cox Energy Gateway
  Site Id           xxxxxxxxxxxxxx
  Din               1232100-00-E--TG123456789ABG
  Mode              backup
  Reserve           20.0
  Current           64.23317468434095
  Grid              11584
  Home              4854
  Battery           -6730
  Solar             0
  Grid Charging     True
  Grid Export Mode  pv_only
```

```bash
# set to self_consumption
python -m pypowerwall set -mode self_consumption
# set reserve
python -m pypowerwall set -reserve 22 
python -m pypowerwall get
```
```
pyPowerwall [0.13.2] - Get Powerwall Mode and Power Levels using cloud mode.

  Site              Cox Energy Gateway
  Site Id           xxxxxxxxxxxxxx
  Din               1232100-00-E--TG123456789ABG
  Mode              self_consumption
  Reserve           22.0
  Current           64.23317468434095
  Grid              9488
  Home              5268
  Battery           -4220
  Solar             0
  Grid Charging     True
  Grid Export Mode  pv_only
```

DaveCudd, 2025-07-26:
Thanks @jasonacox but I get the following error when I run the install.

# pip install git+https://github.com/tdorssers/TeslaPy.git --force
Collecting git+https://github.com/tdorssers/TeslaPy.git
  Cloning https://github.com/tdorssers/TeslaPy.git to /tmp/pip-req-build-b8ivqskg
  Running command git clone --filter=blob:none --quiet https://github.com/tdorssers/TeslaPy.git /tmp/pip-req-build-b8ivqskg
  /opt/QGit/libexec/git-core/git-remote-https: error while loading shared libraries: libgnutls.so.30: cannot open shared object file: No such file or directory
  error: subprocess-exited-with-error

jasonacox, 2025-07-27:
Interesting... your system seems to be missing the gnu security library used to handle the git install.  You could try cloning and installing the repository manually: 

```bash
git clone https://github.com/tdorssers/TeslaPy.git
cd TeslaPy
pip install .
```

MikeScavell, 2025-07-27:
If you are having issues with the git clone command (I've personally never had the issue you are describing), you can also just download the code as a zip file. Then extract the files, and proceed with the `cd TeslaPy` and `pip install .` commands.

Alternatively, you could put some of your errors in to ChatGPT, it can be quite good at assisting with debugging.

jasonacox, 2025-07-27:
Hi @DaveCudd  - If you already had it installed, it may not replace the older version since the version number is the same. You can try uninstalling it first (e.g. pip uninstall teslapy). You can also uninstall pypowerwall and install the latest from  https://github.com/jasonacox/pypowerwall as a zip too. I embedded the patched TeslaPy directly into pypowerwall's code.

DaveCudd, 2025-07-28:
Thanks @jasonacox 
That's solved my issue. :)

I was going to download the whole Codebase for pypowerwall and overlay the TeslaPy code over the top and install it manually, but this saved me having to try that.

Appreciated.

jasonacox, 2025-07-28:
Awesome! Thanks for the confirmation @DaveCudd !

I'll close this issue but feel free to post or reopen.

closed 2025-07-28 by jasonacox