jasonacox/pypowerwall #200: backup

ayshfm1, 2025-07-26:
This probably worth a read

https://teslamotorsclub.com/tmc/threads/powerwall-charge-throttled-at-80-with-25-18-4-setting-pw-reserve-between-81-99-restricted-by-tesla-in-some-modes.347099/

Set-mode supports time and selfpowered.   Any reason it doesn't also support backup?   We might be needing it.   I tested the underlying code works, so it's just the validation and boiler plate that needs extending.

date - set_mode("backup")

(powerwall) ayshfm1@packer:~/powerwall/pypowerwall/tools$ python3 set-mode.py --read
READ: Current Operational Mode: backup with 1 Powerwalls

jasonacox, 2025-07-27:
Yes, the set-mode.py script is a bit dated. It needs some attention or better yet, reworked to be a proxy for pypowerwall's native CLI:

https://github.com/jasonacox/pypowerwall?tab=readme-ov-file#powerwall-command-line

```bash
# Setup Connection with Tesla Cloud
python -m pypowerwall setup

# Get Power Levels, Operation Mode, and Battery Reserve Setting
#
# Usage: PyPowerwall get [-h] [-format FORMAT]
#  -h, --help      show this help message and exit
#  -format FORMAT  Output format: text, json, csv
#
python -m pypowerwall get
python -m pypowerwall get -format json
python -m pypowerwall get -format csv

# Set Operation Mode and Battery Reserve Setting
#
# Usage: PyPowerwall set [-h] [-mode MODE] [-reserve RESERVE] [-current]
#  -h, --help        show this help message and exit
#  -mode MODE        Powerwall Mode: self_consumption, backup, or autonomous
#  -reserve RESERVE  Set Battery Reserve Level [Default=20]
#  -current          Set Battery Reserve Level to Current Charge
#
python -m pypowerwall set -mode backup
python -m pypowerwall set -reserve 30
python -m pypowerwall set -current
```

Note: right now the mode setting is not working due to an API change by Teala that required an update on TeslaPy. That change was accepted but we are waiting on it to be published to pull in to pypowerwall.

JonMurphy, 2025-07-27:
Wait?!? What?  pypowerwall has the ability to send commands to a Powerwall (hopefully a PW3)??  Where can I learn more??

jasonacox, 2025-07-27:
Haha! @JonMurphy Yes works with all Powerwalls. See the CLI options: https://github.com/jasonacox/pypowerwall?tab=readme-ov-file#pypowerwall-command-line-interface-cli

JohnJ9ml, 2025-07-28:
I know it's a paying service and folks aren't always happy about that, but I've been harvesting my car data from TeslaFi for years, and they also offer the ability to send commands through an API as well.

JohnJ9ml, 2025-07-28:
yes, i admit it's pricey, but I've been gathering my data from them off the car for a very long time. I'm able to track my charging state that way, as well as hold onto the charging data from each session. (Never figured out much to do with it, but who can stop storing the stuff, eh?) Biggest bit I guess was knowing how much of my PG&E bill was charging the  car versus running the house.

jasonacox, 2025-07-29:
I suspect TeslaFi uses FleetAPI same as NetZero. Tesla now charges for using FleetAPI but if your usage is small, it may be under the free tier (see https://developer.tesla.com/#usage-based-pricing). I only set up FleetAPI commands in pypowerwall for items related to Powerwalls and Solar Energy. Someone could use it and add vehicle control commands or data polling.

JonMurphy, 2025-07-29:
I had looked into using FleetAPI but I couldn't quite understand the instructions for a raspberry pi.  I need a beginners (idiots) guide.

jasonacox, 2025-08-03:
Totally agree. Tesla made that for commercial businesses to use so it assume you have your own engineering team, own a company and host your own websites.

jasonacox, 2025-08-10:
Updated scripts to use patched teslapy: https://github.com/jasonacox/pypowerwall/tree/main/tools

rkokfhung, 2025-08-10:
> Updated scripts to use patched teslapy: https://github.com/jasonacox/pypowerwall/tree/main/tools

Thanks @jasonacox got it working again!!