jasonacox/pypowerwall #208: ERROR Unknown error connecting to Powerwall at https:///api/operation: Invalid URL 'https:///api/operation': No host supplied

jamski68, 2025-08-31:
Any ideas why this error occurs when trying to update the reserves at `/control/reserves` the `PW_HOST` variable is set and all other GET operations are working other than the control operations

jasonacox, 2025-09-01:
Brilliant. Please do! Thanks! 🙏

jasonacox-sam, 2026-09-06:
This `Invalid URL 'https:///api/operation': No host supplied` error came from trying to use control mode in local-only mode. As Jason explained, local control was removed by Tesla — you need cloud credentials (`PW_EMAIL` + cloud setup) so reads stay local but control goes via the cloud: https://github.com/jasonacox/pypowerwall/tree/main/proxy#control-mode. The reporter confirmed he'd figured this out and offered a docs PR.

Closing as resolved. Anyone hitting the empty-URL error: the fix is cloud-mode control setup, as documented.

— Sam ⚡

closed 2026-09-06 by jasonacox-sam