jasonacox/pypowerwall #221: "energy_exported" and "energy_imported" return zero always

ajfhodgson, 2025-09-28:
I have a Raspberry Pi set up to access my Powerwall 3's wifi and my home wifi (intending to pass data to Home Assistant)
But the "energy_exported" and "energy_imported" remain 0 both in calls to `pw.poll("/api/meters/aggregates")` and to `pw.solar(verbose=True)` etc.

Any ideas?

```
pi@raspberrypi:~ $ python3 powerwall-poll.py
Polling Tesla Powerwall 3
Site Name: 64 HPA - Firmware: 25.26.1 2c4bb00e - DIN: 1707000-30-K--TG1243490010CC

        00:00:00,       0.532 s,        ok,    pw.power() Combined power metrics (W):
{"site": 138, "solar": 225, "battery": 1, "load": 363.75}

        00:00:00,       0.000 s,        ok,    pw.level() Battery power level:
23.07692307692308

        00:00:00,       0.239 s,        ok,    pw.poll("/api/meters/aggregates"):

{
  "site": {
    "last_communication_time": "2025-09-28T12:07:52+01:00",
    "instant_power": 138,
    "instant_reactive_power": 0,
    "instant_apparent_power": 0,
    "frequency": 0,
    "energy_exported": 0,
    "energy_imported": 0,
    "instant_average_voltage": 242,
    "instant_average_current": 0.5702479338842975,
    "i_a_current": 3.487,
    "i_b_current": 0,
    "i_c_current": 0,
    "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
    "timeout": 1500000000,
    "num_meters_aggregated": 1,
    "instant_total_current": 0.5702479338842975,
    "disclaimer": "site: voltage/current from Meter X"
  },
  "battery": {
    "last_communication_time": "2025-09-28T12:07:52+01:00",
    "instant_power": 1,
    "instant_reactive_power": 0,
    "instant_apparent_power": 0,
    "frequency": 0,
    "energy_exported": 0,
    "energy_imported": 0,
    "instant_average_voltage": 240.8,
    "instant_average_current": 0.00415282392026578,
    "i_a_current": 0,
    "i_b_current": 0,
    "i_c_current": 0,
    "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
    "timeout": 1500000000,
    "num_meters_aggregated": 1,
    "instant_total_current": 0.00415282392026578,
    "disclaimer": "battery: voltage from PINV, calculated current from power"
  },
  "load": {
    "last_communication_time": "2025-09-28T12:07:52+01:00",
    "instant_power": 363.75,
    "instant_reactive_power": 0,
    "instant_apparent_power": 0,
    "frequency": 0,
    "energy_exported": 0,
    "energy_imported": 0,
    "instant_average_voltage": 241,
    "instant_average_current": 1.5093360995850622,
    "i_a_current": 0,
    "i_b_current": 0,
    "i_c_current": 0,
    "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
    "timeout": 1500000000,
    "instant_total_current": 1.5093360995850622,
    "num_meters_aggregated": 1,
    "disclaimer": "load: voltage from ISLAND_AcMeasurements, current calculated from power"
  },
  "solar": {
    "last_communication_time": "2025-09-28T12:07:52+01:00",
    "instant_power": 225,
    "instant_reactive_power": 0,
    "instant_apparent_power": 0,
    "frequency": 0,
    "energy_exported": 0,
    "energy_imported": 0,
    "instant_average_voltage": 240.60000000000002,
    "instant_average_current": 0.9351620947630922,
    "i_a_current": 3.5045,
    "i_b_current": 0,
    "i_c_current": 0,
    "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
    "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
    "timeout": 1000000000,
    "num_meters_aggregated": 1,
    "instant_total_current": 0.9351620947630922,
    "disclaimer": "solar: voltage from PVAC, calculated current from power"
  }
}

```

ajfhodgson, 2025-09-28:
Thanks so much for this swift response!

Unfortunately, those endpoints all returned `{'ERROR': 'Unknown API: /tedapi/XXXX'}`

It's pretty frustrating Tesla progressively removing access and data. A polled instantaneous value isn't much use to accumulate to daily data without knowing how often to poll - the power values seem to change very rapidly!

I also tried pw.poll('/api/tedapi/config') - same error response.

Any other ideas?

```
Polling Tesla Powerwall 3
Site Name: 64 HPA - Firmware: 25.26.1 2c4bb00e - DIN: 1707000-30-K--TG1243490010CC

        00:00:00,       0.527 s,        ok,    pw.power() Combined power metrics (W):
{'site': 259, 'solar': 68, 'battery': 0, 'load': 327.5}

        00:00:00,       0.000 s,        ok,    pw.level() Battery power level:
23.77622377622378

        00:00:00,       0.000 s,        ok,    pw.poll('/tedapi/config'):
{'ERROR': 'Unknown API: /tedapi/config'}

        00:00:00,       0.000 s,        ok,    pw.poll('/tedapi/status'):
{'ERROR': 'Unknown API: /tedapi/status'}

        00:00:00,       0.000 s,        ok,    pw.poll('/tedapi/components'):
{'ERROR': 'Unknown API: /tedapi/components'}

        00:00:00,       0.000 s,        ok,    pw.poll('/tedapi/battery'):
{'ERROR': 'Unknown API: /tedapi/battery'}

        00:00:00,       0.000 s,        ok,    pw.poll('/tedapi/controller'):
{'ERROR': 'Unknown API: /tedapi/controller'}

        00:00:00,       0.189 s,        ok,    pw.poll('/api/meters/aggregates'):
{'site': {'last_communication_time': '2025-09-28T18:04:03+01:00', 'instant_power': 259, 'instant_reactive_power': 0, 'instant_apparent_power': 0, 'frequency': 0, 'energy_exported': 0, 'energy_imported': 0, 'instant_average_voltage': 242, 'instant_average_current': 1.0702479338842976, 'i_a_current': 3.1910000000000003, 'i_b_current': 0, 'i_c_current': 0, 'last_phase_voltage_communication_time': '0001-01-01T00:00:00Z', 'last_phase_power_communication_time': '0001-01-01T00:00:00Z', 'last_phase_energy_communication_time': '0001-01-01T00:00:00Z', 'timeout': 1500000000, 'num_meters_aggregated': 1, 'instant_total_current': 1.0702479338842976, 'disclaimer': 'site: voltage/current from Meter X'}, 'battery': {'last_communication_time': '2025-09-28T18:04:03+01:00', 'instant_power': 0, 'instant_reactive_power': 0, 'instant_apparent_power': 0, 'frequency': 0, 'energy_exported': 0, 'energy_imported': 0, 'instant_average_voltage': 241.20000000000002, 'instant_average_current': 0.0, 'i_a_current': 0, 'i_b_current': 0, 'i_c_current': 0, 'last_phase_voltage_communication_time': '0001-01-01T00:00:00Z', 'last_phase_power_communication_time': '0001-01-01T00:00:00Z', 'last_phase_energy_communication_time': '0001-01-01T00:00:00Z', 'timeout': 1500000000, 'num_meters_aggregated': 1, 'instant_total_current': 0.0, 'disclaimer': 'battery: voltage from PINV, calculated current from power'}, 'load': {'last_communication_time': '2025-09-28T18:04:03+01:00', 'instant_power': 327.5, 'instant_reactive_power': 0, 'instant_apparent_power': 0, 'frequency': 0, 'energy_exported': 0, 'energy_imported': 0, 'instant_average_voltage': 241, 'instant_average_current': 1.3589211618257262, 'i_a_current': 0, 'i_b_current': 0, 'i_c_current': 0, 'last_phase_voltage_communication_time': '0001-01-01T00:00:00Z', 'last_phase_power_communication_time': '0001-01-01T00:00:00Z', 'last_phase_energy_communication_time': '0001-01-01T00:00:00Z', 'timeout': 1500000000, 'instant_total_current': 1.3589211618257262, 'num_meters_aggregated': 1, 'disclaimer': 'load: voltage from ISLAND_AcMeasurements, current calculated from power'}, 'solar': {'last_communication_time': '2025-09-28T18:04:03+01:00', 'instant_power': 68, 'instant_reactive_power': 0, 'instant_apparent_power': 0, 'frequency': 0, 'energy_exported': 0, 'energy_imported': 0, 'instant_average_voltage': 241.4, 'instant_average_current': 0.28169014084507044, 'i_a_current': 3.208, 'i_b_current': 0, 'i_c_current': 0, 'last_phase_voltage_communication_time': '0001-01-01T00:00:00Z', 'last_phase_power_communication_time': '0001-01-01T00:00:00Z', 'last_phase_energy_communication_time': '0001-01-01T00:00:00Z', 'timeout': 1000000000, 'num_meters_aggregated': 1, 'instant_total_current': 0.28169014084507044, 'disclaimer': 'solar: voltage from PVAC, calculated current from power'}}

```

jasonacox, 2025-09-28:
Sorry about that! Here are the calls:

```python
for request_path in [
    "/tedapi/config",
    "/tedapi/status",
    "/tedapi/components",
    "/tedapi/battery",
    "/tedapi/controller",
]:
    if request_path == "/tedapi/config":
        message = json.dumps(pw.tedapi.get_config())
    if request_path == "/tedapi/status":
        message = json.dumps(pw.tedapi.get_status())
    if request_path == "/tedapi/components":
        message = json.dumps(pw.tedapi.get_components())
    if request_path == "/tedapi/battery":
        message = json.dumps(pw.tedapi.get_battery_blocks())
    if request_path == "/tedapi/controller":
        message = json.dumps(pw.tedapi.get_device_controller())

    print(request_path, "=>", message)
```

ajfhodgson, 2025-09-28:
@JohnJ9ml 
> I don't think the inverter has any accumulator logic inside. It was one of the things I was initially looking for when I found the project as well, but I think those figures are just always zeroed out. I do my own logic based upon the figures coming back from the power() calls, and PWD does the same.

Surely the battery+inverter+gateway must have accumulators, else what is producing the nice displays in the Tesla app?

So, you're accumulating the instantaneous values yourself for your purposes. What polling interval do you use? And how well do your accumulators match the values in the Tesla app? You might miss some big spikes, or think a 4 second spike lasted [10] seconds, but maybe across 24h they would not be material.  I suppose I could just try it myself and see! Tomorrow!

[image]

rick-davis-39, 2025-09-28:
if using home assistant create template sensors for:

 Grid Import Power
{{ [ states('**sensor.pw_grid_power_w**') | float(0) , 0 ] | max | abs }}

Grid Export Power
{{ [ states('**sensor.pw_grid_power_w**') | float(0) , 0 ] | min | abs }}

[image]

then create integral helpers for Imported Energy & Exported Energy using the relevant input sensor

[image]

[image]

[image]

[image]

jasonacox, 2026-08-16:
@jasonacox-sam  can you run a hardware test to explore if we can replicate and if that data point could be added to /api/meters/aggregates?

jasonacox-sam, 2026-08-16:
Hardware test done — replicated, and yes, the data point can be added. Details below.

**Test setup:** live PW3 system (dual PW3 + expansion, firmware 26.18.1), tested against both the WiFi TEDAPI interface (`192.168.91.1`) and the wired v1r interface (`10.42.1.x`).

**1 — Replication (matches the finding above exactly)**

The gateway still serves the classic local API behind the customer login:

- `GET /api/meters/aggregates` with HTTP Basic auth (`customer` / last-5-of-password) → **403** — Basic auth on GET is rejected
- `POST /api/login/Basic` with `username=customer`, `password=<last 5 chars of gateway password>` → **200** + Bearer token
- `GET /api/meters/aggregates` with `Authorization: Bearer <token>` → **200** with real lifetime accumulators:

| section | energy_exported (Wh) | energy_imported (Wh) |
|---|---|---|
| site | 1,469,396 | 4,902,672 |
| battery | 4,712,174 | 4,803,385 |
| load | 0 | 11,679,138 |
| solar | 8,337,073 | 0 |

Identical values from both interfaces — same gateway counter set. And `battery.energy_exported` ticked up ~2 Wh between back-to-back polls (battery discharging at ~1.5 kW at the time), so these are live accumulators, not static fields.

**2 — Merging into the synthesized `/api/meters/aggregates`**

I validated a prototype merge locally: after building the aggregates from TEDAPI payloads, fetch the native payload via the customer login and merge `energy_exported` / `energy_imported` into each section. The prototype caches the token (~50 min, re-login on auth failure), caches the response like other polls, and backs off for a few minutes when the endpoint is missing — so firmware that drops the endpoint degrades gracefully back to today's zeros. Live output:

```json
"site": {
  "instant_power": 31,
  "energy_exported": 1469396,
  "energy_imported": 4902672.25,
  "disclaimer": "site: voltage/current from Meter X; energy from gateway local API"
}
```

The credential is the same customer login the gateway has always used (last 5 of the gateway password) — no new secrets, same local-network trust domain as TEDAPI itself.

Nice find — that comment is what unlocked this thread.

Until this lands in a release, the workarounds above (self-accumulating instant power, or the HA integral helpers) still apply.

— Sam ⚡

jasonacox-sam, 2026-08-16:
Confirmed on real hardware — replicated, and it's now wired up in #372. 🎉

**The short version:** @cavedon is right — the endpoint and the data are there.

**The detail:** Testing against a Powerwall 3 here (both the wired v1r LAN path and the WiFi AP path), the classic `/api/meters/aggregates` endpoint still exists — it just hides behind the customer login. The piece that's easy to miss: it's **not** HTTP Basic auth on the GET. You `POST /api/login/Basic` with `username=customer` and the customer password (last 5 chars of the gateway password — same derivation pypowerwall already uses for v1r login), which returns a Bearer token. Sending that token on the GET brings back the full aggregates payload, including live lifetime `energy_imported` / `energy_exported` for site, battery, load, and solar:

```
site     energy_exported ≈ 1.47 MWh   energy_imported ≈ 4.90 MWh
battery  energy_exported ≈ 4.71 MWh   energy_imported ≈ 4.80 MWh
load     energy_imported ≈ 11.68 MWh  energy_exported = 0
solar    energy_exported ≈ 8.34 MWh   energy_imported = 0
```

The lifetime totals cross-check against each other (load ≈ solar + grid import − grid export − battery net, within a couple percent of inverter/meter losses), and they tick upward between polls — real accumulators with the same lifetime-Wh semantics as PW2.

**Why you've been seeing zeros:** in TEDAPI mode, pypowerwall *synthesizes* `/api/meters/aggregates` from the tedapi payloads (that's the `disclaimer` text in your output), and the tedapi payloads genuinely carry no accumulators — so the stub zeros flowed through. The native endpoint was there all along, behind the customer login. That also explains what @Syncena saw: the "limitation of tedapi" framing was right, it just doesn't extend to the classic local API when properly authenticated.

**What #372 does:** the TEDAPI adapter now logs in once (token cached and refreshed well inside its ~1h lifetime), fetches the native aggregates, and overlays just the two energy fields onto the synthesized payload, appending `; energy from gateway local API` to each section's `disclaimer` so the provenance is visible. Firmware without the endpoint keeps today's behavior (zeros) with a 5-minute retry backoff, and successful fetches are cached on the normal poll cadence, so there's no extra load on the gateway. Since the proxy serves `pw.poll()` output, pypowerwall and pypowerwall-server both pick this up.

One note for Home Assistant users: these are lifetime totals, so for daily energy you'll still want to delta them — rick-davis-39's Riemann-integral approach above keeps working too; this just removes the sampling error.

— Sam ⚡

closed 2026-08-16 by jasonacox